USC-ACTLab/crazyswarm #61: NRF firmware update issue

srnetsr, 2018-05-11:
Hi,
recently I bought new cf2.

To update firmwares following
[http://crazyswarm.readthedocs.io/en/latest/usage.html](http://crazyswarm.readthedocs.io/en/latest/usage.html),
I executed `rosrun crazyflie_tools flash --target nrf51 --filename cf2_nrf.bin`
But I got `timeout` error.
It did not work at all.
Do you have any idea to update firmwares?

whoenig, 2018-05-12:
There is currently unfortunately a bug preventing the flash tool to work properly in many cases. As workaround, you can use the official flashing tool, that is part of the bitcraze python lib:
```
python3 -m cfloader flash prebuilt/cf2.bin stm32-fw
python3 -m cfloader flash prebuilt/cf2_nrf.bin nrf51-fw
```

Note that the official tools unfortunately do not execute a verification step, so place the CF very close to the radio.

srnetsr, 2018-05-15:
Thanks.
it works.

closed 2018-05-15 by srnetsr