USC-ACTLab/crazyswarm #62: Drone crashes after 8 meters

pPatrickK, 2018-05-15:
Hello! :) 
After updating to the newest version we have a similar problem like [this old issue]( https://github.com/USC-ACTLab/crazyswarm/issues/56).
We tried to change the firmware again (setting `static float const POSITION_LIMIT = 15.0f; // meters`) but now we have new problems: 

- The drone should fly 9 meters but just flew ~4.5m. 
- The drone should fly at a height of 30cm but it actually flies at about 15cm. 
- The drone wiggles when flying. 

What can we do to solve these problems?

whoenig, 2018-05-15:
Hi,
There is now another place where you need to change the value as well: https://github.com/whoenig/crazyflie_cpp/blob/master/src/Crazyflie.cpp#L1283 (crazyflie_cpp is a submodule of crazyflie_ros in the ros_ws/src folder.)

Background:
We integrated the special fork of crazyflie_ros into the official version with that release. In crazyflie_ros, we don't want to have a build dependency on the firmware at this point; thus it needs to be changed in the firmware and host code.

On another note: With the official firmware merge we also changed the respective packets to be in mm. That means by default you'll be able to fly in -16...16m large spaces. We have an internal version of the crazyswarm package that can optionally work with the official firmware, but didn't release it yet.

Hope that helps,
Wolfgang

pPatrickK, 2018-05-16:
Thank you, that worked.

Best regards
Patrick

closed 2018-05-16 by pPatrickK