USC-ACTLab/crazyswarm #64: Drones randomly falls onto the ground

pPatrickK, 2018-05-15:
Sometime a drone randomly falls down.
We see this problem since the new release. It seems like the engines just shut off for no reason.
Do you have some hints for us to solve this problem?
We saw this problem with different scripts. For example, "figure8_csv.py" and others. Even on own scripts with simple goto commands.
We see those "Don't know ack" error messages:

> [ INFO] [1526389834.069563575]: [cf20] Update parameters
> [ INFO] [1526389834.131466331]: Update params: 0.072950 s
> [ INFO] [1526389834.131564272]: Started 1 threads
> [ INFO] [1526389839.818718904]: [cf20] Upload trajectory
> [ WARN] [1526389839.842813651]: Latency too high! Is 0.009258 s.
> [ WARN] [1526389839.956615973]: Latency too high! Is 0.012839 s.
> [ WARN] [1526389839.992218644]: Don't know ack: Port: 4 Channel: 2 Len: 7
> [ WARN] [1526389839.993758488]: Don't know ack: Port: 4 Channel: 2 Len: 7
> [ WARN] [1526389839.998673977]: Don't know ack: Port: 4 Channel: 2 Len: 7
> [ WARN] [1526389840.001823660]: Don't know ack: Port: 4 Channel: 2 Len: 7
> [ WARN] [1526389840.003410951]: Don't know ack: Port: 4 Channel: 2 Len: 7
> [ WARN] [1526389840.005012624]: Don't know ack: Port: 4 Channel: 2 Len: 7
> [ WARN] [1526389840.008210921]: Don't know ack: Port: 4 Channel: 2 Len: 7
> [ WARN] [1526389840.009796982]: Don't know ack: Port: 4 Channel: 2 Len: 7
> [ WARN] [1526389840.013099888]: Don't know ack: Port: 4 Channel: 2 Len: 7
> [ WARN] [1526389840.014687268]: Don't know ack: Port: 4 Channel: 2 Len: 7
> [ WARN] [1526389840.019393615]: Don't know ack: Port: 4 Channel: 2 Len: 7
> [ WARN] [1526389840.022582420]: Don't know ack: Port: 4 Channel: 2 Len: 7
> [ WARN] [1526389840.024152848]: Don't know ack: Port: 4 Channel: 2 Len: 7
> [ WARN] [1526389840.028920479]: Don't know ack: Port: 4 Channel: 2 Len: 7
> [ WARN] [1526389840.032163458]: Don't know ack: Port: 4 Channel: 2 Len: 7
> [ WARN] [1526389840.033717457]: Don't know ack: Port: 4 Channel: 2 Len: 7
> [ WARN] [1526389840.036857618]: Don't know ack: Port: 4 Channel: 2 Len: 7
> [ WARN] [1526389840.038421661]: Don't know ack: Port: 4 Channel: 2 Len: 7
> [ WARN] [1526389840.040019073]: Don't know ack: Port: 4 Channel: 2 Len: 7
> [ WARN] [1526389840.043349168]: Don't know ack: Port: 4 Channel: 2 Len: 7
> [ WARN] [1526389840.044896328]: Don't know ack: Port: 4 Channel: 2 Len: 7
> [ WARN] [1526389840.047966159]: Don't know ack: Port: 4 Channel: 2 Len: 7
> [ WARN] [1526389840.049552269]: Don't know ack: Port: 4 Channel: 2 Len: 7
> [ WARN] [1526389840.052748658]: Don't know ack: Port: 4 Channel: 2 Len: 7
> [ WARN] [1526389840.054290630]: Don't know ack: Port: 4 Channel: 2 Len: 7
> [ WARN] [1526389840.057396114]: Don't know ack: Port: 4 Channel: 2 Len: 7
> [ WARN] [1526389840.058900243]: Don't know ack: Port: 4 Channel: 2 Len: 7
> [ WARN] [1526389840.062170358]: Don't know ack: Port: 4 Channel: 2 Len: 7
> [ WARN] [1526389840.063746538]: Don't know ack: Port: 4 Channel: 2 Len: 7
> [ WARN] [1526389840.066887262]: Don't know ack: Port: 4 Channel: 2 Len: 7
> [ WARN] [1526389840.068442953]: Don't know ack: Port: 4 Channel: 2 Len: 7
> [ WARN] [1526389840.070165117]: Don't know ack: Port: 4 Channel: 2 Len: 7
> [ WARN] [1526389840.073449530]: Don't know ack: Port: 4 Channel: 2 Len: 7
> [ WARN] [1526389840.075050056]: Don't know ack: Port: 4 Channel: 2 Len: 7
> [ WARN] [1526389840.078349725]: Don't know ack: Port: 4 Channel: 2 Len: 7
> [ WARN] [1526389840.089618555]: Don't know ack: Port: 4 Channel: 2 Len: 7
> [ WARN] [1526389840.090293804]: Latency too high! Is 0.011876 s.
> [ WARN] [1526389840.092082812]: Don't know ack: Port: 4 Channel: 2 Len: 7
> [ WARN] [1526389840.094823300]: Don't know ack: Port: 4 Channel: 2 Len: 7
> [ WARN] [1526389840.097462835]: Don't know ack: Port: 4 Channel: 2 Len: 7
> [ WARN] [1526389840.098389554]: Don't know ack: Port: 4 Channel: 2 Len: 7
> [ WARN] [1526389840.099342653]: Don't know ack: Port: 4 Channel: 2 Len: 7
> [ WARN] [1526389840.101932494]: Don't know ack: Port: 4 Channel: 2 Len: 7
> [ WARN] [1526389840.102870241]: Don't know ack: Port: 4 Channel: 2 Len: 7
> [ WARN] [1526389840.103765166]: Don't know ack: Port: 4 Channel: 2 Len: 7
> [ WARN] [1526389840.106355384]: Don't know ack: Port: 4 Channel: 2 Len: 7
> [ WARN] [1526389840.107907640]: Don't know ack: Port: 4 Channel: 2 Len: 7
> [ WARN] [1526389840.112907071]: Don't know ack: Port: 4 Channel: 2 Len: 7
> [ WARN] [1526389840.114475663]: Don't know ack: Port: 4 Channel: 2 Len: 7
> [ WARN] [1526389840.117611540]: Don't know ack: Port: 4 Channel: 2 Len: 7
> [ WARN] [1526389840.119152273]: Don't know ack: Port: 4 Channel: 2 Len: 7
> [ WARN] [1526389840.123952395]: Don't know ack: Port: 4 Channel: 2 Len: 7
> [ WARN] [1526389840.127157452]: Don't know ack: Port: 4 Channel: 2 Len: 7
> [ WARN] [1526389840.128264511]: Don't know ack: Port: 4 Channel: 2 Len: 7
> [ WARN] [1526389840.129214621]: Don't know ack: Port: 4 Channel: 2 Len: 7
> [ WARN] [1526389840.132868530]: Don't know ack: Port: 4 Channel: 2 Len: 7
> [ WARN] [1526389840.133833855]: Don't know ack: Port: 4 Channel: 2 Len: 7
> [ WARN] [1526389840.136345869]: Don't know ack: Port: 4 Channel: 2 Len: 7
[... 369 lines cut ...]
> [ WARN] [1526389886.320147239]: Don't know ack: Port: 4 Channel: 2 Len: 7
> [ WARN] [1526389886.483256626]: Don't know ack: Port: 8 Channel: 0 Len: 5
> [ WARN] [1526389886.495326810]: Don't know ack: Port: 8 Channel: 0 Len: 5
> [ WARN] [1526389886.518482411]: Don't know ack: Port: 8 Channel: 0 Len: 5
> [ WARN] [1526389886.530168482]: Don't know ack: Port: 8 Channel: 0 Len: 5
> [ WARN] [1526389886.541567534]: Don't know ack: Port: 8 Channel: 0 Len: 5
> [ WARN] [1526389886.623797926]: Don't know ack: Port: 8 Channel: 0 Len: 5
> [ WARN] [1526389886.635239798]: Don't know ack: Port: 8 Channel: 0 Len: 5
> [ WARN] [1526389886.646636864]: Don't know ack: Port: 8 Channel: 0 Len: 5
> [ WARN] [1526389886.658150420]: Don't know ack: Port: 8 Channel: 0 Len: 5
> [ WARN] [1526389886.669962605]: Don't know ack: Port: 8 Channel: 0 Len: 5
> [ WARN] [1526389886.740040288]: Don't know ack: Port: 8 Channel: 0 Len: 5
> [ WARN] [1526389886.751419491]: Don't know ack: Port: 8 Channel: 0 Len: 5
> [ WARN] [1526389886.763022609]: Don't know ack: Port: 8 Channel: 0 Len: 5
> [ WARN] [1526389886.775148730]: Don't know ack: Port: 8 Channel: 0 Len: 5
> [ WARN] [1526389886.786696105]: Don't know ack: Port: 8 Channel: 0 Len: 5
> [ WARN] [1526389886.798108752]: Don't know ack: Port: 8 Channel: 0 Len: 5
> [ WARN] [1526389886.810092996]: Don't know ack: Port: 8 Channel: 0 Len: 5
> [ WARN] [1526389886.833107051]: Don't know ack: Port: 8 Channel: 0 Len: 5
> [ WARN] [1526389886.844931844]: Don't know ack: Port: 8 Channel: 0 Len: 5
> [ WARN] [1526389887.498618867]: [cf20] Link Quality low (0.620000)
> [ INFO] [1526389888.373566946]: [cf20] GoTo
> [ INFO] [1526389890.890836045]: Start trajectory!
> [ INFO] [1526389900.238371515]: Land!
>

whoenig, 2018-05-15:
I am sorry about the "Don't know ack" warnings - they are harmless, and I created an issue in crazyflie_cpp to address that issue (see https://github.com/whoenig/crazyflie_cpp/issues/2). 

Did you disable logging? The link quality is actually not analyzed in swarm operation with logging disabled. If logging or any other direct drone<->PC communication is enabled (instead of broadcasts), the broadcasts and other operations with share the radio and you increase the risk of not getting enough broadcast messages about the position. If you have the LED-ring installed, it will show you the link quality by default (green: good; red: bad).

pPatrickK, 2018-05-16:
Observation from today:

I've increased the timeout timeout time as suggested in [https://github.com/USC-ACTLab/crazyswarm/issues/63](url)

I flew just one drone with logging disabled,
randomly the drone just shut off and fell to the ground. This always happened when a new goto was send.

Any ideas?

We install today our realtime spectrum analyzer to monitor the traffic.

pPatrickK, 2018-05-16:
Picture 1 shows the bar of channel 82 while the server is running.
Picture 2 shows channel 82 while the server and the flie is off. The channel seems pretty clean. 

We tested also the following:
-Pull out the dongle while the cf hovers -> cf drifts away
-Kill the server over the task manager -> cf drifts away
-If we shut down the server by ctrl+c -> cf shuts engines off as with our problem.

So at some point the cf must decide to cut of the engines for some reason. What would be the best way to debug this? 

[image]
[image]

whoenig, 2018-05-16:
That's a neat setup - sounds like we can rule out communication issues. A few ideas:
1.) After the crash, is the CF is a failure mode (usually indicated by the red LED being consistently on)? This would indicate that there is an issue with the on-board computation.
2.) Did you verify that your goTo commands are feasible? In other words, what is the distance and desired time you send? I have seen such crashes if a) the distance is too long for the desired time, or b) the desired time is very short (<0.5s). In the first case the resulting trajectory is simply infeasible. In the second case I assume we run into some numerical issue (we only use single-precision computation on-board).
3.) If you can reproduce with a single drone, you can also re-enable logging and look at the setpoint and state information to see if there is a sudden jump (which would indicate a planning issue).

The ultimate/best way to debug the issue is to attach the uSD-card deck and log setpoint, state, and vicon information in every control loop iteration and analyze the data after the crash. We had that functionality in the old firmware, but not in the updated one (we are hoping to use the existing uSD-card infrastructure of the official firmware.) This basically helps you to also figure out which component (state estimation or control) failed first.

pPatrickK, 2018-05-17:
After a crash the red LED is constantly on, yes.
We give the flie enough time for the movement. For example two seconds for one meter. With the last version, those values worked well. We ordered some uSD card decks to check your last point.
If we start logging over the plotter (state and estimated state for example), the drone crashes after a short time.

We attached some images of the plots. Red curves are target values and blue are estimated values. First image is x- second is y- and third is z-value.

[image]

[image]

[image]

We also made a [video](https://youtu.be/Rl2rE8j0cEY) of the flight.

whoenig, 2018-05-17:
The plots and scripts look good.

In your video, I noticed that the red light turns fully on after the first waypoint. The CF then continues one or two more waypoints, before falling down. While (or after) falling down, the CF seems to reboot (and according to your information the red light will be on.)

That the red light is on during the flight while still operating means low battery voltage. (If you have the LED ring installed, the ring will start blinking to indicate low battery.)
On the other hand, the falling and red light after the reboot likely means that there was a firmware issue. The CF has a watchdog that resets the drone if necessary, for example if there was a hardfault (e.g. division by 0), memory access fault, or some computation simply took too long.

Things to try out:
1. Make sure that your batteries are sufficiently charged. You can check using chooser.py before the flight.
2. After the drone crashed, try to connect with the CF client and look at the debug console (or, use the console tool in crazyflie_tools). It might provide you with some details on why the drone crashed.
3. If there is not much detail in 2, you can also try a firmware that is compiled with DEBUG=1. This will have two effects: a) the watchdog is disabled, so if it is just a matter of computation time there won't be a crash, and b) you get debug symbols. If you happen do have the debug adapter kit and a JTAG-debugger, you can attach with gdb after the crash and try to get a stacktrace using the `bt` command.

I might have some time tomorrow to double check the operation on our side with your script. I assume you use the latest crazyswarm master branch, together with the pre-compiled firmware we provide?

pPatrickK, 2018-05-18:
Yes, we use the latest branch and firmware.
We use the motion capture tracking here with Nexus2.6.

We've made the config.mk with 
```
PLATFORM=CF2
DEBUG=1
CLOAD=1
```
This should bring us in debug or?
We currently getting the right connector.

In this [video](https://youtu.be/L6jrqH5coEY) you see that the cf reboots after it falls(which shouldn't be since we are in debug?). After the reboot the red light is flashing as in the [failed self test](https://www.bitcraze.io/getting-started-with-the-crazyflie-2-0/#infostep18).

Best regards.

whoenig, 2018-05-20:
You are correct that the provided config.mk should create a DEBUG build where the CF should never reboot, unless power-cycled. Did you use `make clean` followed by `make`? The Makefile will not detect changes in the config.mk file.

I have not seen such reboots as in your video for a debug build, unless there was some mechanical problem (lose battery wires).

One problem can be related to flashing. If you use the official bitcraze tool, it doesn't use verification. I have seen on multiple instances that flashed firmwares were not identical to the firmware on the PC, resulting in all kinds of weird bugs. You should use the `flash` tool part of `crazyflie_tools` instead, or the JTAG debug adapter (with `make flash`) - both use verification by default. Note, that `flash` was not working in the last release, but I just pushed a fix for that yesterday.

I didn't get around testing your example script in our facility yet, but it's still high up on my ToDo list for next week.

whoenig, 2018-05-25:
I just tested it on my side and I can reproduce the crashes with the crazyswarm firmware. However, as a workaround, it seems to be fine to use the official firmware:
1. `rosrun crazyflie_tools flash --target stm32 --filename ../prebuilt/cf2_bitcraze.bin`
2. Change [this](https://github.com/USC-ACTLab/crazyswarm/blob/master/ros_ws/src/crazyswarm/launch/hover_swarm.launch#L10) line to "bitcraze".

I will update the thread once I found (& fixed) the issue in the crazyswarm firmware. The crazyswarm firmware requires more stack because it has a more complex EKF, so my best guess is that there are some memory corruptions (since the goTo planning code is otherwise identical between the firmwares).

Note that the bitcraze firmware has the disadvantage that it only fuses position information, so you can't hover at a point forever (moving sideways allows the EKF to correct yaw drift).

pPatrickK, 2018-06-04:
We are currently working with the commit from March 22, 2018 (50e81754da9264457ca5b0a7b8ef0710fbbae759). We can observe that with more drones the they get less stable up to a point where the first crashes start. With 20 drones we usually have 2-3 crashes. Right now we are assuming that the error stems from a firmware error. 
Observations: 
 
1. If I pause tracker for half a second and press play afterwards -> drone drifts off a bit and corrects then to the setpoint with high dynamic. If I wait longer the drone exceeds its dynamic limits and crashes.

2. During takeoff one or more of the drones performs a backflip, although all drones are having correct poses in tracker. These backflips sometimes also occur during normal flight. 

3. Another issue occurs if we use a small swarm in contrast to a single drone. One drone can fly highly dynamical and without issues, while e.g. with 10 drones all loose their stable pose during flight (fluctuations in rotation and position) and a few drones crash with a similar behavior as in 1. .

We are assuming two main error sources. First package loss/excessive channel latency during transmission. Second, bad estimations on the drone itself.

Do you have further suggestions on the next steps of the debugging process. We have 2 SD-Card Decks to attach to the drones and a ST-Link Debugger to our disposal. Although we are not sure, whether we are able to connect the ST-Link Debugger to a drone in motion, without influencing its flight behavior. In which directions should we investigate first with the debugging process?

Also is frequency hopping available for the NRF chip ? 
Thanks in advance!

whoenig, 2018-06-05:
We used this particular commit to fly 13 drones without any issues (trajectory execution).

* Are you operating in waypoint following mode or trajectory execution? Unfortunately, we didn't test waypoint following extensively lately.
* You mention that it's correct in tracker. Are you using Tracker for tracking or libmotioncapture? I have seen such backflips only if using Vicon Tracker when axis-flips occurred. If you use Tracker, try to enable a motion filtering model.
* If I remember correctly, you have LED rings and a spectrum analyzer. If the analyzer doesn't show issues and the LED rings are green it should be fine. There is no HW support for frequency hoping in the NRF chip. It might be possible to implement that in SW though.

Debugging:
* For uSD-cards I recommend going to the latest official firmware (and a newer crazyswarm version). We just implemented proper support of logging of any onboard variables - see the wiki here: https://wiki.bitcraze.io/projects:crazyflie2:expansionboards:microsd. If you use the provided config.txt, you can look at the state estimate and see if there are any weird flips. You could also add variables for the received external state estimate to check if there are any latency issues.
* The debugger can't be used while flying. It is mostly useful for three purposes: a) flash firmware quickly (with verification), b) develop and step through firmware, c) figure out why a crash occurred. If you believe the firmware crashed (or rebooted) after you observed odd behavior, you can attach the debugger after the crash occurred and hopefully get an idea where it happened.

whoenig, 2018-06-05:
Just as update for the waypoint following crashes (given your python script): I identified and fixed the issue in the crazyswarm firmware. At the same time I merged the latest official firmware master into our code base, so the crazyswarm firmware now also has proper uSD card logging support.

pPatrickK, 2018-06-12:
Hi,
thanks, that worked.

Still we face a few crashes during a flight of the swarm. We logged some variables with the sd deck and think that we have a combination of issues which could be the reason for the crashes.

In our hall (build out of sheet metal) we have strong interference which results in a lot of spots where the packet rate is 0. 
Our drones differ in performance, picture 3 & 4. In picture 3 (drone that crashed) one can see that strong deviatito ons between the engines. When the dynamic gets to high the engine will run into the threshold of ~65000. (Picture 1&3 = drone1; 2&4 = drone2)  

Questions
1. Is the threshold ~65000 (variable: motor.m1-m4) really the physical maximum or is it limited by software? (is it possible the get more power out of the engines without changing the hardware?)
2. Have you experienced deviations between the hardware of the crazyflies (battery, engines...) ?
3. How should we interpret the CTRL yaw (first plot in pic 1&2,variable: ctrltarget.yaw)
4. Have you guys tweaked the drones in a way that the engine signals are similar to each other?
5. How could you place the markers underneath the rotor? (We had once placed a 1.5 mm rod underneath a rotor which already destroyed the aerodynamics, the drone could not take off)
6. Have you guys ever tried to run the swarm with 250K instead of 2Mbps, any experiences? With that we can achieve a more stable packet rate (varies strongly with location). Best is 0.92.

Any more ideas ?

[image]
[image]
[image]
[image]

whoenig, 2018-06-13:
I don't recall looking at how close our thrust output to saturation is. I can check next time I run experiments. Which trajectory are you flying at which speed? What is the weight of your drone?
There are a few tricks to deal with saturation that we haven't implemented yet, but for "normal" speeds that shouldn't be a problem.

1. The thrust is unit-less and essentially the PWM-output mapped between 0 and 65536 (see https://github.com/bitcraze/crazyflie-firmware/blob/7ae809eeb5b48b5d84649c713accd2d7bf3df338/src/drivers/src/motors.c#L201-L232). The actual thrust depends on the battery voltage, so newer/more powerful batteries can produce more thrust. Note that bigger batteries can cover the antenna, which will make your reception worse, so stick to the ones that are roughly the original size. (We use [these](https://hobbyking.com/en_us/turnigy-nano-tech-260mah-1s-35-70c-lipo-pack-qr-ladybird-genius-cp-mini-cp-5pcs.html?wrh_pdp=7)
2. Definitely, especially after a few crashes. We frequently replace propellers in that case (hard to visually inspect actually), rarely motor mounts (only if visibly damaged), and maybe only once a motor. The I-parts of our controller take care of differences fairly well.
3. This is the yaw setpoint in degrees. I believe your jumps might come from "mode-switches" (e.g., if you take-off, land, or go to a new waypoint). Whenever the on-board planner is called, we plan from the current state to the new goal. You can a "jump" if the current state was not identical to the setpoint. We might change that behavior to plan from the current setpoint instead, to avoid any jumps in the setpoint.
4. No, we just let the I-part of our controller take care of such differences.
5. A while ago I did play around with 3D printed frames etc. for markers. Basically anything I tried had a significant effect on aerodynamics, even when not underneath. One thing you could try is to use active IR LEDs instead of big markers (similar to the VICON Active Wand). This should be easy with the expansion port and should have a much lower effect on aerodynamics. I also just suggested this here: https://www.bitcraze.io/2018/06/mocap-deck/.
6. In our space 250K performed worse than the 2M mode. The data rate will be a bit lower, but I think that shouldn't be a problem for the Crazyswarm.

For the cases where a drone crashes, how does the vicon.dt look like? In your plots it is hard to tell since there is a large initial spike. If you look at vicon.dt right around the crash, this should give you a good indication if there was communication loss first (causing wrong state estimation, causing loss of control), or a control issue (caused, for example, by output saturation).

srnetsr, 2018-08-21:
Hi, is it resolved?
In my case before, when I used goTo function, then sometimes cf2 was just dropped to the ground, and rebooted by itself.
I am not sure rebooting before falling or falling before rebooting happened.
Do I need to upgrade firmware to resolve this issue?

Thank you.

jpreiss, 2020-06-13:
@pPatrickK were you able to resolve this?

jpreiss, 2021-01-30:
closing due to inactivity.

closed 2021-01-30 by jpreiss