USC-ACTLab/crazyswarm #65: Build and launch errors on new machines

pPatrickK, 2018-05-18:
We tried to setup some more machines because we were curious if we only get our problems on our used machine. On one machine, we made a clean install of ubuntu 16.04 and started to install the dependencies and tried to build the repository. We get some strange error while building:

> CMake Warning at /opt/ros/kinetic/share/catkin/cmake/catkinConfig.cmake:76 (find_package):
>   Could not find a package configuration file provided by "crazyflie_cpp"
>   with any of the following names:
> 
>     crazyflie_cppConfig.cmake
>     crazyflie_cpp-config.cmake
> 
>   Add the installation prefix of "crazyflie_cpp" to CMAKE_PREFIX_PATH or set
>   "crazyflie_cpp_DIR" to a directory containing one of the above files.  If
>   "crazyflie_cpp" provides a separate development package or SDK, be sure it
>   has been installed.
> Call Stack (most recent call first):
>   crazyswarm/CMakeLists.txt:7 (find_package)
> 
> 
> -- Could not find the required component 'crazyflie_cpp'. The following CMake error indicates that you either need to install the package with the same name or change your environment so that it can be found.
> CMake Error at /opt/ros/kinetic/share/catkin/cmake/catkinConfig.cmake:83 (find_package):
>   Could not find a package configuration file provided by "crazyflie_cpp"
>   with any of the following names:
> 
>     crazyflie_cppConfig.cmake
>     crazyflie_cpp-config.cmake
> 
>   Add the installation prefix of "crazyflie_cpp" to CMAKE_PREFIX_PATH or set
>   "crazyflie_cpp_DIR" to a directory containing one of the above files.  If
>   "crazyflie_cpp" provides a separate development package or SDK, be sure it
>   has been installed.
> Call Stack (most recent call first):
>   crazyswarm/CMakeLists.txt:7 (find_package)
> 
> 
> -- Configuring incomplete, errors occurred!
> See also "/home/cardinal/Dokumente/crazyswarm/ros_ws/build/CMakeFiles/CMakeOutput.log".
> See also "/home/cardinal/Dokumente/crazyswarm/ros_ws/build/CMakeFiles/CMakeError.log".
> Invoking "cmake" failed

We used another machine where the build ran correctly but we couldn't launch the crazyflie server. It failed waiting for emergency service.

It is strange that nobody else seems to get those issues we face with crazyswarm.

whoenig, 2018-05-20:
For the first issue, are you using `./build.sh`? It looks like the crazyflie_cpp submodule was not checked-out correctly, as done in the following lines: https://github.com/USC-ACTLab/crazyswarm/blob/master/build.sh#L7-L10.

For the second issue, you need to install `sudo apt install ros-kinetic-joy`.

sarvjeetdh, 2018-05-31:
I used this method and getting the same error 

$ mkdir -p ~/crazyflie_ws/src
$ cd ~/crazyflie_ws/src
$ catkin_init_workspace

$ git clone https://github.com/whoenig/crazyflie_ros.git
$ cd ~/crazyflie_ws
$ catkin_make

I have just started with crazyflie and can't locate that ./builld.sh 

Please help

whoenig, 2018-05-31:
@sarvjeetdh: You seem to use crazyflie_ros only and not the crazyswarm. Please open an issue [here](https://github.com/whoenig/crazyflie_ros/issues) instead.

sarvjeetdh, 2018-06-01:
Ok... Thanks

pPatrickK, 2018-06-21:
Hi, 

you were right with the submodule.

Thanks

closed 2018-06-21 by pPatrickK