USC-ACTLab/crazyswarm #68: Server process dies immediately

iron-beagle, 2018-06-15:
SUMMARY
========

PARAMETERS
 * /crazyflieTypes/default/batteryVoltageWarning: 3.8
 * /crazyflieTypes/default/batteryVoltateCritical: 3.7
 * /crazyflieTypes/default/bigQuad: False
 * /crazyflieTypes/default/dynamicsConfiguration: 0
 * /crazyflieTypes/default/firmwareParams/ctrlMel/i_range_m_z: 1500
 * /crazyflieTypes/default/firmwareParams/ctrlMel/i_range_xy: 2.0
 * /crazyflieTypes/default/firmwareParams/ctrlMel/i_range_z: 0.4
 * /crazyflieTypes/default/firmwareParams/ctrlMel/kR_xy: 70000
 * /crazyflieTypes/default/firmwareParams/ctrlMel/kR_z: 60000
 * /crazyflieTypes/default/firmwareParams/ctrlMel/kd_omega_rp: 200
 * /crazyflieTypes/default/firmwareParams/ctrlMel/kd_xy: 0.2
 * /crazyflieTypes/default/firmwareParams/ctrlMel/kd_z: 0.4
 * /crazyflieTypes/default/firmwareParams/ctrlMel/ki_m_z: 500
 * /crazyflieTypes/default/firmwareParams/ctrlMel/ki_xy: 0.05
 * /crazyflieTypes/default/firmwareParams/ctrlMel/ki_z: 0.05
 * /crazyflieTypes/default/firmwareParams/ctrlMel/kp_xy: 0.4
 * /crazyflieTypes/default/firmwareParams/ctrlMel/kp_z: 1.25
 * /crazyflieTypes/default/firmwareParams/ctrlMel/kw_xy: 20000
 * /crazyflieTypes/default/firmwareParams/ctrlMel/kw_z: 12000
 * /crazyflieTypes/default/firmwareParams/ctrlMel/mass: 0.032
 * /crazyflieTypes/default/firmwareParams/ctrlMel/massThrust: 132000
 * /crazyflieTypes/default/markerConfiguration: 0
 * /crazyflies: [{'initialPositio...
 * /crazyswarm_server/broadcast_address: FFE7E7E7E7
 * /crazyswarm_server/enable_logging: False
 * /crazyswarm_server/enable_parameters: True
 * /crazyswarm_server/firmware: crazyswarm
 * /crazyswarm_server/firmwareParams/commander/enHighLevel: 1
 * /crazyswarm_server/firmwareParams/ring/effect: 14
 * /crazyswarm_server/firmwareParams/ring/headlightEnable: 0
 * /crazyswarm_server/firmwareParams/ring/solidBlue: 255
 * /crazyswarm_server/firmwareParams/ring/solidGreen: 0
 * /crazyswarm_server/firmwareParams/ring/solidRed: 0
 * /crazyswarm_server/force_no_cache: False
 * /crazyswarm_server/genericLogTopicFrequencies: [10]
 * /crazyswarm_server/genericLogTopic_log1_Variables: ['ctrlStat.edist']
 * /crazyswarm_server/genericLogTopics: ['log1']
 * /crazyswarm_server/motion_capture_type: vicon
 * /crazyswarm_server/object_tracking_type: libobjecttracker
 * /crazyswarm_server/print_latency: False
 * /crazyswarm_server/save_point_clouds: ~/pointCloud.ot
 * /crazyswarm_server/vicon_host_name: 192.168.15.5:801
 * /crazyswarm_server/world_frame: /world
 * /crazyswarm_server/write_csvs: False
 * /crazyswarm_teleop/csv_file: /home/lab/workspa...
 * /crazyswarm_teleop/timescale: 0.8
 * /dynamicsConfigurations/0/maxFitnessScore: 0.001
 * /dynamicsConfigurations/0/maxPitch: 1.4
 * /dynamicsConfigurations/0/maxPitchRate: 20.0
 * /dynamicsConfigurations/0/maxRoll: 1.4
 * /dynamicsConfigurations/0/maxRollRate: 20.0
 * /dynamicsConfigurations/0/maxXVelocity: 2.0
 * /dynamicsConfigurations/0/maxYVelocity: 2.0
 * /dynamicsConfigurations/0/maxYawRate: 10.0
 * /dynamicsConfigurations/0/maxZVelocity: 3.0
 * /joy/dev: /dev/input/js0
 * /markerConfigurations/0/numPoints: 4
 * /markerConfigurations/0/offset: [0.0, -0.01, -0.04]
 * /markerConfigurations/0/points/0: [0.00122217, 0.00...
 * /markerConfigurations/0/points/1: [0.0654261, 0.001...
 * /markerConfigurations/0/points/2: [-0.00177642, -0....
 * /markerConfigurations/0/points/3: [0.00190696, 0.06...
 * /numDynamicsConfigurations: 1
 * /numMarkerConfigurations: 1
 * /rosdistro: kinetic
 * /rosversion: 1.12.13

NODES
  /
    crazyswarm_server (crazyswarm/crazyswarm_server)
    crazyswarm_teleop (crazyswarm/crazyswarm_teleop)
    joy (joy/joy_node)
    rviz (rviz/rviz)

auto-starting new master
process[master]: started with pid [7104]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 66f92592-70e2-11e8-bd0f-80fa5b0762fb
process[rosout-1]: started with pid [7117]
started core service [/rosout]
process[crazyswarm_server-2]: started with pid [7134]
process[joy-3]: started with pid [7135]
process[crazyswarm_teleop-4]: started with pid [7136]
process[rviz-5]: started with pid [7142]
[ INFO] [1529097833.077857439]: Wait for services...
[ INFO] [1529097833.078298745]: waitForService: Service [/emergency] has not been advertised, waiting...
[ INFO] [1529097833.099317256]: waitForService: Service [/emergency] is now available.
[ INFO] [1529097833.100684095]: Manager ready.
[ INFO] [1529097833.115739764]: Opened joystick: /dev/input/js0. deadzone_: 0.050000.
[crazyswarm_server-2] process has died [pid 7134, exit code -11, cmd /home/lab/workspace/crazyswarm/ros_ws/devel/lib/crazyswarm/crazyswarm_server __name:=crazyswarm_server __log:=/home/lab/.ros/log/66f92592-70e2-11e8-bd0f-80fa5b0762fb/crazyswarm_server-2.log].
log file: /home/lab/.ros/log/66f92592-70e2-11e8-bd0f-80fa5b0762fb/crazyswarm_server-2*.log
[rviz-5] process has finished cleanly
log file: /home/lab/.ros/log/66f92592-70e2-11e8-bd0f-80fa5b0762fb/rviz-5*.log

exit code 11 relates to a segment fault. We cannot print statements after  /*enableObjects*/useMotionCaptureObjectTracking || !interactiveObject.empty(), in file crazyswarm_server.cpp 

We know the radio is talking to the crazyflie because the chooser works. We verified Vicon IP address. Vicon set properly but we think there may be a communication error. Maybe something is wrong with the setup. Running last years version fine on a separate PC in parallel to having this error on another.

iron-beagle, 2018-06-19:
Ran under your conditions and it identified the issue as the _libViconDataStreamSDK_CPP.so_. 

When first spinning up your repo I had trouble with the structure and played with the new Vicon 2.0 files. Must have left one or a few in there. Replaced with 1.7.1, rebuilt and ran like a champ. 

May be worth noting that both the **libmotioncapture** and **libobjecttracker** need to be cloned after crazyswarm. Otherwise the *build.sh* will not run correctly. Doing a recursive clone did not populate these folders in the right way and it took the better part of a day to find the solution. 

Thanks for the solid work on this project and helping me lvl up my Ros skills! 
Closing the issue.

closed 2018-06-19 by iron-beagle

whoenig, 2018-06-19:
Thanks for the update! `build.sh` should clone/update the required submodules.

whoenig, 2018-07-30:
This issue was closed and your error seems to be unrelated to this issue, because in your case the crazyswarm_server does not terminate. Please open a new issue.