USC-ACTLab/crazyswarm #69: Updating the Crazyflie firmware

Pop99, 2018-06-18:
Hi Wolfgang,
In Crazyflie preperation notes, step 4, I need to update the firmware.

Using option 2, when I run the command below from the Ubuntu 16.04 crazyswarm dir 
rosrun crazyflie_tools flash --target nrf51 --filename prebuilt/cf2_nrf.bin

I get [rospack] Error: package 'crazyflie_tools' not found. 

Looking at closed posts it looks like I can run
 python3 -m cfloader flash prebuilt/cf2.bin stm32-fwcd prebuilt

unfortunately I get 
/usr/bin/python3: No module named cfloader

Any thoughts
Thanks

whoenig, 2018-06-18:
In order to use crazyflie_tools, you need ROS to find it, typically by using the following command in the terminal: `source ros_ws/devel/setup.bash` in the crazyswarm folder.

Pop99, 2018-06-18:
Thank you for your quick response, i am getting this returned when running the commands

[image]

Crazyradio PA is in place and works fine with crazyflies

I've checked the issues list, both open and closed but haven't spotted a similar response

whoenig, 2018-06-18:
Do you have the permission for your user set for the Crazyradio?
What does `lsusb` show?

Pop99, 2018-06-18:
Please see
[image]

Pop99, 2018-06-18:
Not sure how to set or check permissions

whoenig, 2018-06-18:
That means even your Linux can't find any USB device, otherwise you should have a line like this:
```
Bus 002 Device 009: ID 1915:7777 Nordic Semiconductor ASA
```

If it's a VM, make sure you pass through the Crazyradio. Also, check your system log after removing/re-inserting the Crazyradio (using `dmesg`).

Pop99, 2018-06-18:
I'm not using a VM
Just Ubuntu 16,04 and Ros Kinetic installed

Just tried dmesg

po78@DESKTOP-4VH4RF8:~/crazyswarm$ dmesg
dmesg: read kernel buffer failed: Function not implemented

whoenig, 2018-06-18:
Seems like some security issue, see https://unix.stackexchange.com/questions/390184/dmesg-read-kernel-buffer-failed-permission-denied. You can try looking at the system log using `sudo`.

Pop99, 2018-06-18:
Sorry new to Ubunto, just trying to find the system log files

Pop99, 2018-06-19:
I first started following the process for 'Flying Multiple UAVs using Ros'. In this paper, there is a point on page 5 where we set PC Permissions which just happens to set USB attributes. It seems plausible that this could be the root of the problem?

whoenig, 2018-06-19:
If you didn't set the permissions, this is certainly a problem. This doesn't explain why `lsusb` doesn't show you any USB devices though. Try `sudo lsusb`, to make sure this is not related to another permission issue.

Pop99, 2018-06-19:
The permissions were set in the Flying multiple UAVs paper and are not referenced in the Crazyswarm instructions. I am trying various installs on different laptops.

Do we need to do the below for Crazyswarm

[image]

I've tried sudo lsusb with the following results

[image]

Thanks for your continuing help

whoenig, 2018-06-19:
Yes, you'll need to set the permission as described in that chapter (or following the Bitcraze documentation for that matter).
I still think that the issue is with your USB and not permission - `lsusb` should show you a bunch of devices, including mouse/keyboard (unless they are connected differently...)

Pop99, 2018-06-20:
One quick thought, do you run this on a Laptop? does anyone?

ataffanel, 2018-06-20:
@Pop99 Just to make sure: are you using the Windows Sybsytem for Linux ([WSL](https://docs.microsoft.com/en-us/windows/wsl/install-win10))? If so, as far as I know, there is no USB support yet so it is not possible to use a Crazyradio in WSL.

Pop99, 2018-06-20:
Hi,
I downloaded Ubuntu 16.04 from the Microsoft store please see pic below
[image]
I did have to select windows system for Linux as stated.
Guess I have to start again?

ataffanel, 2018-06-20:
As far as I could gather about the subject, USB does not work with WSL and so you will have to use Ubuntu either natively on your computer or in a virtual machine.

Pop99, 2018-06-20:
I want to make sure I get the right Ubuntu version, if possible could you point me in the right direction for the correct download please.

I want to fly four crazyflies at the same time, each going on a slightly different path. Could I use a script for this and run it from Pythons Idle environment like we can with other simpler scripts. if so, how do we import cached factory and import swarm as in one of the other Bitcraze scripts. I don't want to get off topic but rather regain some lost time
Thanks alot

whoenig, 2018-06-20:
Thanks @ataffanel for catching that. The crazyswarm does work in a VM, but you'll have trouble with some of the graphical tools and USB latency. The best way is to setup a machine that boots Ubuntu natively. You can download installers at https://www.ubuntu.com/download/alternative-downloads. Make sure to pick 16.04 and a version that reflects your architecture. There are many tutorials online on how to setup Ubuntu coming from Windows.

I don't understand your second question. The official Bitcraze SDK can handle multiple CFs, although you might need more Crazyradios compared to the Crazyswarm and execution is controlled on the PC as it does not have the High-level-mode implemented, yet. Another option is to use crazyflie_cpp, that some people have gotten to work on Windows (but this is not officially supported at the moment.)

Pop99, 2018-06-20:
Thanks Guys for the help. Do you know if I need to delete the existing Ubuntu installation and unset the WSL setting in Windows before I re-install Ubuntu and ROS Kinetic? Shame about the WSL version, would it be an idea to warn others about this on the Crazyswarm project docs?

Still new to Crazyflie development and I was wondering if I could achieve a quick autonomous 4 CF's result using python scripts and the flow deck. which would allow me to take my time with the full Crazyswarm option.

whoenig, 2018-06-21:
I don't know much about WSL or the Ubuntu offered in the Windows store.

It would be certainly good to update the docs to include the information about how to set the permissions as well as some hints on Ubuntu version - PRs are welcome.

For any question related to the official SDK, it is best to seek out for help in the official support forum.

closed 2018-12-07 by whoenig