USC-ACTLab/crazyswarm #70: rviz error: no transformation from [crazyflie/base_link] to [world]

klauswq, 2018-06-20:
Hi, I am integrating optitrack and cf2 with crazyflie_ros and mocap

In mocap configuration file, i set
rigid body:

    '3'：
        pose： optitrack/crazyflie_1/pose 
        child_frame_id: crazyflie
        parent_frame_id: world
 

when i launch hover_vicon.launch , the cf can take off, but it can not hover to target position, and i got error in rviz as follows:
/home/jingwen/Pictures/Screenshot from 2018-06-20 13-23-51.png

my tf tree is as follows:
/home/jingwen/Pictures/Screenshot from 2018-06-20 13-24-34.png

Can you please help me with my rviz error? Your help would be greatly appreciated.

whoenig, 2018-06-20:
This is about crazyflie_ros and has nothing to do with the Crazyswarm, correct?

Your tf tree looks good. Can you share the full launch file you are using? `hover_vicon.launch` does launch the `vicon_bridge`, so I guess you modified it somewhat?

klauswq, 2018-06-20:
Thanks for your reply ,I comment run vicon bridge, does this lead to my error?

<?xml version="1.0"?>
<launch>
  <arg name="uri" default="radio://0/80/2M/E7E7E7E7E7" />
  <arg name="frame" default="crazyflie" />
  
  <arg name="joy_dev" default="/dev/input/js0" />
  <arg name="x" default="0" />
  <arg name="y" default="0.5" />
  <arg name="z" default="0.5" />

  <include file="$(find crazyflie_driver)/launch/crazyflie_server.launch">
  </include>

  <group ns="crazyflie">
    <include file="$(find crazyflie_driver)/launch/crazyflie_add.launch">
      <arg name="uri" value="$(arg uri)" />
      <arg name="tf_prefix" value="crazyflie" />
      <arg name="enable_logging" value="False" />
    </include>

    <node name="joy" pkg="joy" type="joy_node" output="screen">
      <param name="dev" value="$(arg joy_dev)" />
    </node>

    <node name="joystick_controller" pkg="crazyflie_demo" type="controller.py" output="screen">
      <param name="use_crazyflie_controller" value="True" />
    </node>

    <include file="$(find crazyflie_controller)/launch/crazyflie2.launch">
      <arg name="frame" value="$(arg frame)" />
    </include>

    <node name="pose" pkg="crazyflie_demo" type="publish_pose.py" output="screen">
      <param name="name" value="goal" />
      <param name="rate" value="30" />
      <param name="x" value="$(arg x)" />
      <param name="y" value="$(arg y)" />
      <param name="z" value="$(arg z)" />
    </node>

    <node pkg="tf" type="static_transform_publisher" name="baselink_broadcaster" args="0 0 0 0 0 0 1 $(arg frame) /crazyflie/base_link 100" />
  </group>

  
  
  
  
  <param name="robot_description" command="$(find xacro)/xacro.py $(find crazyflie_description)/urdf/crazyflie.urdf.xacro" />
  <node name="rviz" pkg="rviz" type="rviz" args="-d $(find crazyflie_demo)/launch/crazyflie_pos.rviz" required="true" />

</launch>

klauswq, 2018-06-20:
I comment run vicon bridge, because i use optitrack, It this the reason why i run into this error?

[image]

whoenig, 2018-06-21:
No, you should comment out those lines. What did you specify as frame in line 4 (https://github.com/whoenig/crazyflie_ros/blob/master/crazyflie_demo/launch/hover_vicon.launch#L4)? It should be "crazyflie" in your case.

Your tf tree doesn't quite match the launch file. For example, the "baselink_broadcaster" should update at 100 Hz, but according to your tf free there is only an update every 10 Hz...

klauswq, 2018-06-21:
Thanks for your reply, i am new to ROS, can you give me a hint on where to change update frequency from 10Hz to 100Hz?

whoenig, 2018-06-21:
I was actually mistaken, the second parameter is not in Herz, but in milliseconds, see http://wiki.ros.org/tf#static_transform_publisher. So you tf tree looks exactly how it should to me.

Make sure that the coordinate system actually matches the ROS convention - I know that this has caused trouble in the past for optitrack users.

whoenig, 2018-12-07:
If there are more issues, please open a new issue in the crazyflie_ros repository. Closing.

closed 2018-12-07 by whoenig

abdullja, 2021-11-07:
How was this problem solved? I'm getting the same error in Rviz and my crazyflie is not taking off.

whoenig, 2021-11-07:
Please open a new discussion rather than commenting on a closed issue.