USC-ACTLab/crazyswarm #71: Crash on setpoint-step

pPatrickK, 2018-06-21:
Hi guys,

if we face packet loss, the drone drifts away form the aimed trajectory.
After getting a valid packet again, this results in a setpoint-step.
This will then, depending on the size of the step, cause a crash. 

We've observed that the pitch/roll angle (attitude) exceeds 30 degrees, which is in our opinion the cause for the crash.

We would like to implement attitude angle limit. Could you give us a hint, which parts of the code is responsible for controlling the flight angles?

jpreiss, 2018-06-21:
It is true that our controller is not stable beyond about 30 degrees pitch/roll. However, I suspect this is due to some implementation bug and not a fundamental limit of the controller. The controller is  nonlinear and theoretically capable of getting to at least 60 degrees roll/pitch without any problems. The roll/yaw singularity at 90 degrees is a problem, but there are ways to fix that too. I am planning to take a look at this within a few weeks.

For this issue, what's happening is the position part of the controller is asking for an unreasonably high acceleration to correct the drift. We can fix this by clamping the position error term (`r_error` in controller_mellinger.c). I won't be able to test that myself for at least 1 week but I suggest clamping at a magnitude of 0.2 meters to try. If behavior is still unstable, reduce further. This solution lets us still have high gain control when near the setpoint.

jpreiss, 2018-06-21:
Glad to hear it, please go ahead and submit a pull request once you have tested/tuned the clamping parameter. We should have this feature enabled always.