USC-ACTLab/crazyswarm #72: Motor shut down without any identifiable reason

pPatrickK, 2018-06-21:
Hello,

we observed yesterday a crash which was caused by the engines being shut off 
[image]
(This first two plots show the position and velocity of the drone, red is Z. The third plot shows the motor thrusts. We aligned the time axis. As you see the motor cut off is not caused by an invalid pose, instead the cut off happens without any obvious reason which causes the drone to fall down.)

Sadly we did not log the sitAw in this particular flight but have enabled sitAw logging afterwards. We have provoked several crashes and the motor engines shut down as soon as the drone is lying on the ground but we have never seen the 3 sitAw flags to report anything.

So now we are looking for the reasons why the engines shut down but we do not know where to look for it. Which parts of the code have the permission/cause to shut off the engines? And do you have any idea why the sitAw flags never report anything to us?

whoenig, 2018-06-21:
I assume that you use our custom firmware and not the official firmware. In that case, we added additional safety features because we have seen instances where quadrotors would crash and keep spinning motors even while upside down - this causes motor/prop damage for CFs, or is simply inherently dangerous if the bigQuad deck is used.

Our code has one additional condition:
  * The drone is upside down for a single sensor reading according to the accelerometer, see https://github.com/USC-ACTLab/crazyflie-firmware/blob/crazyswarm-next/src/modules/src/stabilizer.c#L164-L175. 

Perhaps you want to disable that condition as it takes data unfiltered - if sitAw works well enough for you in failure cases.

jpreiss, 2020-06-13:
Closing. This appears to be an issue with our old custom firmware. We now recommend the official firmware.

closed 2020-06-13 by jpreiss