USC-ACTLab/crazyswarm #73: Log variables kill firmware

pPatrickK, 2018-06-21:
Hi,

we tried to add some sitAw Log variables which were commented in the 
https://github.com/USC-ACTLab/crazyflie-firmware/blob/ca2882b874d8e4f57eea53f06e6864eb68a895cb/src/modules/interface/sitaw.h#L65-L67 .
It was still possible to build the firmware but after we've flashed the one blue and both red leds were solid.

After re flashing without these changes the drone was working like usual.
Do you have any hints what causes this behavior ?

best regards
Patrick

whoenig, 2018-06-21:
This is most likely caused by this issue in the original firmware: https://github.com/bitcraze/crazyflie-firmware/issues/313. If you only log to uSD-card, you can remove the check here: https://github.com/USC-ACTLab/crazyflie-firmware/blob/crazyswarm-next/src/modules/src/log.c#L194. Otherwise, I recommend getting rid of some logging variables that you don't need.

This is on my plate to get fixed, but will probably take another 2 weeks or so, since it's a bug that affects the official firmware and all clients.

whoenig, 2018-12-07:
This has been fixed.

closed 2018-12-07 by whoenig