USC-ACTLab/crazyswarm #74: Attitude control unstable beyond about 30 degrees roll/pitch

jpreiss, 2018-06-21:
the SE(3) controller from Mellinger/Li should be capable of trajectory tracking with around 60 degrees roll/pitch angle without problems. Our implementation only can get to around 30 degrees. This has been observed by @whoenig and me in a trajectory tracking scenario and by @pPatrickK in #71 in a position setpoint step scenario. I believe this is due to some implementation bug.

whoenig, 2021-08-17:
Can we close this @jpreiss ? I feel this doesn't belong to this repo anymore, since we don't maintain the firmware. Also, I think this is actually due to the low thrust-to-weight ratio of about 1.4, which is significantly lower than the quadrotors used in other experiments.

jpreiss, 2021-08-17:
Sure, but maybe we can open an issue in `crazyflie-firmware`? I am not convinced that our controller implementation is bug-free. For example, when I was working on converting the controller to SI units I believe I found that our feedforward thrust command is not properly scaled like the feedback terms.

You might be right about the low thrust causing problems for fast trajectory tracking, but I think it will help future Crazyflie users to know that the performance limits of the controller have not been thoroughly investigated.

whoenig, 2021-08-17:
Added in https://github.com/bitcraze/crazyflie-firmware/issues/832.

closed 2021-08-17 by whoenig