USC-ACTLab/crazyswarm #76: Can't open prebuilt firmware files

jonesbmj, 2018-06-26:
Hi,

For some reason the command "rosrun crazyflie_tools flash --target nrf51 --filename prebuilt/cf2_nrf.bin" results in "Couldn't open prebuilt/cf2_nrf.bin!"

Earlier we added PC permissions as stated in the "Flying Multiple UAVs with ROS" paper and made sure to source the work space, so not sure what the problem is.

Any help is appreciated!

whoenig, 2018-06-26:
Which folder do you run the command in? The path is relative to the current working dir, so you would need to run that in the "crazyswarm" root folder to work properly.

jonesbmj, 2018-06-26:
That worked, thank you! I am new to linux and crazyflies so sorry for the noob mistake.

closed 2018-06-26 by jonesbmj