USC-ACTLab/crazyswarm #79: Timeout error with chooser and hover_swarm.launch

andresbhoo, 2018-07-12:
Hi whoenig,

Whenever I launch 'hover_swarm.launch' or 'chooser.py' I get a timeout error. I am able to connect to the crazyflie with the python3 client and it shows up with the tools scan, yet i am still getting this time out error. 
My crazyflie is at Address: 0xE7E7E7E7E7 and radio://0/81/2M. my allCrazyflies.yaml is as follows:

```
crazyflies:
  - id: 1
    channel: 81
    initialPosition: [.5, .4, 1.2]
    type: default
#  - id: 40
#    channel: 100
#    initialPosition: [-1.0, -0.5, 0.0]
#    type: default
```

crazyflies.yaml reads:

```
crazyflies:
- channel: 81
  id: 1
  initialPosition: [0.5, 0.4, 1.2]
  type: default
```

and the broadcast address in hover_swarm.launch is "FFE7E7E7E7".

The end of my 'hover_swarm.launch' output reads:

```
NODES
  /
    crazyswarm_server (crazyswarm/crazyswarm_server)
    crazyswarm_teleop (crazyswarm/crazyswarm_teleop)
    joy (joy/joy_node)
    rviz (rviz/rviz)

auto-starting new master
process[master]: started with pid [21179]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 177ac51e-8601-11e8-9c30-000c29022557
process[rosout-1]: started with pid [21192]
started core service [/rosout]
process[crazyswarm_server-2]: started with pid [21209]
process[joy-3]: started with pid [21210]
process[crazyswarm_teleop-4]: started with pid [21211]
process[rviz-5]: started with pid [21219]
[ INFO] [1531419988.943156177]: Wait for services...
[ INFO] [1531419988.946610439]: waitForService: Service [/emergency] has not been advertised, waiting...
[ INFO] [1531419988.974372646]: waitForService: Service [/emergency] is now available.
[ INFO] [1531419988.979788667]: Manager ready.
[ INFO] [1531419988.989793280]: Opened joystick: /dev/input/js0. deadzone_: 0.050000.
ch: 1
terminate called after throwing an instance of 'std::runtime_error'
  what():  timeout
[crazyswarm_server-2] process has died [pid 21209, exit code -6, cmd /home/cf/crazyswarm/ros_ws/devel/lib/crazyswarm/crazyswarm_server __name:=crazyswarm_server __log:=/home/cf/.ros/log/177ac51e-8601-11e8-9c30-000c29022557/crazyswarm_server-2.log].
log file: /home/cf/.ros/log/177ac51e-8601-11e8-9c30-000c29022557/crazyswarm_server-2*.log

```
Any suggestions? And thanks for the help!

andresbhoo, 2018-07-12:
Also, forgot to mention, the 'chooser.py' GUI opens up just fine. It is only when I try to use any of the buttons on the bottom edge that I get the timeout error.

Thank you again for the help.

andresbhoo, 2018-07-12:
Sorry, one more possible piece to the puzzle. With the new firmware on the drone, the cfclient does not react to attitude changes. I am assuming thing is because the crazyflie does trajectory planning on board, so it does not need to broadcast its attitude measurement. However, I am not sure about this, so I thought I would double check. 

Thanks again!

whoenig, 2018-07-12:
For your first issue: If your id is 1, you need to change the CFs address to 0xE7E7E7E701 (using the cfclient; connect over uUSB).

Which firmware did you flash? How did you flash it? The latest version should work with the client as well, but older versions only work with the Crazyswarm and not the client.

andresbhoo, 2018-07-12:
Oh, ok, I see. Changing the address to 01 allowed the crazyflie to communicate with 'chooser.py'.

Now that I can communicate via the "chooser" I find that I am only able to use "sysOFF" after I run "battery" and "version" first. The same sequence is required to run "reboot." However, more often than not, "reboot" gives me a timeout error. The reboot output is as follows:
```
1
timeout
```
Also, I got the 'hover_swarm.launch' to launch correctly one time while I had the "chooser" open in another consul! However, when it ran it kept giving me a latency warning and I have been unable to get it to launch successfully since. I am getting the exact same timeout error. Any suggestions?

Also I flashed the firmware using Option 2 on the documentation [website](http://crazyswarm.readthedocs.io/en/latest/usage.html).

Thank you so much for the help and the quick response!

whoenig, 2018-07-13:
A few comments, since I am not sure which exact sequence of operations work and which sequence of operations doesn't:
* "battery" should work in any state (unless you manually turn the CF off)
* "version" only works if the CF is fully on (i.e., not if you previously selected sysoff)
* before running the launch file, you should use "reboot" in chooser (i.e., you will get a timeout if you were in sysoff mode)
* chooser.py can run in parallel to the launch file, however you won't be able to use the buttons while the launch file is active

whoenig, 2018-07-14:
Note that the launch file attempts to call a sysOff at the end (if you use Ctrl+C). In contrary, it does not enforce a reboot, so you'll need to do that manually.

I have seen this kind of behavior before in two cases:
a) If you use the official client in between. The client enables a special mode called safelink that is different from the mode the latest crazyswarm version uses.
b) If you use the latest crazyflie_cpp (i.e., you updated manually compared to the version that is in the crazyswarm), but not the latest nrf-firmware. The latest crazyflie_cpp implements safelink, and the latest nrf-firmware fixes a bug in the firmware related to reboot, battery commands etc. that occurs when safelink is enabled.

If none of the above cases are true for you, I still suggest trying upgrading crazyflie_cpp and nrf firmware. Also, try different USB ports - I have seen one machine where one of the ports causes significant packet drops of the radio.

andresbhoo, 2018-07-16:
Ok, I believe I have now updated the crazyflie_cpp manually and upgraded the firmware. Unfortunately, nothing has changed. I got the new _cpp from your [crazyflie_cpp](https://github.com/whoenig/crazyflie_cpp) git hub and I used the new cf2_nrf.bin that was created when I used the "make" command after downloading what I thought was the upgraded nrf-firmware from your [crazyflie2-nrf-firmware](https://github.com/bitcraze/crazyflie2-nrf-firmware) git hub. Are these, indeed, the updated packages?

whoenig, 2018-07-16:
That seems right. Did you run catkin_make after upgrading crazyflie_cpp?

andresbhoo, 2018-07-18:
Yes, unfortunately. I used build.sh on the whole work space, which I believe runs catkin_make.

I am running on a virtual machine and am beginning to think this is the cause of the problem.

LZMHIT, 2018-10-18:
Dear Author, I have a problem when I launch hover_swarm.launch with single crazyflie2.0, and This is the screenshot, could you please tell me what is wrong with it?
[image]

whoenig, 2018-10-18:
This means that you are likely not using the Crazyswarm firmware. You can either change your firmware, or set https://github.com/USC-ACTLab/crazyswarm/blob/master/ros_ws/src/crazyswarm/launch/hover_swarm.launch#L10 to "bitcraze" and https://github.com/USC-ACTLab/crazyswarm/blob/master/ros_ws/src/crazyswarm/launch/hover_swarm.launch#L25 to "2". Also, I suggest disabling logging (which is the default, see https://github.com/USC-ACTLab/crazyswarm/blob/master/ros_ws/src/crazyswarm/launch/hover_swarm.launch#L53).

jpreiss, 2020-06-13:
@andresbhoo did you switch from a VM to hardware? Any updates to share?

andresbhoo, 2020-06-14:
Hey, I am sorry I did not leave an update since a few summers ago. It has
been a few years since I've worked with the crazyswarm workspace, but if
memory serves, I continued using the VM. My problem was that I was setting
up my drone facing the positive y direction or the positive x direction (I
don't recall which) instead of facing it in the correct coordinate
direction. After I changed this, everything worked ... a very silly and
time consuming mistake, but it's made for a good story the last few years.

Other than this, there were a few bugs, but I believe they had to do with
the software selecting which drones were active and which were not, but it
has been too long for me to remember any of the specifics. The workspace
functioned sufficiently well to get the drones up and working together.

jpreiss, 2020-06-14:
Thank you for taking the time to update us!

I'm not sure why the initial yaw angle would cause the communication-related issues you saw. However, it can definitely cause issues with flight stability. The initialization routine in `libobjecttracker` should be able to handle any initial yaw angle, but the takeoff always commands a desired yaw of zero, which can result in an unstable takeoff. Bitcraze recently added support for an arbitrary target yaw in the takeoff radio command. This will be added to Crazyswarm soon.

closed 2020-06-14 by jpreiss

andresbhoo, 2020-06-18:
Certainly!

Very exciting! Actually, I believe my problem with the angle was with the
Vicon software, not with the crazfly (or crazyswarm - I cannot remember
which is correct) workspace. I believe I built my object in Vicon facing
the wrong way, so all of the commands Vicon sent were off by 90°.