USC-ACTLab/crazyswarm #83: Firmware Controller

andresbhoo, 2018-07-20:
Hi Whoenig,

I am trying to modify pieces of the on-board controller, but thus far I am unable to find the controller within the various firmware files on the crazyswarm github.

Sorry for the trouble. It feels like I am probably missing something in plane sight, but thus far I have had no luck.

Thanks for the help!

andresbhoo, 2018-07-20:
Fantastic! Thank you.

jpreiss, 2018-07-20:
You're welcome, please share with us if you make improvements to the controller!

closed 2018-07-20 by jpreiss

emmasmith100, 2019-02-15:
Hi there!
I have been going through the mellinger paper and the code controller_mellinger.c in order to understand its implementation.
I do have many questions below and I was wondering if someone knows the answers? THANKS A LOT FOR YOUR HELP!
Note: my level of C coding is very basic and since there are not many comments in the code it gets difficult for newcomers. Especially the used attitude representations (and the chosen conventions) are not always clear to me.

**Question1:** 
`#include "position_controller.h"`: why do you include this? I don’t see where this is used?

**Question2:**
I understand controller gains are static since they will never change during flight. Under the comments 
`// Helper variables` and `// Logging variables` you also defined e.g. `i_error_x, prev_omega_roll, static struct vec z_axis_desired, roll_desired` as static variables. But these are variables must be changing over time? So it seems I don’t fully understanding the concept static.
Related to this, why is e.g. `prev_omega_roll` not assinged to 0? 

**Question3:** 
`i_range_xy`: what's the procedure to choose the range to saturate/clamp the integral error? Is this clamping strictly necessary or should it also work without clamping?

**Question4:**
I don’t understand the use of the functions `controllerMellingerInit()` and `controllerMellingerTest()`, since there are not called in the code.

**Question5:**
Concerning the function 
```
void controllerMellingerNew(control_t *control, setpoint_t *setpoint,
                                         const sensorData_t *sensors,
                                         const state_t *state,
                                         const uint32_t tick)
```
I was able to find more information in `stabilizer_types.h`, but I have some questions related to that file.
**5a)** For` typedef struct quaternion_s`, what is the difference between these 2 representations (q0,q1,q2,q3) vs (x,y,z,w)? Are they fully equivalent? In `controller_mellinger.c` code, it seems only the second one is used. So why do you have 2 representations?
**5b)** Are the quaternion representations _unit_ quaternions? Which _convention_ do you assume through the code? Hamilton or JPL? https://fzheng.me/2017/11/12/quaternion_conventions_en/
**5c)** `typedef struct sensorData_s { // Axis3f: see imu_types.h` // are these measurements the _raw_ sensor measurements? Or they the _filtered_ values after the Kalman filter? I would like to know which values used by the controller are raw and which are filtered? Are all of the on-board measurements _sampled_ at 1kHz?
 ```
 Axis3f acc;               // Gs  --> what does this unit mean?
  point_t position;         // m // is this the position of the center of mass of the cf2 measured by vicon? Is the sampling frequency 100Hz?
#ifdef LOG_SEC_IMU          // where is LOG_SEC_IMU defined? What is difference between accSec and acc and between gyroSec and gyro?
  Axis3f accSec;            // Gs
  Axis3f gyroSec;           // deg/s
#endif
  uint64_t interruptTimestamp; // what is this? difference with timestamp?
} sensorData_t;
```
**5d)**
```
typedef struct state_s { //
  attitude_t attitude;      // deg (legacy CF2 body coordinate system, where pitch is inverted) // So how is the CF2 body coordinate system chosen? x-axis to front of cf2, y-axis to right of cf2 and z-axis using the right hand rule (so pointing down)? What do you mean with pitch inverted? With respect to which case is pitch inverted?
  quaternion_t attitudeQuaternion; // So which convention (Hamilton or JPL?) is used for these (unit?) quaternions? Is this quaternion also corresponding to the orientation of the  CF2 body coordinate system, where pitch is inverted or is it only for attitude?
} state_t;
```
**5e)**
```
 typedef struct control_s {
  int16_t roll;
  int16_t pitch;
  int16_t yaw;
  float thrust;
} control_t; // about which axis are these control variables working (moment roll , moment pitch, moment yaw, thrust) defined? roll, pitch, yaw about which axis? Is it wrt CF2 body coordinate system, where pitch is inverted? Does thrust represent the total thrust or the thrust of 1 motor? Do these variables represent the physical moments and forces with units (e.g. Nm, Nm, Nm and N)?
// why does does this control_s not have a timestamp assigned like e.g. setpoint_t?
```

**5f)**
```
typedef enum mode_e {
  modeDisable = 0, // 
  modeAbs, // assinged automatically to 1 (previous +1)
  modeVelocity // assinged automatically to 2 (previous +1)
} stab_mode_t; // what do these modes: modeDisable, modeAbs and modeVelocity represent? Where in the code is decided if  stab_mode_t = 0,1 or 2?
```

**5g)** 
```
[... 162 lines cut ...]
```

**Question 23:**
in `motors.c`:
```
// Ithrust is thrust mapped for 65536 <==> 60 grams
void motorsSetRatio(uint32_t id, uint16_t ithrust)
{
  if (isInit) {
    uint16_t ratio;
    ASSERT(id < NBR_OF_MOTORS);
    ratio = ithrust;
  #ifdef ENABLE_THRUST_BAT_COMPENSATED
    if (motorMap[id]->drvType == BRUSHED)
    {
      float thrust = ((float)ithrust / 65536.0f) * 60; // so this is thrust expressed in grams (so a force if you would also multiply by gravity constant?)
      // is there any documention (paper/work) where I can find this mapping explained?
      float volts = -0.0006239f * thrust * thrust + 0.088f * thrust; // is there any documention where I can find this mapping?
      float supply_voltage = pmGetBatteryVoltage(); // how significant is the influence of the battery voltage drop on the thrust coefficients?
      float percentage = volts / supply_voltage;
      percentage = percentage > 1.0f ? 1.0f : percentage; // what does this line mean?
      ratio = percentage * UINT16_MAX; // what does UINT16_MAX represent?
      motor_ratios[id] = ratio;
    }
```

whoenig, 2019-02-15:
Hi,
Those are too many questions for me to respond to - you seem to have several subquestions in a number of the "main" questions... For some of the question, you will need to read the rest of the code of the firmware, to understand units and conventions. Some of your confusion can be clarified by two facts:
1. The firmware supports different kinds of setpoints, for example for velocity control, or position control. Thus, the controller has to support different "modes".
2. The units in the firmware are not SI units. The units in the controller itself, are also non standard, i.e., there is no physical meaning.

About some of your questions:
1. This might be a "bug". You can include "stabilizer_types.h" directly, instead.
2. Static in global scope in C just means that the variable cannot be used from a different C file.
3. This bounds the I-part and can be obtained by guessing how large the steady-state error can be. This is common for PID controllers.
4. They are called using function pointers, see controller.c
5a. This uses a union, so you get different names for the same components.
5b. Hamilton
23. https://wiki.bitcraze.io/misc:investigations:thrust

emmasmith100, 2019-02-16:
Hi @whoenig,
Thanks for considering some of my questions.
I am very sorry if I asked too many questions in one post. 
I don't know which rules I should follow if I ask questions here, that's not clear to me at all.
I just listed all the questions I had when I went chronologically (and multiple times) through the mellinger paper, `controller_mellinger.c`, and related files.
I just thought that these questions could help me and several other people in the future who start working with the crazyflie.

My follow up questions on your answers to part of my questions:
> For some of the question, you will need to read the rest of the code of the firmware, to understand units and conventions

I would be very happy if you could point me to files where it is explicitly stated which axis definitions are used. In the files I referred to in my questions (mainly `math3d.h`) I have not seen any axis definitions, neither definitions of Euler angle conventions used (axis order, intrinsic/extrinsic), or quaternions (Hamilton/JPL). Because of your answer I know you use the Hamilton convention, but it was not clear from `math3d.h`. So what's the convention used for RPY?
Halfway in controller_mellinger.c there start appearing many - signs to convert to other axis definitions. It is not clear which frame definition wrt CF2 you consider in the first part for the code and which in the second part. Note also in the mellinger paper they define axis on a quadrotor along the lines that connect 2 motors. The crazyflie uses the X configuration, which is different.

> The units in the firmware are not SI units. The units in the controller itself, are also non standard, i.e., there is no physical meaning.

Thanks for pointing me to [https://wiki.bitcraze.io/misc:investigations:thrust](https://wiki.bitcraze.io/misc:investigations:thrust), now I understand  the mapping `float volts = -0.0006239f * thrust * thrust + 0.088f * thrust;` as defined in `motors.c`.
Though still it is unclear to me in `power_distribution_stock.c` why you divide by 2:
```
int16_t r = control->roll / 2.0f;
int16_t p = control->pitch / 2.0f;
```
If you cannot answer my other questions, do you know someone who you think can respond to my questions?
Thank you for your time!

whoenig, 2019-02-18:
I didn't want to discourage you to ask questions and I hope that you'll find all the answer eventually. All I wanted to say is that this project is mostly run in free time, so questions that are very time consuming might not get answered.

For the conventions, there might be no direct comments in the code, but you can infer from the equations that are used in the code.

For the negative signs and the power distribution: Unfortunately, there are many inconsistencies in the firmware. You can find a non-exhaustive list here: https://lib.dr.iastate.edu/etd/15575/ (Appendix A). Essentially, those signs compensate for those issues.

You can ask generic firmware questions in the official forum, which has more people reading and writing there than the github issues here.

emmasmith100, 2019-02-19:
Thanks for your answer @whoenig.
Unfortunately, the download link seems not to be working.
Could you please send me (emma.smith100@hotmail.com) the PDF version?
Thanks!

whoenig, 2019-02-19:
I sent an email.

emmasmith100, 2019-02-19:
Thank you! This will help me forward!