USC-ACTLab/crazyswarm #85: ERROR: Could not add the interface to the multicast group

klauswq, 2018-07-31:
i got error message when i launch hover_swarm.launch. The error message is as follows:
[image]
The chooser.py works fine, there is no response when i press the takeoff button as well as when i run figure8_csv.py in another terminal.

whoenig, 2018-08-01:
This looks like the crazyswarm_server is unable to connect to your optitrack system (and basically waiting for the connection to be established). Make sure the optitrack settings are correct (https://github.com/USC-ACTLab/crazyswarm/blob/master/ros_ws/src/crazyswarm/launch/hover_swarm.launch#L44-L45). You might also want to check if your optitrack system works with ROS (other than using the crazyswarm).

klauswq, 2018-08-01:
Sorry to bother you again.

Do i have to specify the address of local IP and server IP? 
like   optitrack_local_ip: "192.168.0.1" and   optitrack_server_ip: "192.168.0.2" .

my setting is currently like this : 

<?xml version="1.0"?>
<launch>
  <arg name="joy_dev" default="/dev/input/js0" />

  <rosparam command="load" file="$(find crazyswarm)/launch/crazyflieTypes.yaml" />
  <rosparam command="load" file="$(find crazyswarm)/launch/crazyflies.yaml" />

  <node pkg="crazyswarm" type="crazyswarm_server" name="crazyswarm_server" output="screen" >
    <rosparam>
      firmware: "crazyswarm" # one of "crazyswarm", "bitcraze"
      broadcast_address: "FFE7E7E7E7"
      world_frame: "/world"
      genericLogTopics: ["log1"]
      genericLogTopicFrequencies: [10]
      # genericLogTopic_log1_Variables: ["ekfprof.usec_setup", "ekfprof.usec_innov", "ekfprof.usec_gain", "ekfprof.usec_corr", "ekfprof.usec_cov"]
      # genericLogTopic_log1_Variables: ["profiling.usec_ekf", "profiling.usec_traj", "profiling.usec_ctrl", "profiling.usec_idle"]
      genericLogTopic_log1_Variables: ["stabilizer.x", "ctrltarget.x", "vicon.x", "stabilizer.z", "ctrltarget.z", "vicon.z"]
      # genericLogTopic_log1_Variables: ["stateEstimate.roll", "ctrlMel.rolld"]
      genericLogTopic_log1_Variables: ["ctrlStat.edist"]
      genericLogTopic_log1_Variables: ["stateEstimate.yaw", "ctrltarget.yaw", "stateEstimate.x", "ctrltarget.x"] #, "ctrltarget.x", "stateEstimate.roll", "ctrlMel.rolld"]
      firmwareParams: # for all (independent of type)
        commander:
          enHighLevel: 1
        ring:
          effect: 14 # 6: double spinner, 7: solid color, 14: packetRate
          solidBlue: 255 # if set to solid color
          solidGreen: 0 # if set to solid color
          solidRed: 0 # if set to solid color
          headlightEnable: 0
        #ekf:
        #  ext_var_xy: 1.5e-7 # 1e-7 # 1.5e-7
        #  ext_var_vel: 2e-4 # 2e-4
        #  ext_var_q: 4.5e-3 # 2e-3 # 4.5e-3
        #  gyro_var_xy: 0.2e-2 # 0.2e-4
        #  gyro_var_z: 0.2e-2 # 0.2e-4
        #  acc_var_xyz: 2.4e-3 # 2.4e-3
      # tracking
      motion_capture_type: "optitrack" # one of vicon,optitrack
      object_tracking_type: "libobjecttracker" # one of motionCapture,libobjecttracker
      # vicon_host_name: "vicon"
      optitrack_local_ip: "localhost"
      optitrack_server_ip: "optitrack"
      save_point_clouds: ~/pointCloud.ot
      print_latency: True
      write_csvs: False
      force_no_cache: False
      enable_parameters: True
      enable_logging: True
    </rosparam>
  </node>

  <node name="joy" pkg="joy" type="joy_node" output="screen">
    <param name="dev" value="$(arg joy_dev)" />
  </node>

  <node pkg="crazyswarm" type="crazyswarm_teleop" name="crazyswarm_teleop" output="screen">
    <param name="csv_file" value="$(find crazyswarm)/launch/figure8_smooth.csv" />
    <param name="timescale" value="0.8" />
  </node>

  <node name="rviz" pkg="rviz" type="rviz" args="-d $(find crazyswarm)/launch/test.rviz"/>

  <node pkg="rqt_plot" type="rqt_plot" name="rqt_plot_x" args="/cf2/log1/values[0]"/>
  <node pkg="rqt_plot" type="rqt_plot" name="rqt_plot_roll" args="/cf1/log1/values[2] /cf1/log1/values[3]"/>

</launch>

whoenig, 2018-08-01:
Yes, you need to specify the IPs (hostnames should work as well though).

klauswq, 2018-08-02:
I try to specify the IP address of optitrack local and server, it seems to work. However, the crazyswarm_sever keeps dieing. The error log is as follows:

process[crazyswarm_server-1]: started with pid [17718]
process[joy-2]: started with pid [17719]
process[crazyswarm_teleop-3]: started with pid [17720]
process[rviz-4]: started with pid [17735]
[ INFO] [1533218489.148220303]: Wait for services...
[ INFO] [1533218489.154315509]: Manager ready.
NatNetClient connecting: 192.168.0.2 to 192.168.0.1...
WARNING: Could not set receive buffer size. Asked for 1048576B got 0B
[Client] Server Software: Motive
[Client] NatNetVersion: 3.0
[Client] ServerVersion: 2.0
NatNet server version: 16820416
ch: 1
[ INFO] [1533218489.266151404]: Opened joystick: /dev/input/js0. deadzone_: 0.050000.
[ INFO] [1533218489.282824726]: Adding CF: cf2 (radio://0/70/2M/E7E7E7E702, cf2)...
[ INFO] [1533218489.294249764]: CF ctor: 0.011273 s
[ INFO] [1533218489.305312215]: [cf2] Requesting parameters...
[ INFO] [1533218489.315430250]: Found variables in cache.
[ INFO] [1533218489.696739515]: [cf2] reqParamTOC: 0.402403 s
[ INFO] [1533218489.696794099]: Requesting memories...
[ INFO] [1533218489.704861132]: Memories: 3
[ INFO] [1533218489.717090899]: [cf2] Ready. Elapsed: 0.422655 s
[ INFO] [1533218489.717778805]: CF run: 0.423482 s
update commander/enHighLevel to 1
update ring/effect to 14
update ring/headlightEnable to 0
update ring/solidBlue to 255
update ring/solidGreen to 0
update ring/solidRed to 0
update ctrlMel/i_range_m_z to 1500
update ctrlMel/i_range_xy to 2
update ctrlMel/i_range_z to 0.4
update ctrlMel/kR_xy to 70000
update ctrlMel/kR_z to 60000
update ctrlMel/kd_omega_rp to 200
update ctrlMel/kd_xy to 0.2
update ctrlMel/kd_z to 0.4
update ctrlMel/ki_m_z to 500
update ctrlMel/ki_xy to 0.05
update ctrlMel/ki_z to 0.05
update ctrlMel/kp_xy to 0.4
update ctrlMel/kp_z to 1.25
update ctrlMel/kw_xy to 20000
update ctrlMel/kw_z to 12000
update ctrlMel/mass to 0.032
update ctrlMel/massThrust to 132000
[ INFO] [1533218489.773254347]: [cf2] Update parameters
[ERROR] [1533218489.773348805]: Could not find param commander/enHighLevel
[ERROR] [1533218489.776487083]: Could not find param ctrlMel/i_range_m_z
[ERROR] [1533218489.776531228]: Could not find param ctrlMel/i_range_xy
[ERROR] [1533218489.776559820]: Could not find param ctrlMel/i_range_z
[ERROR] [1533218489.776587815]: Could not find param ctrlMel/kR_xy
[ERROR] [1533218489.776611856]: Could not find param ctrlMel/kR_z
[ERROR] [1533218489.776634687]: Could not find param ctrlMel/kd_omega_rp
[ERROR] [1533218489.776657145]: Could not find param ctrlMel/kd_xy
[ERROR] [1533218489.776681244]: Could not find param ctrlMel/kd_z
[ERROR] [1533218489.776704015]: Could not find param ctrlMel/ki_m_z
[ERROR] [1533218489.776726893]: Could not find param ctrlMel/ki_xy
[ERROR] [1533218489.776749156]: Could not find param ctrlMel/ki_z
[ERROR] [1533218489.776771778]: Could not find param ctrlMel/kp_xy
[ERROR] [1533218489.776795079]: Could not find param ctrlMel/kp_z
[ERROR] [1533218489.776816876]: Could not find param ctrlMel/kw_xy
[ERROR] [1533218489.776838989]: Could not find param ctrlMel/kw_z
[ERROR] [1533218489.776860939]: Could not find param ctrlMel/mass
[ERROR] [1533218489.776882842]: Could not find param ctrlMel/massThrust
[ INFO] [1533218489.788485931]: Update params: 0.070085 s
[ INFO] [1533218489.789442884]: Started 1 threads
[crazyswarm_server-1] process has died [pid 17718, exit code -11, cmd /home/jingwen/crazyswarm/ros_ws/devel/lib/crazyswarm/crazyswarm_server __name:=crazyswarm_server __log:=/root/.ros/log/e70a153a-90cb-11e8-ae22-54ee75238326/crazyswarm_server-1.log].
log file: /root/.ros/log/e70a153a-90cb-11e8-ae22-54ee75238326/crazyswarm_server-1*.log

whoenig, 2018-08-02:
How does your allCrazyflies.yaml look like? Also, your second CF does not have the correct firmware.

whoenig, 2018-08-03:
1. Please add `launch-prefix="xterm -e gdb --args"` in the hover_swarm launch file in order to debug why the server crashes. You can find more information here: http://wiki.ros.org/roslaunch/Tutorials/Roslaunch%20Nodes%20in%20Valgrind%20or%20GDB. I haven't seen that error in a while and don't have a good guess at the moment.

2. You can see `Could not find param` in the log, which indicates a wrong firmware version.

LZMHIT, 2018-10-09:
> 
> 
> I try to specify the IP address of optitrack local and server, it seems to work. However, the crazyswarm_sever keeps dieing. Can you give me some hint about this error. The error log is as follows:
> 
> process[crazyswarm_server-1]: started with pid [17718]
> process[joy-2]: started with pid [17719]
> process[crazyswarm_teleop-3]: started with pid [17720]
> process[rviz-4]: started with pid [17735]
> [ INFO] [1533218489.148220303]: Wait for services...
> [ INFO] [1533218489.154315509]: Manager ready.
> NatNetClient connecting: 192.168.0.2 to 192.168.0.1...
> WARNING: Could not set receive buffer size. Asked for 1048576B got 0B
> [Client] Server Software: Motive
> [Client] NatNetVersion: 3.0
> [Client] ServerVersion: 2.0
> NatNet server version: 16820416
> ch: 1
> [ INFO] [1533218489.266151404]: Opened joystick: /dev/input/js0. deadzone_: 0.050000.
> [ INFO] [1533218489.282824726]: Adding CF: cf2 (radio://0/70/2M/E7E7E7E702, cf2)...
> [ INFO] [1533218489.294249764]: CF ctor: 0.011273 s
> [ INFO] [1533218489.305312215]: [cf2] Requesting parameters...
> [ INFO] [1533218489.315430250]: Found variables in cache.
> [ INFO] [1533218489.696739515]: [cf2] reqParamTOC: 0.402403 s
> [ INFO] [1533218489.696794099]: Requesting memories...
> [ INFO] [1533218489.704861132]: Memories: 3
> [ INFO] [1533218489.717090899]: [cf2] Ready. Elapsed: 0.422655 s
> [ INFO] [1533218489.717778805]: CF run: 0.423482 s
> update commander/enHighLevel to 1
> update ring/effect to 14
> update ring/headlightEnable to 0
> update ring/solidBlue to 255
> update ring/solidGreen to 0
> update ring/solidRed to 0
> update ctrlMel/i_range_m_z to 1500
> update ctrlMel/i_range_xy to 2
> update ctrlMel/i_range_z to 0.4
> update ctrlMel/kR_xy to 70000
> update ctrlMel/kR_z to 60000
> update ctrlMel/kd_omega_rp to 200
> update ctrlMel/kd_xy to 0.2
> update ctrlMel/kd_z to 0.4
> update ctrlMel/ki_m_z to 500
> update ctrlMel/ki_xy to 0.05
> update ctrlMel/ki_z to 0.05
> update ctrlMel/kp_xy to 0.4
> update ctrlMel/kp_z to 1.25
> update ctrlMel/kw_xy to 20000
> update ctrlMel/kw_z to 12000
> update ctrlMel/mass to 0.032
> update ctrlMel/massThrust to 132000
> [ INFO] [1533218489.773254347]: [cf2] Update parameters
> [ERROR] [1533218489.773348805]: Could not find param commander/enHighLevel
> [ERROR] [1533218489.776487083]: Could not find param ctrlMel/i_range_m_z
> [ERROR] [1533218489.776531228]: Could not find param ctrlMel/i_range_xy
> [ERROR] [1533218489.776559820]: Could not find param ctrlMel/i_range_z
> [ERROR] [1533218489.776587815]: Could not find param ctrlMel/kR_xy
> [ERROR] [1533218489.776611856]: Could not find param ctrlMel/kR_z
> [ERROR] [1533218489.776634687]: Could not find param ctrlMel/kd_omega_rp
> [ERROR] [1533218489.776657145]: Could not find param ctrlMel/kd_xy
> [ERROR] [1533218489.776681244]: Could not find param ctrlMel/kd_z
> [ERROR] [1533218489.776704015]: Could not find param ctrlMel/ki_m_z
> [ERROR] [1533218489.776726893]: Could not find param ctrlMel/ki_xy
> [ERROR] [1533218489.776749156]: Could not find param ctrlMel/ki_z
> [ERROR] [1533218489.776771778]: Could not find param ctrlMel/kp_xy
> [ERROR] [1533218489.776795079]: Could not find param ctrlMel/kp_z
> [ERROR] [1533218489.776816876]: Could not find param ctrlMel/kw_xy
> [ERROR] [1533218489.776838989]: Could not find param ctrlMel/kw_z
> [ERROR] [1533218489.776860939]: Could not find param ctrlMel/mass
> [ERROR] [1533218489.776882842]: Could not find param ctrlMel/massThrust
> [ INFO] [1533218489.788485931]: Update params: 0.070085 s
> [ INFO] [1533218489.789442884]: Started 1 threads
> [crazyswarm_server-1] process has died [pid 17718, exit code -11, cmd /home/jingwen/crazyswarm/ros_ws/devel/lib/crazyswarm/crazyswarm_server __name:=crazyswarm_server __log:=/root/.ros/log/e70a153a-90cb-11e8-ae22-54ee75238326/crazyswarm_server-1.log].
> log file: /root/.ros/log/e70a153a-90cb-11e8-ae22-54ee75238326/crazyswarm_server-1*.log

Hello, I have the same problem with connecting with Optitrack, Could you please tell me how to confirm the optitrack_local_ip and optitrack_server_ip?

LZMHIT, 2018-10-09:
[image]
I have a problem, When I launch mocap_helper.launch, and I got this message, I am sure  my computer and the computer running Optitrack system are in the same network, and the Ip of my is 192.168.222.111, while the ip of the computer running Optitrack system is 192.168.222.153, so could you please tell me how to modify the args at the correspond area in the launch file.
[image]

LZMHIT, 2018-10-09:
> running

Thank you for your replying, I give you a screenshot, as I can see, It is Version 1.9.0. [image]

whoenig, 2018-10-09:
This version should work (to the best of my knowledge). It looks like you have an object defined in motive. In your setting (libobjecttracker), you should not define any objects in Motive.

LZMHIT, 2018-10-09:
> This version should work (to the best of my knowledge). It looks like you have an object defined in motive. In your setting (libobjecttracker), you should not define any objects in Motive.

I tried to change, unfortunately，it dosen't work. I give you more pictures, on the computer with windows system running the Optitrack system, I looked at the ip address, and on the computer with ubuntu system,I also did that, I wonder whether I make wrong Ip address.
[image]
[image]
And the settings change
[image]
[image]
Besides, I wonder know what else software should I run besides Optitrack?

whoenig, 2018-10-09:
Can you ping your windows machine from the Ubuntu machine? You shouldn't need to run any other SW.

LZMHIT, 2018-10-09:
> Can you ping your windows machine from the Ubuntu machine? You shouldn't need to run any other SW.

Yes, I can, I close  net firewall both in windows and ubuntu, I ping 192.168.222.153 and it really works.
[image]

whoenig, 2018-10-09:
The latency is very high, so you probably want a physical ethernet connection between the machines instead of WiFi. (Of course, this won't solve the connection issues you are seeing, unless your wifi/firewall is blocking some ports.)

jpreiss, 2020-06-13:
closing, I doubt this is a general issue with Crazyswarm.

closed 2020-06-13 by jpreiss