USC-ACTLab/crazyswarm #87: Support for arbitrary polynomial powers

iron-beagle, 2018-08-11:
The 4k byte memory limit on the trajectory only allows for polynomials of up to degree seven to be executed. I am curious to know what the best way to go about increasing this limit would be?

As of right now it is my thought to create a buffer to allow loading of a longer trajectory from either uSD or over the radio. Theoretically this would increase the trajectory piece space to the point where the major limiting factor behind flight length or complexity is the radio power and battery life. 

This could be achieved by sending the trajectory from-

* radio to board
* radio to uSD then to board when starting the trajectory
* pre-staged on uSD to be used on board when starting the trajectory

Our use-case is to run a path finding curve generation algorithm that is currently limited by the polynomial power.

jpreiss, 2018-08-11:
@whoenig recently added support for arbitrary size polynomial uploads in length but not degree. I'm pretty sure this would not be too difficult to add.

On the other hand, it's quite possible that numerical stability issues will appear with very high degree polynomials. Degree 7 is sufficient to ensure a continuous fourth derivative, making it physically plausible for a quadrotor. You could also consider using a spline fitting tool to approximate your high degree polynomials with lower degree and more pieces.

whoenig, 2018-08-13:
The Crazyswarm and official firmwares are almost identical at this point (and in fact, you can use the crazyswarm with the latest official firmware with a few minor drawbacks). In both firmwares, you have to look at the high-level-commander:
https://github.com/USC-ACTLab/crazyflie-firmware/blob/crazyswarm-next/src/modules/src/crtp_commander_high_level.c

The actual polynomial evaluation is done here:
https://github.com/USC-ACTLab/crazyflie-firmware/blob/crazyswarm-next/src/modules/src/pptraj.c

The actual upload just fills in a buffer, so you only need to change the logic in the firmware on how you interpret the buffer (and crazyflie_ros to upload in a different data format).

Our simulator directly uses the C-code from the firmware, so the best way to get started would be to get it to work within the simulation.

jpreiss, 2020-06-13:
@iron-beagle did you ever make progress on this?

iron-beagle, 2020-06-22:
I did not. The rest of the team is not in a position to share code legally. As far as I know they did not make progress either though.

closed 2020-07-03 by jpreiss