USC-ACTLab/crazyswarm #89: Crazyflie Trajectory Generation

YimingWan, 2018-08-27:
Hi,
I'm planning on using quintic polynomial trajectories to traverse paths defined by via points. I have the position, velocity and acceleration at each point. 
1) Is there a way to use this data as input to create a trajectory rather than using the API to upload and start a trajectory? 
2) Can I get the full state estimate of the crazyflie as it moves along the trajectory. 
3) Also, Is it possible to implement dynamic trajectory generation to alter the trajectory if an obstacle is detected mid way to the goal?

whoenig, 2018-08-27:
There are two ways: 
1. You implement everything in the firmware. This is a major addition, as you'll need to alter the communication protocol and add a new way to evaluate trajectories. The plus side is that you have full access to the state estimate. For your third point, you might not have enough on-board computational power, depending on your planning method.
2. You evaluate trajectories and re-plan on a host-computer and sent the current setpoint (including position, velocity, acceleration etc.) to the CF. The downsides are that this requires more radio bandwidth and that you won't have access to the full state estimate (you can stream parts of the state estimate at ~100 Hz though).

YimingWan, 2018-08-27:
Thank you for the reply. Would using a 7th order polynomial give access to the full state estimate without changing the firmware? and what commands do I use to send the current setpoint to the CF and to subscribe to the state estimate?

closed 2020-06-13 by jpreiss