USC-ACTLab/crazyswarm #91: Non-smooth outputs from uav_trajectories

jpreiss, 2018-09-17:
I tried the uav_trajectories library and it works really well. Tough when generating a trajectory for fixed waypoints at every goal all the derivatives are 0 consistently that makes the flight really discontinuous. See my output attached:

[image]

I checked the crazyswarm project and found some matlab scripts for generating a time variable continuous trajectory for the figure8, circle etc. My only problem this script looks a bit complicated for me...
My question would be:

- Is there a setting for the uav_trajectory or other similar library that generates continuous trajectory for all derivatives thus making a smooth flight between the waypoints.

- Is there a documentation or more high level script for the matlab scripts for just entering the waypoints and constraints in time. Or is there a way to do that dynamically inside crazyswarm project?

- Or shall I just understand those and generate the path for myself?

All your works are highly appreciated and looking for your answer!

_Originally posted by @RozDavid in https://github.com/USC-ACTLab/crazyswarm/issues/59#issuecomment-422053678_

jpreiss, 2018-09-17:
I would not recommend the Matlab tools in `crazyswarm` because they do not include minimum-snap trajectory optimization using quadratic programming. Those scripts are based on the idea of finding a particular solution to the underdetermined system of linear equations imposed by the waypoints, and using the null space of the system as a reduced-dimensionality space for optimization with arbitrary cost functions (i.e. the approach described in [1]).

The particular solution returned by this method is the Moore-Penrose pseudoinverse which minimizes the L2 norm of the polynomial coefficients. However, this objective does not actually correspond to any particular desirable property of the trajectory, although it is usually fairly reasonable.

So I would suggest trying to find the source of the issue in `uav_trajectories` rather than switching to the Matlab tools in `crazyswarm/matlab/trajectorygen/`.

What exactly are you plotting in your output? Are those the L2 norms of the velocity, acceleration, and omega vectors?

[1] Hausman, K., Preiss, J., Sukhatme, G. S., & Weiss, S. Observability-Aware Trajectory Optimization for Self-Calibration With Application to UAVs. IEEE Robotics and Automation Letters, 2(3), 2017. [http://sci-hub.tw/10.1109/LRA.2017.2647799](http://sci-hub.tw/10.1109/LRA.2017.2647799)

whoenig, 2018-09-17:
I guess this should be an issue in uav_trajectories. In general, we only add constraints for the first and last waypoint for higher-order derivatives. You can also build your own tools on top of https://github.com/ethz-asl/mav_trajectory_generation. I suggest opening a new issue in uav_trajectories that includes your input file. Thanks!

RozDavid, 2018-09-18:
Hi @whoenig, @jpreiss,

Thank you very much for your feedback, I started testing today with the mav_trajectory_generation. There is a way in the code, where we can set additional constraints for [speed](https://github.com/ethz-asl/mav_trajectory_generation/blob/24e6c2638e3c67f4a0e2185243948109d9f61f5d/mav_trajectory_generation_ros/test/test_feasibility.cpp#L76) at the waypoints, that made the result better in this specific elliptic case.