USC-ACTLab/crazyswarm #93: M2 flashes fast

netkimjh, 2018-09-21:
Hi,
I find M2 led flashes fast when I put cf2 on the ground.
This is causing that cf2 flies badly.
I recorded three videos.

1. putting on the ground
[https://youtu.be/FL4V12uCIB8](https://youtu.be/FL4V12uCIB8)

2. rebooting using chooser.py on the ground
[https://youtu.be/KN-phQ6FmWI](https://youtu.be/KN-phQ6FmWI)

3. rebooting using chooser.py not on the ground
[https://youtu.be/Pg3_HYev5h0](https://youtu.be/Pg3_HYev5h0)

Do you have any idea about it?

Thank you.

whoenig, 2018-09-21:
This is the expected behavior. If you hold the CF in your hand, it can't calibrate (that is find the gyro bias), so it will blink at 1Hz. Once you put it down, calibration will succeed and blink at 2Hz. System alive means that the firmware is not crashed (which shouldn't happen with any current firmware anyways).

netkimjh, 2018-09-21:
Thank you for your reply.