USC-ACTLab/crazyswarm #95: Simulation mode has some issues

Alexwei92, 2018-09-25:
Hi guys,

I installed the package today and tried to run 'python figure8_csv.py --sim' to view the trajectory before I can implement it on my robots. However, I got the error message as follows:
```
Traceback (most recent call last):
  File "figure8_csv.py", line 5, in <module>
    from pycrazyswarm import *
  File "/home/peng/crazyswarm/ros_ws/src/crazyswarm/scripts/pycrazyswarm/piecewise.py", line 1, in <module>
    from .cfsim import cffirmware as firm
ImportError: cannot import name cffirmware
```
The problem comes from the 'piecewise.py' file when it tried to import the 'cffirmware' from 'cfsim'. I also 
 ran other files which use 'cfsim' package but none of them worked. Then, I went to the 'cfsim' folder and tried to compile it, but I got the following errors which I don't know what to do next:
```
swig -python -I../../../../../../crazyflie-firmware/src/modules/interface cffirmware.i
cffirmware.i:12: Error: Unable to find 'packetdef.h'
Makefile:6: recipe for target 'swig' failed
make: *** [swig] Error 1
```
Can someone tell me what shall I do in order to use the simulation feature?

Thanks,
Peng

whoenig, 2018-09-26:
I think I might have pushed a wrong link to the crazyflie-firmware. Can you please try the following:

```
cd crazyflie-firmware
git checkout crazyswarm-next
cd ../
./build.sh
```
And report if that fixes your issue? Once you are on the right branch, you should have packetdef.h in crazyflie-firmware/src/modules/interface/.

Thanks!
Wolfgang

Alexwei92, 2018-09-26:
Hi Wolfgang,

Thanks for your help! Yes, it does solve the problem. But I added 'git
checkout crazyswarm-next' into the Makefile because it switches back to the
'master' by default.

Thanks again,
Peng

whoenig, 2018-09-26:
Thanks for the update, Peng! I pushed a fix that points to the right branch and should fix the problem for everyone else.

closed 2018-09-26 by whoenig