USC-ACTLab/crazyswarm #97: About wind disturbances

srnetsr, 2018-10-03:
Hi,
I am testing two cf2s with wind disturbances.
The wind from propellers is stronger than what I expected, so the cf2 failed to hover under the other cf2.
The difference of altitude was 1 meter.

So, I would like to have a look and apply nonlinear controller, could you let me know where is the control part in crazyswarm package?

Is there any method to hover cf2 under the other cf2?

Thank you.

jpreiss, 2018-10-12:
The controller is in [crazyflie-firmware/src/modules/src/controller_mellinger.c](https://github.com/USC-ACTLab/crazyflie-firmware/blob/crazyswarm-next/src/modules/src/controller_mellinger.c).

The air stream from the upper quadrotor is usually called "downwash". In our tests, we found that 0.6m was sufficient distance in the z axis for two Crazyflies using the controller linked above. So, I am a little surprised that you are having issues at 1m. Are you using the Crazyswarm controller or the default Bitcraze controller?

In [our paper](act.usc.edu/publications/Hoenig_TRO2018.pdf) [1], we develop multi-robot motion planning algorithms that take downwash into account. The quadrotors must maintain a large distance in the vertical axis while they are allowed to fly much closer together in the horizontal axes.

A different approach is to try to design a controller that can counteract the downwash force, or more generally any external wind disturbance. I am not sure what's the state of the art in this area.

> [1] W. Hönig, J. A. Preiss, T. K. S. Kumar, G. S. Sukhatme, and N. Ayanian. "Trajectory Planning for Quadrotor Swarms", in IEEE Transactions on Robotics (T-RO), Special Issue Aerial Swarm Robotics, vol. 34, no. 4, pp. 856-869, August 2018.

srnetsr, 2018-10-18:
Hi @jpreiss , I appreciate you for your reply, and apologize for my tardy reply.
I am using crazyswarm pacakge, and I did not change any setting before and the test was that two quadrotors hover at the same position in 2D space with different altitude (1.5m, 0.5m).
At that time, I just executed `roslaunch crazyswarm hover_swarm.launch`.
and used python code including `goTo`.

Is there any setting for using the controller ([crazyflie-firmware/src/modules/src/controller_mellinger.c](crazyflie-firmware/src/modules/src/controller_mellinger.c))?
What should I do to use the controller you mentioned?

Thank you.

whoenig, 2018-10-18:
The crazyswarm uses the mellinger controller by default (see https://github.com/USC-ACTLab/crazyswarm/blob/master/ros_ws/src/crazyswarm/launch/hover_swarm.launch#L26). The gains can be adjusted here: https://github.com/USC-ACTLab/crazyswarm/blob/master/ros_ws/src/crazyswarm/launch/crazyflieTypes.yaml#L9-L26). The 0.6m we measured was with the default settings provided in the package, with CFs that have 4 mocap markers and the LED ring (33g total weight). However, it might depend on the age of your battery as well.

srnetsr, 2018-10-21:
Thank you for your reply, @whoenig .
I think all setting for me is the same as you described.
The battery is the new one and showing proper voltage and condition.
I will check the weight soon.
Do you think is there any other parameter which can affect on this issue?

Thank you very much in advance.

whoenig, 2018-10-23:
I can't think of any SW parameter (assuming your tracking system performs well and has a low latency). The motion capture system can have an effect, as well as the battery brand.

jpreiss, 2020-06-13:
@srnetsr were you able to resolve this?

jpreiss, 2021-01-30:
closing due to inactivity.

closed 2021-01-30 by jpreiss