USC-ACTLab/crazyswarm #98: vicon connection issues and directory confusion

pete9936, 2018-10-05:
Hi,
I believe I have a connection issue between Vicon and crazyswarm. When I run hover_swarm I have not been able to see any cf on rviz. A similar concern was brought up in Issue #37 but I think something else may be occurring here. 

I am successfully able to run the crazyflie_ros > crazyflie_demo files using the vicon_bridge that you outlined in your previous work. For this I am connecting to vicon over our remote network as opposed to locally with an ethernet connection but that shouldn't be an issue. 

I also noticed in Issue #29 that some pieces with the vicon installation have maybe recently changed. I followed what was done there and downloaded DataStreamSDK 1.7 and placed it in the directory specified `ros_ws/src/crazyflie_ros/externalDependencies/libmotioncapture/externalDependencies/vicon_sdk ` but had some confusion here. It seems that on this path you have already added `vicon-datastream-sdk`. The vicon_sdk structure I have is as follows:
[image]
Where just a portion of the vicon-datastream-sdk that you have prebuilt is:
[image]
I was wondering have you recently changed things and added all of this? I haven't seen anyone bring it up. Would it be alright to just delete the `vicon-datastream-sdk` that is preloaded and simply use my 'vicon_sdk' directory? Or are there may changes that would need to be made?

Following off that, I moved forward anyway and as you suggested in #37 I changed the hostname (`vicon`) to the IP address of our network when trying to run `hover_swarm.launch` by changing hostname in `crazyswarm_server.cpp` as follows:
[image]
And also tried this for `mocap_helper.launch` by modifying `mocap_helper.cpp` as follows:
[image]
And when I run this as detailed in the [Configure Marker Arrangement](https://crazyswarm.readthedocs.io/en/latest/usage.html#configure-marker-arrangement) section I get no output.
Just to be thorough, while attempting to run `hover_swarm.launch` I chose to use `motionCapture` as the `object_tracking_type` and had "track" enabled for my objects cf1 and cf2 with the proper uri convention specified. And when I ran `mocap_helper.launch` I was ceratin to change the object_tracking_type to `libobjecttracker` and not to have "track" enabled on the Tracker software. I certainly tried with and without for both.

Am I going about this process correctly? And what could/should I change? And sorry for the long question, just wanted to be thorough.

whoenig, 2018-10-05:
VICON recently published the SDK as open source, so there is no need anymore in manually putting the SDK (the readthedocs should reflect those changes). 

For your IP address: You should change the address in the launch file and not in the code, i.e., https://github.com/USC-ACTLab/crazyswarm/blob/master/ros_ws/src/crazyswarm/launch/hover_swarm.launch#L43 or https://github.com/USC-ACTLab/crazyswarm/blob/master/ros_ws/src/crazyswarm/launch/mocap_helper.launch#L9. The other things (tracking mode) sound perfectly good to me.

Thanks for the detailed report and please let me know if it works when you specify the IP address in the launch file.

pete9936, 2018-10-05:
Thank you for getting back so quickly. Yes the process is very simple and straight forward now! The `mocap_helper.launch` works now that I changed the IP address in that file and I am streaming data from Vicon as expected.

However, I made the same modifications in `hover_swarm.launch` and I am getting an output error. I made the modification as you suggested:
[image]
But for some reason I get an error when I run `roslaunch crazyswarm hover_swarm.launch`
[image]
I'm not sure if it's even associated with the Vicon changes. Rviz shows up but it is empty. For reference I am using Tracker 3.6 to track two crazyflies, cf1 and cf2. With `uri:=radio://0/80/2M/E7E7E7E701` and `uri:=radio://0/80/2M/E7E7E7E702`. I have changed the `allCrazyflies.yaml` file accordingly and sourced the ros_ws into my .bashrc.

Any suggestions?

pete9936, 2018-10-05:
I went poking around other issues and I found that this has come up a couple of times. It looks like Issue #77 came across a similar vicon issue and solved it the same way. However, they also had the same `process has died   exit code -6` error as described above. 

So I followed that thread which led to debugging using the technique of Issue #68 . This is where I'm now stuck. So I did the following like you said:
`$cd ros_ws
$rm -rf build/ devel/
$catkin_make -DCMAKE_BUILD_TYPE=Debug`
And then updated the `hover_swarm.launch` file with the debugging launch prefix:
[image]
Then in gdb I would `run` the node and then use the command `bt` to display the issue. However, I am unable to figure out what is wrong. Here is what I get from gdb:
[image]

Any ideas moving forward? I think I'm officially stuck for now.

pete9936, 2018-10-05:
Well, I suppose it looks like the issue is at `crazyswarm_server.cpp` but I have left that completely untouched.

whoenig, 2018-10-07:
The timeout exception means that the server couldn't connect to one or both CFs. Please double check the URIs and try to connect using the official client.

pete9936, 2018-10-08:
I looked back in the crazyflies.yaml file and my values were incorrect. Changed that and everything is working very nicely! I can successfully hover both of my Crazyflies!

Thanks so much for helping me out with this, and awesome work on the updated controller. The stability is really solid! 

I will close this issue since all is well.

closed 2018-10-08 by pete9936

whoenig, 2018-10-08:
Awesome, thanks for the update!

anikasaraf, 2024-02-02:
> I looked back in the crazyflies.yaml file and my values were incorrect. Changed that and everything is working very nicely! I can successfully hover both of my Crazyflies!
>
> Thanks so much for [...]

How did you decide what values to put in your crazyflies.yaml