USC-ACTLab/crazyswarm #99: Cannot fly multiple drones

Alexwei92, 2018-10-06:
Hi guys,

I was testing the multiple drones flying yesterday and got the following error when I have two CFs active (the single one works well). The program terminated when adding the second CF to the server I guess. Does anyone know the reason?

```
terminate called after throwing an instance of 'std::runtime_error'
  what():  No matching USB Device with devid = 1 found!
[crazyswarm_server-2] process has died [pid 25299, exit code -6, cmd /home/peng/crazyswarm/ros_ws/devel/lib/crazyswarm/crazyswarm_server __name:=crazyswarm_server __log:=/home/peng/.ros/log/c667424a-c9a5-11e8-aead-34e12d1fe9b0/crazyswarm_server-2.log].
log file: /home/peng/.ros/log/c667424a-c9a5-11e8-aead-34e12d1fe9b0/crazyswarm_server-2*.log
```

Thanks,
Peng

whoenig, 2018-10-07:
You need one radio per channel. It seems like you have only one radio plugged in, but configured the two CFs on two different channels.

Alexwei92, 2018-10-07:
Yes, that solved my problem. Sorry for this stupid question.

Thanks,
Peng

closed 2018-10-07 by Alexwei92