USC-ACTLab/crazyswarm #102: Timeout when updating firmware

sahibdhanjal, 2018-10-08:
Hi,
I was trying to get crazyswarm_ros running up on my desktop but was stuck with updating the firmware (at step 4) as per the [documentation](https://crazyswarm.readthedocs.io/en/latest/usage.html). Please find the details below:

I have 4 crazyflies on channel <strong>80</strong> with the addresses from ```0xE7E7E7E7{01-04}```. When I scan for each of them using ```rosrun crazyflie_tools scan --address 0xE7E7E7E7{01-04}```, I'm able to detect each of them. However, when I try flashing each of them in bootloader mode using the command mentioned with the respective address - ```rosrun crazyflie_tools flash --target nrf51 --filename prebuilt/cf2_nrf.bin --uri radio://0/80/1M/E7E7E7E7{01-04}```, I get a ```timeout``` error which I was able to trackback to the ```Crazyflie::sendPacketOrTimeout()``` function in ```Crazyflie.cpp```. 

Could you please help me resolve this issue?

whoenig, 2018-10-08:
If you put the CFs manually in bootloader mode (by pressing the power button for 3sec), you should not use the `--uri` flag.

sahibdhanjal, 2018-10-09:
Thanks for getting back this quick. I also tried putting them in reboot mode and not using the ```--uri``` flag, but in that case, it times out for handling requests in the ```Crazyflie::handleRequests``` function. I have also tried the following approaches, but all have been to no avail:
1. Booting the CF up and using ```flash``` both with and without the ```--uri``` flag
2. Booting the CF in bootloader mode and using ```flash``` without the ```-uri``` flag
3. I have also tried doing the same by connecting the CFs to the computer using USB as well as the CF Radio

How should I proceed now?

sahibdhanjal, 2018-10-09:
So I tried 1) and make cload was able to update the firmware. For 2), yes the blue LEDs were blinking, However, I don't know why I wasn't able to communicate with the CF. I'll proceed with flashing the firmware using approach 1
Thanks!

closed 2018-10-09 by sahibdhanjal