USC-ACTLab/crazyswarm #103: NATNET FRAME TIMEOUT When running mocap_helper.launch

LZMHIT, 2018-10-09:
Dear Author, 
  I am so sorry to bother you again, but I am so confused with the system that I can not make any progress in my work. As for the question I asked in #90 ,I found that I need to modify the Local interface ip address in the Optitrack SW. Like following pictures.
[image]
However,I meet another problem. I launch the mocap_helper.launch again, and I got this messages
[image]
It flashed with the same message. And no topics are pub.
[image]
And I have another Question, Do I need to turn on the crazyflie2.0 and plug in the CrazyRadio when I just launch the mocap_helper.launch file?
I really hope to receive your replay, Thank you so much.

whoenig, 2018-10-09:
mocap_helper does not use the crazyflie/crazyradio, so no need to plug anything else in. Unfortunately, the qualisys effort is community-driven and I haven't tested the system myself, so I am not sure what the issue is.

whoenig, 2018-12-07:
Closing this issue, see issue #122 for more discussions on this.

closed 2018-12-07 by whoenig