USC-ACTLab/crazyswarm #104: Crazyradio firmware update and chooser. py timeout error

astridtepie, 2018-10-10:
Hello,
I am actually working on the crazyswarm tutorial, I can't launch the hover_swarm file and I always get timeout error on the chooser.py server. I could update successfully my crazyflie but for one reason or another the firmware update of the crazyradio is not successful. The errors I have are shown on the pictures below, does any one have any clue on what should be done to get pass this problem?
Thanks in advance!
[image]
[image]

whoenig, 2018-10-11:
For the first error: It can't find the Crazyradio - did you set the udev rules?

For the second issue: you specify a folder, but you need to specify a file instead. ("python crazyradio-firmware/usbtools/nrfbootload.py flash prebuilt/cradio.bin").

astridtepie, 2018-10-12:
Yes I did set the udev rule and when I run the command with the specific file, the following error pop up.
[image]
[image]

whoenig, 2018-10-12:
You will need to use python instead of python3 (and install any dependencies that you might be missing for python2).

Is your current user account part of the plugdev group (type `groups` on the shell)?

astridtepie, 2018-10-15:
Yes I am, actually I can now visualize the cf on rviz, I just hqd to chqnge the parameters of the cf on the file. Thanks a lot for your help sir!

closed 2018-10-16 by whoenig