USC-ACTLab/crazyswarm #105: Chooser.py Timeout Error

sahibdhanjal, 2018-10-12:
Hi,
I was successfully able to flash the firmware for both the crazyflie and the crazyradio as per the documentation. However, when I run chooser.py, there is a timeout error whenever I press any button on the GUI. I was able to track it down and it is produced from the ```Crazyflie::sendPacketOrTimeout()``` function. I'm able to use scan each of the CF using ```rosrun crazyflie_tools scan``` by specifying the address, however, when I try to use ```rosrun crazyflie_tools battery```, that again gives me a timeout from ```Crazyflie::handleRequests()```. Could you suggest on how I should proceed to fix this issue?

whoenig, 2018-10-12:
Can you post the output of `rosrun crazyflie_tools scan`? Did you specify the uri when running `rosrun crazyflie_tools battery`?

sahibdhanjal, 2018-10-12:
Yeah, I specified the uri flag when running battery. The CF is connected via both the USB and the Radio. I added a small statement to help me track down the issue, which can be seen in the 3rd image. 
The screenshots are attached below:
[image]
[image]
[image]

whoenig, 2018-10-12:
Is the CF turned on? Can you connect using the official python client over the radio? Also, I suggest using 2M as datarate (this should not be an issue, however).

sahibdhanjal, 2018-10-12:
Yes, the CF is on and I can connect using the cfclient as well. I have also tried using all the available data rates but couldn't fix it.

sahibdhanjal, 2018-10-15:
Hi, any updates on the same?

whoenig, 2018-10-16:
Which firmware are you running? You can try pulling crazyflie_ros manually (git pull in ros_ws/src/crazyflie_ros) - there was a recent fix for support of more than 255 parameters that you'll need with newer firmwares.

sahibdhanjal, 2018-10-18:
I'm running the STM32 firmware and uploaded it to the CFs using the instructions [here](https://wiki.bitcraze.io/doc:crazyflie:dev:starting). For the radio, I followed the instructions on the main readthedocs page [here](https://crazyswarm.readthedocs.io/en/latest/usage.html). It is funny that when I launch ```python chooser.py``` and click on flash(STM), I'm able to successfully flash the CF. But none of the other buttons work (give me a timeout). Even flash (NRF) works fine. Anything wrong with my approach in flashing or any other reason why the other buttons don't work? 

PS - I also pulled the latest ```crazyflie_ros``` and re-flashed everything, but it still doesn't seem to work

Attaching screenshots of successful flashing of the devices for your reference below:
1. ```chooser.py flash```
[image]

2. flashing the CF
[image]

3. flashing the radio
[image]

whoenig, 2018-10-19:
Did you also flash the NRF firmware (either using `make cload` from the crazyflie2-nrf-firmware folder or using `rosrun crazyflie_tools flash --target nrf51 --filename prebuilt/cf2_nrf.bin` in the crazyswarm folder)? I believe the version button is not fully supported with the current version, but the battery check should work with any latest NRF firmware.

sahibdhanjal, 2018-10-19:
Yes, I flashed both the firmwares using ```make cload```. I thought of trying to run crazyswarm without the battery/version button running, but again, I face the same issue where the crazyswarm_server is crashing and throwing a runtime error from ```Crazyflie::handleRequests``` as in the attached image.  The error I get is as follows:
```bash
process[crazyswarm_server-1]: started with pid [13351]
process[rviz-2]: started with pid [13352]
ch: 1
[ INFO] [1539974086.090482944]: Adding CF: cf1 (radio://0/80/2M/E7E7E7E701, cf1)...
[ INFO] [1539974086.104601668]: CF ctor: 0.014038 s
terminate called after throwing an instance of 'std::runtime_error'
  what():  timeout in handling batch
[crazyswarm_server-1] process has died [pid 13351, exit code -6, cmd /home/dasc/Documents/CrazySwarm_WS/ros_ws/devel/lib/crazyswarm/crazyswarm_server __name:=crazyswarm_server __log:=/home/dasc/.ros/log/8f6b5b0e-d3c9-11e8-a0a3-4ccc6aa6bf2d/crazyswarm_server-1.log].
log file: /home/dasc/.ros/log/8f6b5b0e-d3c9-11e8-a0a3-4ccc6aa6bf2d/crazyswarm_server-1*.log
```

> [image]

Is there any other way of getting it to work?

sahibdhanjal, 2018-10-19:
I think the CF I was using is bricked after I ran the above command. Only M2 turns on, when I try to reboot it. I'm not able to access or even flash the CF using ```make cload``` anymore. Anyway how I can recover the CF?

whoenig, 2018-10-20:
For recovery you might need two people:
1. Remove battery & USB cable.
2. Press power button
3. Attach battery (while keep pressing the power button)
4. Keep pressing the power button for 3 seconds, until the blue led starts blinking.
5. Flash (e.g., using flash cload)

If that doesn't work, you'll need the debug adapter and JTAG programmer for recovery.

I am not sure what's going on in your case - are you running on a native installation or is it a VM?

sahibdhanjal, 2018-10-20:
I'm running the whole setup on native Ubuntu-16.04. Would you suggest me trying on a different system? I've currently tried it out on 2 native systems, but can try out on more.
I'll give what you suggested a try. Btw, I can't reboot into bootloader mode at all, and the only way to turn on/off the CF is unplugging/re-plugging the battery. I have a debug adapter, and will try that out. Thanks for that.

sahibdhanjal, 2018-10-20:
Sure, I'll try using the JTAG and flashing it. Will let you know if it works

sahibdhanjal, 2018-10-22:
Hi, I was able to revive the CF that died using the process you mentioned. Thanks for that. I'm really running out of options on how to get this thing running. I'm pointing out the process I follow after cloning and building the repository. Could you please see if anything is wrong in it? I really appreciate you being so patient and helping out!

**Assign IDs to each CF**
1. Assign 4 CFs IDs from ```0xE7E7E7E7{01-04}``` on channel 80 at 2MB datarate

**Upload the STM32 and NRF51 firmware using cload**
2. Goto ```crazyswarm/crazyflie2-nrf-firmware``` and ```git checkout master```
3. ```make``` and then ```make cload```
4. Repeat steps 2,3 for ```crazyswarm/crazyflie_firmware```

**Update the crazyradio firmware**
5. Update the firmware as directed in Option 2.

**Update Config Files**
6. Update ```allCrazyflies.yaml, crazyflieTypes.yaml, and hover_swarm.launch``` as directed
7. Finally source the workspace and run ```hover_swarm.launch```

whoenig, 2018-10-23:
For the STM32 firmware, you don't want to use the master branch, but the "crazyswarm-next" branch. Also, to avoid any issues that might be caused by compiling the firmware, I suggest using the prebuilt firmware binaries (if switching branches doesn't help); they are in the prebuilt folder (see https://github.com/USC-ACTLab/crazyswarm/tree/master/prebuilt).

sahibdhanjal, 2018-10-23:
Switching branches also doesn't help. I was initially uploading the crazyswarm-next branch itself initially. Since it boils down to uploading the prebuilt binaries, is there any way I can do that without using ```rosrun crazyflie_tools flash --target nrf51 --filename prebuilt/cf2_nrf.bin``` and ```rosrun crazyflie_tools flash --target stm32 --filename prebuilt/cf2.bin```? I encounter the problem of bricking the CFs when I use these commands. I can do that via  ```chooser.py``` but even that doesn't help out for the connection.

whoenig, 2018-10-23:
It is actually very worrisome that the crazyflie_tools don't work well for you. Is the link quality good (as shown in the cfclient)? How far away is the CF from the radio when you flash it?

You can flash those binary files using the official python tool chain as well. Use `python3 -m cfloader flash <filename> stm32-fw` for the STM32 firmware or append `nrf51-fw` for the NRF firmware.

sahibdhanjal, 2018-10-29:
The link quality as shown in the cfclient) is nearly full. The CFs are literally kept right beside the radio on the desktop. I even flashed them using the method you suggested and flashed ```cf2.bin``` and ```cf2_nrf.bin```  but it still doesn't seem to work and gets stuck everytime I try to run the ```crazyswarm_server``` from ```hoverswarm.launch```. It times out after approximately ~2 sec.

PS - sorry for the late update, I had my midterms going on

whoenig, 2018-11-03:
We just debugged this issue and it looks like a regression:
* Version 20cee6b10a2e8f5aca3b069df0a2da5ee7b2a63f (add vicon-sdk as submodule; whoenig committed on Jul 28) works fine
* In the master branch, if safelink is disabled, at least NRF-related functions work fine (i.e., battery check)
* Even with safelink disabled, there are issues with getting logging variables/parameters. This seems to be related to the V2 protocol.

ToDo: 
* since Arnaud also reported issues with safelink, this should be disabled by default. (Perhaps have it a runtime option instead of being hard-coded)
* add logging mode to crazyflie_cpp/tools to gather communication data exchanged over radio to debug further.

NicksonYap, 2018-12-17:
@whoenig 

Is the issue solved?

I'm new to Crazyswarm and ROS in general
I've been a minor contributor to Crazyflie firmware repo & was able to fly multiple crazyflies using Loco deck and Bitcraze's Python API

I'm trying to use Loco deck with crazyswarm

I'm using Ubuntu 16.04 VMware

I've updated to the latest Crazyradio firmware: (that commit marked as "Verified")
https://github.com/bitcraze/crazyradio-firmware/commit/d928afd3d34c6d5ce596cdb226a82bc1f72fc56c

and the latest Crazyflie firmware release:  (2018.12)
https://github.com/bitcraze/crazyflie-firmware/tree/2018.12

Chooser.py is not stable, and the errors are of course, from crazyflie_tools

[image]

Getting Battery value works fine for many times, that is until Version is called once (which works), other buttons (Battery, Version itself, sysOff, reboot) will only respond with timeout

Funnily, Flashing will work, and the Crazyflie reboots itself. 
Even after flashing & self reboot, the Reboot button works, but Battery, Version & SysOff responds with an error

I suppose calling Version will cause STM related code to break?

Unplugging and replugging back the radio will reset the condition to allow Battery and others to work again (did not restart crazyflie, did not restart Chooser.py)

whoenig, 2018-12-18:
When you updated to the latest commit, did you also update all the git submodules? What you describe sounds like it might be an issue related to safelink, which seems to be buggy in crazyflie_cpp. The latest crazyflie_cpp disables safelink by default. The latest chooser.py also doesn't have a version button anymore, I believe.

NicksonYap, 2018-12-18:
I see, I'll try to update all submodules, especially the ones from your repo?

whoenig, 2018-12-18:
I believe if you run `git pull` followed by `./build.sh` it should update all submodules as well.

Huzayfaa, 2019-02-26:
Hi,

I believe that I am facing the same/similar issue when running either of the following commands using ROS;

```
cd ros_ws/src/crazyswarm/scripts
python chooser.py
```
or 

```
source ros_ws/devel/setup.bash
roslaunch crazyswarm hover_swarm.launch
```

I have read and understood majority of this existing issue raised and have thereafter implemented the changes I deemed necessary to try and diagnose the problem better and try to get the command(s) to work. However, I have now spent much time on this and can not get it to work. I receive the following issue/error when I run either of the commands (as stated above);

`timeout`

or 

```
[crazyswarm_server-2] process has died [pid 25697, exit code -6, cmd /home/huzayfaa/crazyswarm/ros_ws/devel/lib/crazyswarm/crazyswarm_server __name:=crazyswarm_server __log:=/home/huzayfaa/.ros/log/061f4cf4-39f2-11e9-ad17-e839355c91e0/crazyswarm_server-2.log].
log file: /home/huzayfaa/.ros/log/061f4cf4-39f2-11e9-ad17-e839355c91e0/crazyswarm_server-2*.log
```

I have made the following changes to the configuration files, as stated in the installation guide
(https://crazyswarm.readthedocs.io/en/latest/usage.html) and some of the older issues that are now solved.The process I follow  includes: 

1. Assigned CF2.0 using a USB cable and the cfclient to **uri:=radio://0/100/2M/E7E7E7E701**
(Using the cfclient I can then scan for the CF2.0 on that address and channel and am able to
connect to it too)
2. Upgraded the firmware of the CF2.0 using the cfclient and bootloader mode to release **2019.02.1**
3. Upgraded the firmware of the Crazyradio using option 2 (prebuilt binary) and the version of the Crazyradio
is now **99.55**
4. Ran the ** ./pc_permissions.sh** script and this too has executed successfully
5. My **allCrazyflies.yaml** configuration file reads
```
crazyflies:
  - id: 1
    channel: 100
    initialPosition: [1.5, 1.5, 0.0]
    type: default
```
6. ** crazyflieTypes.yaml** - No changes were made to this configuration file
7. The following changes were made to ** hover_swarm.launch**
```
firmware: "bitcraze"
estimator: 2
motion_capture_type: "none"
object_tracking_type: "motionCapture"
```

The cfclient works absolutely fine and I am able to connect the CF2.0 using the same channel and address whenusing the client. However, both **chooser.py** and **hover_swarm.launch** do not work  and thia I understand may be due to the timeout. 

I understand that this may be a connection issue and that the Crazyradio is unable to connect to the CF. However, I don’t understand what I am doing wrong or even not doing and how to go about fixing this issue. I have two crazyflies and have now tried with both. 

Many thanks in advance, any help and direction would be highly appreciated.

------------------------------------------------------------

p.s. when I run `./build.sh` it all seems to compile fine and reaches 100%, however the following warning message is also displayed four times and I believe this relates to Issue #16 

```
CMake Warning at /opt/ros/kinetic/share/catkin/cmake/catkin_package.cmake:166 (message):
  catkin_package() DEPENDS on '/usr/lib/x86_64-linux-gnu/libusb-1.0.so' but
  neither '/usr/lib/x86_64-linux-gnu/libusb-1.0.so_INCLUDE_DIRS' nor
  '/usr/lib/x86_64-linux-gnu/libusb-1.0.so_LIBRARIES' is defined.
```

Huzayfaa, 2019-03-02:
Hi, 

Can I please ask if anyone has had a chance to acknowledge my previous comment in regards to an issue similar to this one. 

Do I have to raise it as a new issue? or is what I have done correct

I'd appreciate any help and look forward to hearing back from someone, thanks 

Huzayfaa

whoenig, 2019-03-04:
Hi Huzayfaa,
Sorry for the delay. I think it would be better to have this as a new issue, since you have your own full description on things that you tried already. If needed I can always refer to other issues. I created #137 on your behalf.

jpreiss, 2021-01-30:
Closing due to sprawl. Please start a new issue and reference this one if necessary.

closed 2021-01-30 by jpreiss