USC-ACTLab/crazyswarm #106: cf cannot take off

astridtepie, 2018-10-15:
Hello everyone!
Recently I had some problems when launching the hover_swarm file. I can visualize the cf on rviz but the only command that work with the controller is emergency(). Also
1.I have update and convergence warning 
[image]
2.The advanced flight option python figure8_csv.py works only in simulation but not while the server is running.
[image]
Your help is really appreciated, thanks in advance!

whoenig, 2018-10-16:
For 1: Does the tracking actually work (in rviz)? It looks like you use the libobjecttracker tracking mode, but the marker configuration doesn't match the config file. This warning is actually critical if you see it several times in a row.

For 2: You need to use python2 - it seems that it tries to use python3 instead.

astridtepie, 2018-10-16:
1. I can see on rviz that the point cloud has an offset with the origin of the cf1 axis, but when moving the cf, cf1 also moves on rviz
[image]

whoenig, 2018-10-17:
Did you adjust the marker configuration in crazyflieTypes.yaml according to your setup? Perhaps they don't match, which can cause the ICP to fail in regular intervals.

astridtepie, 2018-10-19:
Yes I have adjusted the markers configuration using the following command
roslaunch crazyswarm mocap_helper.launch
However I still get the same problem,
[image]
Now I can see takeoff! and Land! on the terminal when pressing the appropriate keys on the xbox360 controller, but the cf would not execute any of those actions.
I have failed to mention that the Vicon system is actually operated on another computer, connection with the Vicon computer is set-up via wifi (The Vicon computer creates a host network.)

whoenig, 2018-10-20:
What was the output of `roslaunch crazyswarm mocap_helper.launch` and how did you adjust the crazyflieTypes.yaml file based on that output?

Do you have the LED ring? If so, which color does it show?

Which crazyflie_ros tutorial are you referring to? Which mocap system are you using?

astridtepie, 2018-10-22:
Dear sir,
1.The output was a list of the coordinates of 4 points, I copy pasted them in the file.
2.No sir, I do not have the LED ring
3.The tutorial I am refering to is https://github.com/whoenig/crazyflie_ros/blob/master/README.md
4. The motion capture I am using is the Vicon system and like stated before it is connected to another computer and the bridge connection is via wifi.
Thanks in advance.

astridtepie, 2018-10-22:
Dear sir,
1.I have the coordinates of a list of points, I simply copy paste them in the file.
2.I do not have a led ring
3. It is not really a tutorial, but I followed the step from this github https://github.com/whoenig/crazyflie_ros/blob/master/README.md
4. I m using the Vicon motion capture system on Nexus. The Vicon system is operated on another computer and the bridge connection is made via wifi.

Thanks in advance.

whoenig, 2018-10-22:
I suggest that for testing, you use the tracking system in VICON Nexus (i.e., create a rigid object there) to see if the issue is related to the tracking or to software/firmware issues. Also, I suggest switching from a wifi connection to a cable connection to avoid latency issues.

astridtepie, 2018-10-24:
Hello,
When using the Tracker system by creating an object cf1(motionCapture), nothing is shown and there is a warning saying that the cf1 object is not updated and nothing is shown on rviz.
[image]
I tried to reupload the marker configuration and now there s no more offset with the cf reference frame.

[image]
But there is still the 

-low correspondance
-ICP did not converge
-Dynamic checkout failed 

And the cf still do not respond to the controller.

Not directly related to tutorial (if possible)
I also have a problem with setting up the network ethernet connection. Indeed,
the Vicon system is connected via an ethernet cable to the Vicon pc. The computer workstation is also connected on another ethernet port (connection tested with ping and successful for both ). By doing that, I do not get any information from the Vicon. I suspect there is no connection between the two ethernet cable despise being on the same computer. I would like to know if possible how to get them to communicate properly.

Thanks in advance!

whoenig, 2018-10-25:
First problem: This sounds like the object is not enabled in Vicon Tracker/Nexus? If you move the CF around, does the object named "cf1" move in Tracker/Nexus?

Objecttracking issue: lets first make sure that the vicon tracking works, then we can revisit the objecttracking again.

Ethernet issue: This is the normal setup (the vicon pc needs two network cards for best performance). Does ping work with ip address or hostname? Did you try giving the crazyswarm the ip address instead of the hostname of your vicon PC? You might also be able to get help on this from your schools IT department, since there could be firewall issues as well.

astridtepie, 2018-10-25:
Dear sir,
1.By moving the cf, the object cf1 also moves on Nexus
2.ping worked with the ip address and was done on both  computers.

astridtepie, 2018-10-29:
Dear sir, 
I have managed to successfully set up the ethernet connection between the vicon computer and my computer however the same error persist. I have also failed to mention that when connecting with motionCapture option,
1. On rviz it is said that no world frame was found and
2.Impossible to make the transformation between world frame and baselink

[image]

astridtepie, 2018-10-30:
Dear sir, 

0.Below are photos of how the nexus was setup : 
[image]
![20181030_135809](https://user-images.githuWhen using the Vicon bridge it actually works, that is:
1.When the object named cf1 is tracked it appears on nexus and the terminal is as shown below : 
[image]
2.When the object is not properly seen, then the following is seen : 
[image]
3. The tf tree is as shown : 
[frames.pdf](https://github.com/USC-ACTLab/crazyswarm/files/2529687/frames.pdf)busercontent.com/43744147/47720059-47ccd980-dc4d-11e8-8562-c3c7f56e686b.jpg)
[image]
[image]
It looks like there are multiple markers, although there are just single.
[image]

When using the Vicon bridge it actually seems to work, that is:
1.When the object named cf1 is tracked it appears on nexus and the terminal is as shown below : 
[image]
2.When the object is not properly seen, then the following is seen : 
[image]
3. The tf tree is as shown : 
[frames.pdf](https://github.com/USC-ACTLab/crazyswarm/files/2529687/frames.pdf)

astridtepie, 2018-10-30:
1.Do the above results of the Vicon bridge connection seem correct?
2.hover_vicon.launch  in the crazyflie_ros package was tested and still there was nothing shown on rviz
[image]

3. The following command hover_swarm.launch was tested using motionCapture, nothing shown on rviz except cf1 not updated like before 
Is there any issue with vicon connection?

Thanks in advance

whoenig, 2018-10-31:
1. I never used Nexus (we used Tracker instead). In Tracker, the name is "vicon/cf1/cf1", i.e., I am wondering where "cff" comes from?
2. This is strange. Did you look at the tf_frames while hover_vicon was running?
3. I think this is related to the "cff" versus "cf1" issue and not to a connection issue. 
4. I don't think the object_tracker file is used in this mode.

glennliu, 2018-10-31:
I think I meet the same problems.

In my case, I use Optitrack(ethernet connection works well). And I followed the steps,
1. firmware flashed fine.
2. cfclient works fine (control cf through xbox joy).
3. chooser.py works fine.
4. mocap_helper.launch works fine.(copy the marker config to .yaml file)
5. hover_swarm.launch, display cf position correct on rviz. **Use xbox to takeoff, no response.** 

Screenshot as below, Any advice appreciated.

[image]

[image]

whoenig, 2018-11-02:
@glennliu Do you have the LED ring? If yes, what color does it show?

glennliu, 2018-11-03:
> @glennliu Do you have the LED ring? If yes, what color does it show?

Unfortunately, I don't have the ring.
Also, the LED4 on cf doesn't turn green. In `roslaunch crazyflie_demo teleop_xbox360.launch`, the LED4 is fast blink in green and cf can takeoff by xbox360 controlled.

I was think is it could be a communication problem?

Thanks

whoenig, 2018-11-04:
The communication in the Crazyswarm is very different from the one used in the crazyflie_demo (Crazyswarm uses broadcasts; the demo uses bidirectional links). The LEDs therefore cannot be used to indicate the communication status. The LED ring is the easiest way to diagnose, but one could also use the uSD card for logging or the JTAG adapter to check if the communication works fine.

glennliu, 2018-11-05:
@whoenig 
I can see tf data on rostopic. So this data should come from mocap alone without imu data?

glennliu, 2018-11-05:
@whoenig 
I just realized it's probably because I flashed crazyflie firmware from `bitcraze/crazyflie_firmware`
 instead of `USC-ACTLab/crazyswarm`. 
I'll flash cf firmware from your git and try again.

Thanks

pete9936, 2019-04-04:
@glennliu Did flashing the cf firmware from the USC-ACTLab/crazyswarm work? I believe I am encountering a similar issue where I updated the firmware from bitcraze yesterday and I get an initial motor command and then it cuts out (when I press start from my xbox controller the motors turn on for a brief moment then shut off).

glennliu, 2019-04-04:
@pete9936 No, I didn't make it succeed. I reflashed the firmware and still cannot achieve stable hovering.

jpreiss, 2020-06-13:
@glennliu @astridtepie were you ever able to resolve this?

jpreiss, 2021-01-30:
closing due to inactivity.

closed 2021-01-30 by jpreiss