USC-ACTLab/crazyswarm #107: Problems using more crazyradios (6+)

The-SS, 2018-10-16:
Hello

In a [previous issue](https://github.com/USC-ACTLab/crazyswarm/issues/92#issuecomment-423003531) you suggested I use a crazyradio for every 4 crazyflies (or something similar) when sending full_state commands. 

I have seven crazyradios, so I was trying to use them all simultaneously. Here is how I set them up:
zero-based index: 60-70-80-90-100-110-120 respectively (each radio would control one cf)
channel is unchanged (100)
rate is 2M
I connected 5 Crazyradio PA and 2 Crazyradio through SuperSpeed USB 3.0 hubs.
My computer has a Xeon E3 processor with 8 cores (16 threads).

I can control at most any 5 of them reliably at a time. When I start using the 6th (or 7th) radio, 5 of the crazyflies work normally, but the rest either do not takeoff or fail to hover and crash. Note than all of them appear in RVIZ

I do not know why that is happening. Is it because I'm mixing Crazyradio and Crazyradio PA? Or is there some other limitation that I am unaware of?

Thank you. 
Sleiman

whoenig, 2018-10-17:
What do you mean by "zero-based index"? Each radio should operate on a separate channel. If you use the 2M datarate, the channels need to be spaced apart by 2 MHz. Each CF should have a unique address. You can freely mix Crazyradio and Crazyradio PA - they are essentially identical apart from a different power strength.

The-SS, 2018-10-17:
Sorry, I think I messed up in my description. What I meant to say is that the 60-70-...-120 are the channels. The zero-based index is zero for all of them. e.g.: 0/60/E7...E701, 0/70/E7...E702, ...

whoenig, 2018-10-17:
This should work and I am not aware of a limitations (there is a maximum number of supported radios, but it will warn you when you exceed the number). Did you check if the channels are working well? 110 and 120 usually don't work very well compared to 80.

whoenig, 2018-10-17:
Do you have the LED ring to see how the communication link look likes (red = lots of packet losses; green otherwise)? Another idea would be to distribute the radios over more physical USB ports - each hub will limit the overall bandwidth.

The-SS, 2018-10-17:
I don't have the LED rings. I only have 4 physical USB ports and the radios are divided over them. USB 3.0 should have a bandwidth over 500 megabytes per second, so I think it is unlikely for it to be the limit.

whoenig, 2018-10-17:
The radios are just USB 2.0. Also, they don't use the burst-mode for high-bandwidth transport (each packet requires a separate USB request to be sent). We only tested with up to 3 radios so far. Do some of the drones simply not get the takeoff command, or do they takeoff and then become unstable? What is the physical separation between the drones?

The-SS, 2018-10-17:
The drones are lined up with 0.5m separation. Some drones might not take off. When they occasionally take off, they are unstable and end up crashing. Does this mean I'm limited by the USB port I have?

whoenig, 2018-10-19:
With 4 physical ports you should be totally fine - that essentially means only up to 2 radios per port? 0.5m separation should also be fine, as long as there are no walls very close-by (which can worsen the effect). Up to how many drones work well for you? Or in other words, what is the smallest system size where you do see the issues arise?

ataffanel, 2018-10-19:
I have never used so many Crazyradio at the same time but I can offer some insight about how USB2 full speed (12Mbits, the sped Crazyradio is using) works since you might start to need to care about that with so many radios: Unfortunately when using USB3 ports, the USB full speed is not using the USB3 bandwidth, it is using the USB2 high speed (480Mbit/s) lines. So you when using Crazyradio you only use the USB2 part of your ports.

To pass from 480Mbit/s to 12Mbit/s USB hubs use a "transaction translator" (TT), some hub have a single TT and so you will have only 12Mbit/s shared for all the port, some are multi TT with one TT per port and this means that you will have 12Mbit/s per port.  If you use a multi TT hub you should have no problem supporting many Crazyradio on it, a single TT hub will cause problems.

When using a USB hub you can check if it is single or multi TT with lsusb, first find out the VID:PID of your hub with lsusb and then:
```
$ lsusb -d <vid:pid> -v | grep TT
Couldn't open device, some information will be missing
  bDeviceProtocol         1 Single TT
```

Different ports of a computer should have full independent bandwidth but it is not guaranteed, it depends how the manufacturer has cabled things on the motherboard.

The-SS, 2018-10-19:
To answer @whoenig, I was actually using 4 crazyradio on one USB hub, 2 connected directly, and one connected with the keyboard/mouse/controller USB hub. We do not have any walls close nearby. The closest objects are about 1.5m away from the drones. With that crazyradio configuration I was only able to control 5 drones reliably (each crazyradio is assigned one crazyflie only). I have not experimented with more crazyflies per radio yet.
Regarding @ataffanel's post, thank you for the clarification. I did what you recommended and got the following:
```
bDeviceProtocol         2 TT per port
    bInterfaceProtocol      1 Single TT
    bInterfaceProtocol      2 TT per port
```
Does that mean that each USB hub has 2 TTs?

Here are some videos for my experiments:
[7 drone fails](https://youtu.be/1Y6f5jF44WI)
[5 drone success](https://youtu.be/l5VxIFxLBhg)
[5 drone fail](https://youtu.be/4lTl7JohaaE)
[4 drone fail](https://youtu.be/KJI-qYN2FFs)
[6 drone success](https://youtu.be/tkupHUok5ms)

The experiments have different radio placements. Those are clarified in the description of each videos. 

The only conclusions I could reach are that:
- I cannot go beyond 5 in my current configuration (I don't know what will happen when I have multiple crazyflies over one radio)
- The USB radio connections to the physical USB ports on the computer makes a difference.

whoenig, 2018-10-20:
From the videos, I don't think your issues are radio related. We have tested up to 16 drones per radio without much issues. What I am mainly concerned with is that the controller performance is really bad in your videos. This most likely means that either your motion capture system does not track very well, or that your overall latency is really high.

For the motion capture system: Which system are you using? Do you let the mocap do the tracking or do you use libobjecttracker? Make sure that if you put the CFs on the ground, that the state estimate is stable, otherwise there is an issue.

For latency: Avoid using a VM. Make sure that the mocap computer has a physical ethernet connection with the Ubuntu machine that runs the Crazyswarm. There is also a way to measure the total latency (described in the Crazyswarm paper), but it currently requires a JTAG adapter and some firmware changes.

The-SS, 2018-10-20:
I don't think that we have a detection problem. We've hovered 28 drones before on three radios and we've run other experiments with up to 16 drones (3 radios).
I am using VICON Nexus and libobjecttracker. The VICON detects the markers and they are sent over to the main computer. All markers appear well and there are no problems with ICP. 
I am not using VM, I'm running Linux natively.

With the same setup, I get better performance with 3 radios than when I increase the number of radios.

whoenig, 2018-10-23:
I have no idea what this could be. If you have time, you could try with Ubuntu 18.04, which comes with a newer kernel/lib-usb and see if that changes anything. I don't think we ever tested with more than 3 radios ourselves, but I don't see a particular reason why the performance should suddenly drop.

The-SS, 2018-10-23:
I'm actually running Ubuntu 18.04. When I use 5 drones over 3 radios, things operate reasonably well. When I use 5 drones with 5 radios, the operation of the drones becomes much worse. 
I am investigating the issue. I'll update this thread once I have an idea about what might be going wrong.
Thanks.

jpreiss, 2020-06-13:
@The-SS for the sake of future users who might find this issue, were you able to resolve this?

The-SS, 2020-06-13:
I haven't used Crazyswarm in a while, so new commits might have resolved the issue. Personally, I did not find a solution. 
My observations were:
- at most 3 radios and using `goto`  function, controlling 20+ Crazyflies was fine
- with more than 3 radios, even with a small number of Crazyflies we sometimes got failures.

I speculate that it might have to do with USB ports being dependent. Two potential solutions might be:
1- using [more than one machine](http://wiki.ros.org/ROS/Tutorials/MultipleMachines) each limited to 3 radios (set up one ros master). This will require some modifications but could expand the platform capabilities.
2- using a USB-PCIe card with independent ports [e.g.](https://www.amazon.com/StarTech-com-Express-SuperSpeed-Dedicated-Channels/dp/B00HJZEA2S)/[e.g.](https://www.startech.com/Cards-Adapters/USB-3.0/Cards/PCI-Express-USB-3-Card-4-Dedicated-Channels-4-Port~PEXUSB3S44V).

closed 2020-06-13 by The-SS