USC-ACTLab/crazyswarm #108: INFO about crazyswarm LPS and ROS

lorenzol92, 2018-10-17:
Hi, 
I'm using ROS kinetic on ubuntu 16.04 (on a VM).
I would like to drive a mini swarm of 4 CF's, for the localization I'm using the LPS with 6 anchors in TdOA mode.
Do the crazyswarm demos work with this setup? Are any changes in code required? 
I was using the official directory (https://github.com/bitcraze/lps-ros) but I had problem when using .launch with more than 2 CF's.
Thank you very much for the help.

whoenig, 2018-10-18:
This is supported - simply change the motioncapture to "none" here: https://github.com/USC-ACTLab/crazyswarm/blob/master/ros_ws/src/crazyswarm/launch/hover_swarm.launch#L41.

lorenzol92, 2018-10-18:
Hi whoenig, 
Thanks for quick reply!
So What I need to do is:
1)set the anchors position through the official pc client (I don't see any .yaml file that does this)
2)change the motioncapture to "none" 
3)set the correct address of the CF's in /crazyflies.yaml
and then start the hover_swarm.launch in ROS?
I have some difficulty in understanding:
1)what to put on   broadcast_address: "FFE7E7E7E7"
2)what to uncomment If I want to use the EKF 
3)for update the crazyflie and crazy radio PA firmware is mandatory to do the procedure described there: https://crazyswarm.readthedocs.io/en/latest/usage.html
or it is sufficient the official one: https://www.bitcraze.io/getting-started-with-the-crazyflie-2-0/

Thank you very much for your patience and help.

whoenig, 2018-10-18:
Your approach seems correct. From what I remember, the anchor positions are just parameters? If that's the case you can extend https://github.com/USC-ACTLab/crazyswarm/blob/master/ros_ws/src/crazyswarm/launch/hover_swarm.launch#L22-L32 to include those parameters and your values. This will be uploaded automatically after connection to each CF.

I suggest using the latest official firmware (from github, not the latest release) and changing https://github.com/USC-ACTLab/crazyswarm/blob/master/ros_ws/src/crazyswarm/launch/hover_swarm.launch#L10 to "bitcraze" and https://github.com/USC-ACTLab/crazyswarm/blob/master/ros_ws/src/crazyswarm/launch/hover_swarm.launch#L25 to "2". You will need to update the crazyradio firmware, though.

ataffanel, 2018-10-19:
A note about the LPS: the anchor position is now set in the anchor themselves, not by parameter in the Crazylfie anymore. The anchor position should be set using the Crazyflie client but once that is done there is nothing more to configure in Crazyflie and so in noting Crazyswarm as well.

Generally, if the Crazyflie can fly position-hold from the Crazyflie client, it will be able to fly using Crazyswarm with motioncapture set to "none" pretty much out of the box.

lorenzol92, 2018-10-19:
Hi ,
Thanks a lot for reply .
Next week I will make some trials!
Only few things unclear :
1)using Crazyswarm with motioncapture set to "none" I need to add some of the nodes of the dwm_loc_ekf_hover.launch (https://github.com/bitcraze/lps-ros/blob/master/launch/dwm_loc_ekf_hover.launch) in order to know the CF position or the CF automatically recognize his position in the space ?
2)to use the figure8.csv the origin of the system where is supposed to be? In the center or could be in a corner of the flight space? (actually I have 1 anchor set in the origin of the system)
3)to execute a waypoint following is sufficient to change the figure8.csv with the waypoints.csv ?

Thanks a lot for the patience and the precious help.

whoenig, 2018-10-19:
1) You should not need to use anything in lps-ros (in fact, this repo is not useful for the newer TdoAv3). CFs will compute their position on-board.
2) The trajectory will be shifted to your current location, so you can execute it anywhere in the space.
3) No, figure8.csv contains a trajectory, not waypoints. For waypoints you can either first generate a trajectory or use goTo commands.

lorenzol92, 2018-10-23:
Hi,
Thanks a lot for the tips.
Today I’ ve tried to simply takeoff the CF with the hover_swarm.launch and LPS with 6 anchors in TdOA2.
Before to do this I had:
1)set the motioncapture to “none “ , the firmware to “bitcraze” and the estimator to “2” as suggest
2)update the CF at the latest official firmware (posted few days ago): 2018.10 following Option 1 described at: https://crazyswarm.readthedocs.io/en/latest/usage.html
3)update the Crazyradio PA with 99.55 following Option 1 described at: https://crazyswarm.readthedocs.io/en/latest/usage.html
4)update the lps anchors with the latest firmware 2018.10.dfu using the apposite tool

I have some questions/issues:
1)When I launch the hover_swarm in ROS I have no error but no LEDs on the CF nor on the crazyradio blinks ( as the CF is not connected to the radio ) is it normal?
2)When I press the start button on joystick(xbox360) the CF takes off and when I press back it lands but the big issue is that if I press the Emergency button the motors don’t shut off (even if on ROS seems that Emergency service is executed correctly)
3)Is it normal that in Rviz the CF doesn’t appear?

Thanks a lot fot the patience and the precious help!

whoenig, 2018-10-23:
1. That is normal - everything will run on-board. data will only be exchanged if you send a command (such as "takeoff"). If you use a mocap, the difference is that the mocap data is frequently sent as well (for UWB this is not needed).
2. I think this is a bug. For the mocap system, the emergency button simply meant to not send mocap data anymore. The firmware initiated the actual emergency shut-off after ~500ms. For the UWB case we would need to send some kind of emergency indication to the drone.
3. Yes - there is no data streaming. One possibility would be to enable logging, log the current state estimate, and sent this to rviz. This hasn't been implemented so far (and you would need a good number of radios for a high update rate.)

lorenzol92, 2018-10-23:
Hi whoenig ,
Thank you very much !
Do you think that this bug could be solved ? 
Any idea/tip on how to do this ?

Thanks a lot for the help

lorenzol92, 2018-10-25:
Hi whoenig,
Thanks a lot for all these info!
I think that the best approach is to Ping CFs, this need only to be activated ( in the bitcraze firmware ) or require to be implemented? In this case what is the main “file” to change/modifiy ?
I prefer, in particular for the first trials, to have an emergency functionality that can be activated via joystick, for the moment I have only 4 CFs (not a large swarm).
Thank you very much for patience and help!

P.S : It is better if I open another “issue” on GitHub relating the Emergency stop or I can continue this thread?

NicksonYap, 2018-12-17:
This whole issue should be turned into a guide!

I think the number of LPS users getting interested in Crazyswarm is growing, so am I!

The lack of implementation or description of how the results should be, kept making me think I did something wrong.

Turns out a lot of stuff is not finalised for LPS yet.

whoenig, 2018-12-18:
I agree! If you like to see that as part of the Crazyswarm documention, pull requests are very welcome (see https://github.com/USC-ACTLab/crazyswarm/tree/master/docs for the "source" files for the Crazyswarm documentation.) The bitcraze wiki also allows external contributors.

NicksonYap, 2018-12-18:
Yep, would definitely update the docs and submit PR for it (after i understand how it works, lol)

I have an xbox360 controller connected but other than pressing the start button to take off, I couldn't land it (pressing the reverse start button) or navigate around
It just takes off and stays there

I tried running figure8.py and from what I recall, it also takes off only, and won't move after that.

The way to stop the drone is by CTRL+C then it just drops.

I was using 2018.12 crazyflie firmware and I recall I was using "bitcraze", estimator "2"

Though I must note that I'm placing the lower anchors 43cm away from ground.
When crazyflie placed on the floor, it would think it is 10-20cm away from ground

So it may confuse any landing algorithm
But it lands well when using CFclient & xbox  controller to land

Did I miss out something?
Is the Crazyflie supposed to a be able to navigate around in Rviz?

timsblade, 2019-08-06:
Hi,

I'm currently working on this as well and I followed the steps mentioned above, however I receive these errors:

ERROR: cannot launch node of type [crazyswarm/crazyswarm_server]: can't locate node [crazyswarm_server] in package [crazyswarm]
ERROR: cannot launch node of type [crazyswarm/crazyswarm_teleop]: can't locate node [crazyswarm_teleop] in package [crazyswarm]

When I checked the packages, the specified nodes are available. Is there a step that I missed?

jpreiss, 2020-06-13:
Is anyone willing to convert this information into a PR for the Crazyswarm documentation? If not, I will close this issue. I'm not thrilled about having important information buried in closed issues, but I also don't want to scare away potential users by having tons of old open issues.

easy98, 2020-12-05:
Hi,

Is there any way I can log the crazyflie's position directly?

easy98, 2020-12-07:
> A note about the LPS: the anchor position is now set in the anchor themselves, not by parameter in the Crazylfie anymore. The anchor position should be set using the Crazyflie client but once that [...]

Hi, how can I fly position-hold from the Crazyflie Client? Do I need to set any desired position manually? Thanks.

jpreiss, 2020-12-07:
@easy98 this repo is for crazyswarm-specific questions. Please ask your question in the bitcraze crazyflie-client issues or the bitcraze forums.

jpreiss, 2021-01-30:
Asking again if anyone can convert this into documentation :)

sabdi1, 2021-03-05:
Hello, I am using Crazyswarm on Ubuntu 18.04 with Ros Melodic and an 8 anchor LPS in system and I'm having some trouble retrieving the state estimate messages. I tried to use the goto function to test with 1 cf in TWR mode, and the cf hovers, but does not go to the goal point. I printed out the cmdPositionMsg and cmdFullStateMsg.pose and both were all zeros. I wanted to look at the state estimates to understand where the cf thinks it is and am having trouble moving forward from here. Any help would be appriciated.

whoenig, 2021-03-06:
@sabdi1 Please open a new [discussion](https://github.com/USC-ACTLab/crazyswarm/discussions) instead of using this issue that is about writing improved documentation. You may also find https://github.com/USC-ACTLab/crazyswarm/discussions/339 interesting.

closed 2021-10-05 by jpreiss