USC-ACTLab/crazyswarm #110: Crazyswarm testing failed

bconvens, 2018-10-30:
Hi,

I am new to ROS/crazyflie and I am trying to test the capabilities of the crazyswarm code, but I am having some problems with the tutorial and have questions about stuff that are not 100% clear to me. 

My setup: I have a vicon nexus system running on a windows PC. The latter is connected with an Ethernet cable to my laptop running ubuntu 16.04 and ROS kinetic. Currently I have only 1 crazyflie 2.0, but I would like to test for a swarm later.
I have already been able to teleoperate the crazyflie 2.0 using an xbox controller via the cfclient before I installed crazyswarm.

1) First of all after building the crazyswarm code with `./build.sh.` Is there anything else not explained in the tutorial I need to do or run in the terminal? I also executed `~/crazyswarm/ros_ws/build$ make`. Anything else I need to make or do?
2) I assigned `0xE7E7E7E701` to cf1. I assign cf1 to channel 100, and not to 80 as in your tutorial. Is this a problem?
3) The crazyflie firmware is upgraded by doing `crazyflie-firmware$ make cload`.
[image]
4) I tried to update the firmware of your Crazyradios but none of the methods works. Below are some pictures of the error I get with option 2. Why is none of these commands working?
[image]
[image]
[image]
5) When I put `rosrun crazyflie_tools scan` in the terminal I get something different than what is expected from the tutorial (it should give `Found Crazyradio with version 99.55`). See the picture below what I obtain. What can be the reason for this?
[image]
6) the original ros_ws/src/crazyswarm/launch/allCrazyflies.yaml file looks like:
 crazyflies:
```
  - id: 1
    channel: 100
    initialPosition: [1.5, 1.5, 0.0]
    type: default
  - id: 40
    channel: 100
    initialPosition: [-1.0, -0.5, 0.0]
    type: default
```
Is it a problem to keep id 40 even if I only test for 1 cf?
7) the ros_ws/src/crazyswarm/launch/crazyflieTypes.yaml file I did not change. Is this ok?
8) In ros_ws/src/crazyswarm/launch/hover_swarm.launch I changed the following lines:
`motion_capture_type: "vicon" # one of none,vicon,optitrack,qualisys
 object_tracking_type: "motionCapture" # one of motionCapture,libobjecttracker
 vicon_host_name: "134.184.20.148"`
I did not change the default `broadcast_address: "FFE7E7E7E7"`, is this correct?
9) When trying to monitor the swarm I first do `~/crazyswarm/ros_ws/devel$ source setup.bash`, but then get an error when I click on battery (same for other buttons): 
[image]
[image]
In the tutorial is stated that it the tool reads the ros_ws/src/crazyswarm/launch/all49.yaml file, but I don't have this file in the folder, is this normal?
10) Basic flight. How can I run `crazyflie_server`? I found it in /crazyswarm/ros_ws/devel/lib/crazyflie_driver, but don't know how to run it. When I execute `source ros_ws/devel/setup.bash
roslaunch crazyswarm hover_swarm.launch` I get the following error:
`[ WARN] [1540915958.700318395]: No updated pose for motion capture object cf1
[ WARN] [1540915958.706759654]: No updated pose for motion capture object cf1
[ WARN] [1540915958.716307725]: No updated pose for motion capture object cf1`
What can be the reason?
11) For the vicon sdk is it SDK (version 1.7.1) from http://www.vicon.com as specified in the readme file or is it possible to use a more recent SDK version? And how should the folder tree exactly look? Is it correctly shown in #51? Since it is very confusing with the default folder tree that seems to not have a externalDependencies folder in the crazyflie_ros folder, but it has one a folder layer higher as a subfolder of src. Could you please show your tree?
12) Note that when testing the vicon_bridge this actually seems to work:
[image]
Only when we occlude markers on purpose we get these warnings:
[image]
Is this the result I should get for a working vicon bridge?

Thanks a lot for the help!

whoenig, 2018-10-31:
1. build.sh should be fine. It runs catkin_make as well.
2. No, that's fine. Just provide the correct channel in allCrazyflies.yaml
4. You have to use python(2). For that version, you seem to be missing usb. Try `sudo apt install python-usb`
5. The output changed in newer version. You can still use `--verbose` to get more detailed output as before.
6. You can keep whatever CFs you want. But you should use chooser.py to enable/disable the ones you want to fly.
7. Yes, not need to change if you fly a standard CF2.0.
8. This is correct. Broadcast_address is actually not used (anymore).
9. Try `sudo apt install python-yaml`. all49.yaml has been renamed to allCrazyflies.yaml. In which part of the documentation did you still see all49.yaml?
10. There seems to be an issue with using Nexus (we tested with Tracker). Essentially, the object should show up as "cf1/cf1" and not "cf1/cff". I don't have Nexus to test. Do you guys have Tracker as well?
11. The latest version comes with the SDK included (no need to manually install anything).
12. See 10.

Note that 10/12 seems the problem in issue #106 as well. So if you figure out where that "cff" comes from, that would be very helpful!

bconvens, 2018-10-31:
The numbers below correspond the the numbers of my questions:
3. Is your answer 3 to my my question 3 or 4? In my question/point 3 I wanted to ask if the shown result when I do make cload is correct? I connect the crazyradio PA, put crazyflie in bootload mode and just type make cload in the terminal and it automatically chooses python3 (see the line python3 -m cfloader flash cf2.bin stm-fw). So I don’t see how I can change this to python 2.
[image]
4. I assume that your answer 3 is to my question 4. I had to indeed install 1 new package by sudo apt install python-usb. I plug in the crazyradio PA, execute python2 crazyradio-firmware/usbtools/launchBootloader.py and then I execute sudo python2 crazyradio-firmware/usbtools/nrfbootload.py flash prebuilt/cradio.bin. 
[image]
What is the reason python 3 is not working and how can I see that this must be executed with python 2?
What’s the reason why it only works with sudo?
5. When unplugging and replugging the crazyradio PA and activating the crazyflie 2.0 I get (without and with connecting it also with usb cable):
[image]
Is crazyradio version 99.55 the last version?
What is meant with “version 2” in “Found crazyflie connected via USB cable with version 2”?
9.  After sudo apt install python-yaml I get the same error. The radio is plugged in and crazyflie is activated. Unplugging and replugging does not help.  Any idea why it keeps complaining about yaml? 
[image]
[image]
[image]
[image]
[image]
In the tutorial section Usage/Monitor Swarm is stated that this tool  (the crazyflie chooser gui) reads the ros_ws/src/crazyswarm/launch/all49.yaml file, but I don't have this file in the folder (because it is renamed probably). So maybe this can be changed in the tutorial.
10/12. The names can be changed in Nexus too. If I name the subject (equivalent of an object in tracker) and the segment both to "cf1" so that it shows up as cf1/cf1. 
[image]
[image]
Unfortunately, I still get the warning that the pose of cf1 is not updated.  In rviz I still don't see anything. What should I get there? But, when I press start/back on the xbox controller, the crazyflie takes off and lands (although not staying at a fixed altitude).
[image]
Is it normal that as long as 9. is not solved, I will have problems with this step or is there another cause?
Do you still think it is because of nexus vs tracker software?
Unfortunately I don't have tracker to test it now, but I will ask a trial license. On your tutorial I see it is working with tracker 3.4, but are newer versions of tracker also working?

Thanks!

whoenig, 2018-11-03:
3. Yes your usage of `make cload` was correct.
4. The script is old and was written for python2 and python3. I don't think there is a way to find out, apart from looking at the error message. You need sudo because the radio goes into a different mode after you start the bootloader. This mode had a different device/vendor id, which you probably didn't include in your udev-rule.
5. 99 just means you compiled yourself. I believe 55 is the latest version. I don't know what "version 2" exactly means - that's just the version query we get back from the CF.
9. chooser.py actually executes some other scripts with python3, so try `sudo apt install python3-yaml`. Note that we use different python versions here because ROS only supports python2, while we usually prefer python3 for new development.

You should see the frame of the CF in rviz if the connection works. Nexus should work equally well as Tracker, especially since you figured out how to change the segment name there. We are working with the latest Tracker (3.7 or so). Perhaps the easiest would be to debug the issue by adding some print outs here: https://github.com/USC-ACTLab/libmotioncapture/blob/master/src/vicon.cpp#L83-L105.

bconvens, 2018-11-03:
Hi,

Thanks for the quick responses. Here is an update.
1) Chooser.py
`sudo apt install python3-yaml` and `pip3 install pyyaml` seemed to solve my problems with chooser.py
If I execute the chooser.py I obtain the following results when pressing the buttons in the GUI.
[image]
[image]
The timeout occurs when the crazyflie is in power save mode after clicking sysOff button. As explain in your tutorial, at that moment the version button does not work and I think it gives therefore timeout error.
Can you please confirm if this is the expected result?
Should I execute python `chooser.py`, `python2 chooser.py` or `python3 chooser.py`.
Since up to know it only works with `chooser.py`, `python2 chooser.py`?
When doing `python3 chooser.py` I still get error referred to Tkinter.
[image]

2) Basic flight
I closed the chooser.py gui, while running this next line. Is this possible or should I always leave the chooser GUI opened?
If I test without being plugged to a vicon PC, I get the following results when I chronologically press takeoff, land and the emergency button on the xbox joystick and finally manually stop it:
[image] 
[image]
[image]
[image]
I don't see anything in rviz, neither I get an error (which seems normal according the next lines).
In the crazyswarm tutorial documentation is stated: 
> If there is an error (such as a faulty configuration or a turned-off Crazyflie) an error message will be shown and the application exits. If there is a problem in the communication between the motion capture system and the Crazyswarm server, the application will not exit but the positions of the Crazyflies will not appear in rviz.

When the application is running and I shut down the crazyflie 2.0, the application does not exit in my case, nothing new is displayed in the terminal. After it is turned off, I can even still execute takeoff and land commands and they appear in the terminal.
If the crazyflie was turned off before I execute the hover_swarm.launch I get:
[image]
So according to what is expected from the tutorial documentation this is already a problem. Any idea why this is?

Thanks for helping me out!

bconvens, 2018-11-05:
1. I tried to install as in https://stackoverflow.com/questions/4783810/install-tkinter-for-python. Unfortunately, it does not help.
[image]
[image]
Any idea why it works for python(2) and not for python 3 even if I have it installed for python3?

How can I send using the crazyswarm Python API https://crazyswarm.readthedocs.io/en/latest/api.html  position,yaw commands to multiple cfs using just 1 radio? It seems like if I use `goTo(self, goal, yaw, duration, groupMask = 0)` that all crazyflies would move the same relative distance (if they start from different initial conditions). So I would like to send position commands using 1 radio so that cf1 goes to (2m,1m,1m), cf2 goes to (3m,0m,1m) and cf3 goes to 1m 2m,2m)?
How can I do this? I think it has to to with the `groupMask`, but I am not sure.

I will soon test the code with the vicon system in the loop. Hopefully it works now! :-)
Thanks!

whoenig, 2018-11-06:
I think in python3 the module is called "tkinter" (i.e., lower-case). Nevertheless, if it works with python2, there is no strong reason to use python3 for this script.

The goTo command can be called on an individual CF object (see https://crazyswarm.readthedocs.io/en/latest/api.html#individual-crazyflie). You can simply make 3 function calls, the crazyswarm will just use one radio if the CFs share the same channel. Note that this won't be synchronous. If you want perfect synchronous execution, you will need to use relative commands (and broadcasts), or compute the trajectories offline, upload the trajectories while the CFs are on the ground, and then broadcast a startTrajectory() command.

bconvens, 2018-11-06:
I tested with vicon in the loop, but unfortunately it still does not work.
The terminal output below is what I get if I don't press start on the xbox controller.
If I press start the cf take off in a very weird, unstable way and crashes (or I stop it before with the xbox controller). I get the same warning as before. I never see any frame of the cf in rviz. 
`brybuntu@brybuntu-ThinkPad-W541:~/crazyswarm/ros_ws$ roslaunch crazyswarm hover_swarm.launch
... logging to /home/brybuntu/.ros/log/cb3d3440-e1ac-11e8-bea6-54ee754e3928/roslaunch-brybuntu-ThinkPad-W541-19803.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://brybuntu-ThinkPad-W541:33803/

SUMMARY
========

PARAMETERS
 * /crazyflieTypes/default/batteryVoltageWarning: 3.8
 * /crazyflieTypes/default/batteryVoltateCritical: 3.7
 * /crazyflieTypes/default/bigQuad: False
 * /crazyflieTypes/default/dynamicsConfiguration: 0
 * /crazyflieTypes/default/firmwareParams/ctrlMel/i_range_m_z: 1500
 * /crazyflieTypes/default/firmwareParams/ctrlMel/i_range_xy: 2.0
 * /crazyflieTypes/default/firmwareParams/ctrlMel/i_range_z: 0.4
 * /crazyflieTypes/default/firmwareParams/ctrlMel/kR_xy: 70000
 * /crazyflieTypes/default/firmwareParams/ctrlMel/kR_z: 60000
 * /crazyflieTypes/default/firmwareParams/ctrlMel/kd_omega_rp: 200
 * /crazyflieTypes/default/firmwareParams/ctrlMel/kd_xy: 0.2
 * /crazyflieTypes/default/firmwareParams/ctrlMel/kd_z: 0.4
 * /crazyflieTypes/default/firmwareParams/ctrlMel/ki_m_z: 500
 * /crazyflieTypes/default/firmwareParams/ctrlMel/ki_xy: 0.05
 * /crazyflieTypes/default/firmwareParams/ctrlMel/ki_z: 0.05
 * /crazyflieTypes/default/firmwareParams/ctrlMel/kp_xy: 0.4
 * /crazyflieTypes/default/firmwareParams/ctrlMel/kp_z: 1.25
 * /crazyflieTypes/default/firmwareParams/ctrlMel/kw_xy: 20000
 * /crazyflieTypes/default/firmwareParams/ctrlMel/kw_z: 12000
 * /crazyflieTypes/default/firmwareParams/ctrlMel/mass: 0.032
 * /crazyflieTypes/default/firmwareParams/ctrlMel/massThrust: 132000
 * /crazyflieTypes/default/markerConfiguration: 0
 * /crazyflieTypes/large/batteryVoltageWarning: 11.4
 * /crazyflieTypes/large/batteryVoltateCritical: 11.1
 * /crazyflieTypes/large/bigQuad: True
 * /crazyflieTypes/large/dynamicsConfiguration: 0
 * /crazyflieTypes/large/firmwareParams/ctrlMel/i_range_m_z: 1500
 * /crazyflieTypes/large/firmwareParams/ctrlMel/i_range_xy: 2.0
 * /crazyflieTypes/large/firmwareParams/ctrlMel/i_range_z: 4.0
 * /crazyflieTypes/large/firmwareParams/ctrlMel/kR_xy: 30000
 * /crazyflieTypes/large/firmwareParams/ctrlMel/kR_z: 40000
 * /crazyflieTypes/large/firmwareParams/ctrlMel/kd_omega_rp: 100
 * /crazyflieTypes/large/firmwareParams/ctrlMel/kd_xy: 3.0
 * /crazyflieTypes/large/firmwareParams/ctrlMel/kd_z: 3.0
 * /crazyflieTypes/large/firmwareParams/ctrlMel/ki_m_z: 500
 * /crazyflieTypes/large/firmwareParams/ctrlMel/ki_xy: 1.0
 * /crazyflieTypes/large/firmwareParams/ctrlMel/ki_z: 2.0
 * /crazyflieTypes/large/firmwareParams/ctrlMel/kp_xy: 8.0
 * /crazyflieTypes/large/firmwareParams/ctrlMel/kp_z: 8.0
 * /crazyflieTypes/large/firmwareParams/ctrlMel/kw_xy: 9000
 * /crazyflieTypes/large/firmwareParams/ctrlMel/kw_z: 10000
 * /crazyflieTypes/large/firmwareParams/ctrlMel/mass: 0.38
 * /crazyflieTypes/large/firmwareParams/ctrlMel/massThrust: 5400
 * /crazyflieTypes/large/markerConfiguration: 2
 * /crazyflieTypes/medium/batteryVoltageWarning: 7.6
 * /crazyflieTypes/medium/batteryVoltateCritical: 7.4
 * /crazyflieTypes/medium/bigQuad: True
 * /crazyflieTypes/medium/dynamicsConfiguration: 0
 * /crazyflieTypes/medium/firmwareParams/ctrlMel/i_range_m_z: 1500
 * /crazyflieTypes/medium/firmwareParams/ctrlMel/i_range_xy: 2.0
 * /crazyflieTypes/medium/firmwareParams/ctrlMel/i_range_z: 2.0
 * /crazyflieTypes/medium/firmwareParams/ctrlMel/kR_xy: 50000
 * /crazyflieTypes/medium/firmwareParams/ctrlMel/kR_z: 60000
 * /crazyflieTypes/medium/firmwareParams/ctrlMel/kd_omega_rp: 200
 * /crazyflieTypes/medium/firmwareParams/ctrlMel/kd_xy: 1.0
[... 279 lines cut ...]
[ WARN] [1541499293.193262934]: No updated pose for motion capture object cf1
[ WARN] [1541499293.201109206]: No updated pose for motion capture object cf1
[ WARN] [1541499293.210853978]: No updated pose for motion capture object cf1
[ WARN] [1541499293.222205336]: No updated pose for motion capture object cf1
^C[ WARN] [1541499293.230770617]: No updated pose for motion capture object cf1
[ WARN] [1541499293.240572793]: No updated pose for motion capture object cf1
[ WARN] [1541499293.250696113]: No updated pose for motion capture object cf1
[ WARN] [1541499293.260388330]: No updated pose for motion capture object cf1
[ WARN] [1541499293.271405146]: No updated pose for motion capture object cf1
[ WARN] [1541499293.280256254]: No updated pose for motion capture object cf1
[ WARN] [1541499293.290714460]: No updated pose for motion capture object cf1
[ WARN] [1541499293.302606254]: No updated pose for motion capture object cf1
[ WARN] [1541499293.311783715]: No updated pose for motion capture object cf1
[rviz-5] killing on exit
[crazyswarm_teleop-4] killing on exit
[joy-3] killing on exit
[ WARN] [1541499293.321987550]: No updated pose for motion capture object cf1
[crazyswarm_server-2] killing on exit
[ WARN] [1541499293.330503624]: No updated pose for motion capture object cf1
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done
`

whoenig, 2018-11-07:
Did you check with vicon_bridge, that you actually get vicon/cf1/cf1 now as object?

bconvens, 2018-11-08:
Hi!

Thanks for the help! I have some good news. I checked the vicon_bridge and it appears as vicon/cf1/cf1. Using crazyswarm the object tracking seems to be working in rviz. I don't get the warnings anymore, unless I occlude the cf1. :+1: 
Taking off does not work since the world frame axis and object/segment axis of cf1 do not correspond in orientation/colors. The world frame orientation/colors are the same in rviz and nexus. I tried rotating the cf1s axis in Nexus, but I am forced in Nexus to first define the blue (z-axis) and then the y axis of the cf1. But when placing the markers on the cf1 as in your tutorial, I cannot select the z-axis, since on the cf1 there is no line connected by 2 markers that is orthogonal to the plane of the drone available. 
[image]
[image]

Do you have any idea how I can overcome? Is this also a problem in tracker? Where in your code could I rotate the axis of the cf1 90° over its own y axis?

To be sure: RGB is always x, y z?

Thanks!

closed 2020-06-13 by jpreiss