USC-ACTLab/crazyswarm #111: Using CrazySwarm for Reinforcement Learning

arya-k, 2018-11-01:
Hi,

I love the work you've been doing with Crazyswarm - I have a couple of questions regarding whether it can be used for RL tasks:

1. In your [demo video on youtube](https://youtu.be/D0CrjoYDt9w?t=75), for 'interactivity', you show someone walking through the space, holding a wand, and the drones moving to avoid it. How would I get the python library to recognize the wand as an object and track it? I'm guessing you're giving the drones new GoTo commands if they are too close to the wand - is that correct?

2. Regarding your simulator, how accurate is it? Does it simply display trajectories (no matter how unrealistic), or does it take into account the physical constraints of the crazyflies?

3. If I wanted to train RL algorithms (we've been evaluating DDPG and PPO), we'd ideally have an OpenAI Gym-style environment, where we can manually step the simulator by 1ms and train the NNs, since our scripts would be a significant bottleneck. Does the simulator have support for this lock-step style of programming? If not, how complex of a task would it be to implement something of the sort?

This is a sample of lock-step style programming:
```import gym
env = gym.make("Taxi-v1")
observation = env.reset()
for _ in range(1000):
  action = env.action_space.sample() # your agent here (this takes random actions)
  observation, reward, done, info = env.step(action)
```
Thanks a lot for the help!

pete9936, 2018-11-02:
I have a very similar question as 1.) and don't think it's worth creating a New Issue for. We are attempting to track a ground vehicle using vicon and want to have the crazyflies interact with that ground vehicle (i.e. land on it). I noticed in Issue #53 you suggested using _libobjecttracker_ to take in both the cfs and a human. I was wondering if there is any way to use the _motioncapture_ object tracking type to perform this? 

Also like arya-k said, would you suggest using GoTo commands once you have that positioning data for the human/object or use a method similar to Issue #54? Since I have attempted to use the method form that issue but am having problems with it (I made a comment on that issue as well with a lot more detail). Though I'm wondering if that can be solved with clarification of my above questions?

Thanks again for all the great work on this project!

whoenig, 2018-11-02:
I actually think it would be easier to have separate issues for different questions.

@arya-k 

1. For that demo video we actually use a form of artificial potential algorithm that runs on-board the CFs. The basestation is only sending the position of obstacle to each drone, the new goal point is computed on-board (the exact equation is given in the Crazyswarm paper).

2. Our simulator just displays trajectories - there is no simulation of any kind of noise or dynamic limits.

3. I don't think there is a big benefit of using our simulator in a gym environment. The simulator was simply developed to test high-level python scripts before executing them on the swarm. It's probably not hard to develop lock-step for this simulator, but I don't think the simulator provides things that are actually useful in an RL setting.

@pete9936 

1. Yes, you can use motioncapture and just assign any name to your ground vehicle. The Crazyswarm will just consider objects of the form "cf<id>". For your ground vehicle, you can use VRPN or vicon_bridge to get the current pose.

2. GoTo commands should work in that case (and I'll take another look at #54).

arya-k, 2018-11-02:
Hi @whoenig thank you for the prompt feedback!

I have a couple followup questions:

1. While the simulator does not model the dynamic limits of the drones, do the python libraries ever check the feasibility of a trajectory? So if I were to send an impossible trajectory to the drone, would it attempt to follow it, or will the libraries catch/discard the commands before they are sent to the drones?

2. I'm considering building my own simulator or integrating with an existing one (probably Gazebo). Do you have the dynamic constraints of Crazyflies modeled anywhere in the code, or would I have to gather this information on my own?

3. If I wanted to track a ground vehicle with _libobjecttracker_, could I add the wand as a markerConfiguration to `crazyflieTypes.yaml` and track it from there? Or would I have to use _motioncapture_ and `vicon_bridge`?

Thank you!

whoenig, 2018-11-03:
1. We have functions that can find max a and w, but we currently do not check before execution if the trajectory is feasible (neither in simulation nor on the HW). We also didn't spend too much time to push the CF controller to its limit, since we do not work on aggressive maneuvers. 

2. We don't have those limits, and they depend on how many you fly or other external disturbances and the controller you use. There are papers that have all parameters required for simulation (such as inertia and motor parameters), and there are a few projects that attempted to integrate that into simulations (e.g., RotorS, CrazyS).

3. At the moment you would probably just track the ground robot using VICON and then use vicon_bridge (or ros_vrpn_client). It is possible to add it to crazyflieTypes, but the current code assumes that all vehicles are CFs and we try to connect to each robot - it's easy to change, but only worth the effort if you can't use the vicon object tracking for some reason.

closed 2019-11-12 by whoenig