USC-ACTLab/crazyswarm #114: Implementation of another stabilizing controller and navigation layer

bconvens, 2018-11-05:
Hi,

I would like to discuss how to start with the development of another stabilizing controller and a navigation layer for the crazyflie 2.0 using the crazyswarm code (with vicon system) to start from. Currently I just focus on 1 cf, but it must be later easily expanded to multiple cfs.

The **navigation** layer (based on potential fields) takes as input the cf pose set-point [r_x,r_y,r_z,r_yaw] send from the ground PC, cf state measurements of the cf centre of mass [x,y,z,roll,pitch,yaw, dotx,doty,dotz,dotroll,dotpitch,dotyaw] from Vicon/onboard sensors, the 4 motor forces [f1,f2,f3,f4], the state of external possibly moving obstacles measured by vicon [x_obs,y_obs,z_obs,roll_obs,pitch_obs,yaw_obs, dotx_obs,doty_obs,dotz_obs,dotroll_obs,dotpitch_obs,dotyaw_obs] (or any other attitude and rate representation). With this information and some defined navigation parameters/gains, it calculates the position set-point for the cf stabilizing controller, i.e. [v_x,v_y,v_z,v_yaw].

The **stabilizing controller** is a nonlinear cascade inner-outer loop controller. It basically takes as input [v_x,v_y,v_z,v_yaw] (=output of navigation layer), the cf state measurements [x,y,z,roll,pitch,yaw, dotx,doty,dotz,dotroll,dotpitch,dotyaw] (or any other attitude and rate representation) and calculates based on some controller parameters/gains the 4 motor forces [f1,f2,f3,f4]. The rate of the attitude loop is about 10 times faster than the position loop. The rate of the navigation layer is the same as the position control loop.

I found the existing controllers, estimators and commanders/navigators in this folder .../crazyswarm/crazyflie-firmware/src/modules/src, but I am a bit overwhelmed by the number of files. It is not clear to me which are used in the crazyswarm tutorial and which can be ignored.
I am basically searching for an estimator/controller/commander example that already satisfies most of my needs and where I can identify the following variables with the names in the code: 
- [r_x,r_y,r_z,r_yaw]
- [x,y,z,roll,pitch,yaw, dotx,doty,dotz,dotroll,dotpitch,dotyaw]. _Please show both the raw measurements and filtered/estimated values. Is there any advice you can give in the type of estimator to use? Also, the controller needs the center of mass position, so is there a part of the code where you translate a marker position to the actual center of mass position used for the control?_
- [f1,f2,f3,f4]. _Please indicate where I can see the mapping from actuator forces in Newton (calculated by the controller) to PWM values send to the motor driver. Do you also know where I can find the motor limits (min and max force or PWM) in the code?_
- [x_obs,y_obs,z_obs,roll_obs,pitch_obs,yaw_obs, dotx_obs,doty_obs,dotz_obs,dotroll_obs,dotpitch_obs,dotyaw_obs]. I think the obstacle is similar to the wand pose in one of the videos.

_Could you please when answering the above questions make references to the relevant parts of the existing code so it is more clear which of the many files are used and how these variables are called in the existing code, and how the information flow proceeds?_

_After I changed the code, if I want to compile this new firmware, what should I do so that the firmware upload buttons in the chooser.py gui compile this new firmware (using either option 1 or option 2)?_ 

_Is there already implemented a way to save the logged data and convert it to matlab plots or to excel?_

Thanks!

whoenig, 2018-11-06:
You might get better answers related to the firmware development in the official forum.

Navigation layer: We evaluate trajectories on-board in our "high-level-commander". I think it would be possible to add other commanders. A good starting point is here: https://github.com/bitcraze/crazyflie-firmware/blob/master/src/modules/src/crtp_commander_high_level.c#L210.

Controller: We already have a nonlinear controller, which is here: https://github.com/bitcraze/crazyflie-firmware/blob/master/src/modules/src/controller_mellinger.c. It does output PWM values instead of newtons (although some people have tried to rectify that in the past). The actual output PWM values are computed here: https://github.com/bitcraze/crazyflie-firmware/blob/master/src/modules/src/controller_mellinger.c.

State estimate: The state estimate is with respect to the center of mass. You can find it here: https://github.com/bitcraze/crazyflie-firmware/blob/master/src/modules/src/estimator_kalman.c. For the obstacle, we just send the position of the wand, no higher order states.

I suggest that you try read the code and understand the current implementation using the pointers I provided (rather than me telling you exact variable names).

For flashing, just use `make flash` in the firmware folder. For logging, I suggest using the uSD-card deck, which has a wiki page explaining how to get and plot the data (in Python, but you could import to matlab if you want to).

whoenig, 2018-12-07:
Closing. Please open a new issue if there are additional questions.

closed 2018-12-07 by whoenig