USC-ACTLab/crazyswarm #115: crazyflie2.0 takeoff flipped under Optitrack

glennliu, 2018-11-06:
Hi 

I was running crazyswarm demo starting from one crazyflie. I followed the crazyswarm wiki and bitcraze wiki closely. And `roslaunch crazyswarm hover.launch`, and sent takeoff command through `/cf1/takeoff`. However, the drone flipped every time in takeoff. 

**Preparation**
1. `roslaunch crazyswarm mocap_helper.launch`  and copy pose of the 4 markers to `crazyflieTypes.yaml`.
2. set Z axis as the up axis in Optitrack.
3. cf front points to Optitrack x axis direction.
4. manually move cf and tf in rviz follows correctly.
[image]

I made several guess for possible reasons.
**Guess 1: coordinator config**
`mocap_helper.launch` gives different marker pose every time I placed the cf. Do I have to copy and paste it to `crazyflieTypes.yaml` every time? 
When I manual move cf, rviz looks all correct. But there are warning messages showed,
[image]
At takeoff,
[image]
I think the crazyswarm_server cannot recognize cf pose once it takeoff. But not sure why this happens.

**Guess 2: hovering point at takeoff**
Do I have to set target hovering position? Maybe its because the hovering position is horizontally different from the drone position. So it move horizontal at takeoff and flipped. 

**Guess 3: takeoff parameter**
In `/cf1/takeoff`, there are height and duration parameters. 
`height=0.3, duration=5', it flipped.
'height=0.05, duration = 3.5', rotor spins a bit and drone doesn't move at all.
Is it can caused by takeoff parameters? Also, there are other paramters in crazyflie_controller, which are default values and I didn't change at all.

Thanks

whoenig, 2018-11-08:
Your approach seems correct to me. The warnings you get are worrisome - there should be no warnings during normal execution. The fitness value mismatch indicates that the relative positions of your markers change. If they are rigidly attached, this can only be caused by issues with your motion capture system.

I suggest trying to use the OptiTrack motion capture mode (and a single CF) first, in order to test the system. This avoid debugging libobjecttracker for now.

For your guesses: The firmware determines the exact position automatically, based on the current position and the given height. Your duration is a bit long. try height=0.5, and duration=2 (or 3). If you have an xbox controller, you can also simply press the "start" button (and "back" to land).

glennliu, 2018-11-08:
Hi @whoenig 

I just tested it with Optitrack "motionCapture"  function. 

_hover_swarm.launch_ configured as below,
```
motion_capture_type: "optitrack" # one of none,vicon,optitrack,qualisys
object_tracking_type: "motionCapture" # one of motionCapture,libobjecttracker
vicon_host_name: "vicon"
optitrack_local_ip: 192.168.1.166 # client ip
optitrack_server_ip: 192.168.1.168 # motive ip
```

Rigid body named `cf1`.

However, there is warning about cf1 data update
[image]

I'm using Optitrack actually. But it shows `VICON Latency high`.
Besides, I got steady `/tf` data update. And when I move crazyflie manually, the rviz update pose correctly. 

Even so, I tried takeoff through xbox joystick and the drone doesn't response. Only `Takeoff!` printed on terminal. Next, I called service `/cf1/takeoff` with height =0.5 and duration = 3. The rotor spins a bit and stopped. The drone doesn't move at all.

I see similar error at #110  . But I have rviz update correctly.

whoenig, 2018-11-14:
If the LED is red, the takeoff command shouldn't do anything (because of safety mechanisms in the firmware). Which (STM32) firmware did you flash exactly?

glennliu, 2018-11-14:
I clone the code from crazyswarm master branch and update all its submodules.
> crazyflie-firmware branch: head detached 0858430
> crazyflie2-nrf-firmware: 6331295

Is it the right branch?

glennliu, 2018-11-15:
Hi @whoenig 

I switched to "master" branch for `"USC-ACTLab/crazyflie-firmware"`. When `make`, it shows
> make[1]: *** No rule to make target 'src/utils/src/outlierFilter.c', needed by 'outlierFilter.o'.  Stop.
> Makefile:327: recipe for target 'build' failed
> make: *** [build] Error 2

Do you know which exact branch for "crazyflie-firmware" and "crazyflie2-nrf-firmware" should use?

Thanks

glennliu, 2018-11-15:
Hi @whoenig 

I flashed prebuild firmware, the drone flipped as usual. I think I should decouple the problem. The LED ring is solid red once `hover_swarm.launch` launched. Do you know what's the possible reason to cause LED ring red?

My radio works fine with cfclient and chooser.py. I use firmware from `/bitcraze/crazyradio-firmware` with "master" branch.

glennliu, 2018-11-20:
Hi @whoenig 

I did more tests. Currently, there are two bugs showed before takeoff
1. warning about cf pose
> [ WARN] [1542724085.265473544]: No updated pose for motion capture object cf1
> NATNET FRAME TIMEOUT
> [ WARN] [1542724085.275612429]: No updated pose for motion capture object cf1
> NATNET FRAME TIMEOUT
> [ WARN] [1542724085.285749708]: No updated pose for motion capture object cf1
> NatNet reconnected after 0.0589902 sec
[image]

It kept shows above warning at terminal. However, there is cf1 pose update from rviz

2. LED ring kept red
Placing drone on the ground, the LED kept red.
And strange thing is that if I hold cf on my hand in the air. The LED turned green.
Also, if I kept call `/takeoff` service or `/land` service, LED blink between green and red. When I stop sending, the LED is back to solid red.

Could you please give me some advice on solving this?

Thanks

glennliu, 2018-11-24:
The warning messages,
` NATNET FRAME TIMEOUT`
came from NatNet, which means your PC cannot receive data from Motive correct. In my case, I use Motive to publish rigid body at 120Hz. But `/tf` messages I received on my PC is at 80Hz.

Strange thing is that I also tried `mocap_optitrack` to run on my PC (https://github.com/ros-drivers/mocap_optitrack). It can receive rigid body data steadily, which is exactly at 120Hz. Also, I tried NatNet sample program downloaded from Optitrack website. The NatNet sample program doesn't have this TIMEOUT issues. 

So, I think it is probably something wrong with my crazyswarm code.

Unfortunately, I didn't find what's in crazyswarm caused this problem yet.

Mic542, 2018-11-26:
Hello.

Is there any update on this issue?
I got NATNET FRAME TIMEOUT in mocap helper as well as hover launch.

I tried using ros vrpn client and sent broadcast vrpn from motive before and I can get the data there.

I am using motive 1.10.2.

whoenig-overleaf, 2018-11-27:
It might be worthwhile to reach out to the OptiTrack support. Basically the issue is that we use a library that was developed a few years back using reverse engineering to match the Motive protocol. Things have changed over time and it seems (from the bug reports; we do not have access to an optitrack system) that there are compatibility issues. I talked to a OptiTrack representative at a conference, and he said that they have internally a Linux C++ library. Unfortunately, he was not willing to share it with me as a non-customer (but he said they might share the library with their customers upon request). What I would need to improve the support is this library, and the ability to put the library code on github (using whatever license they prefer for their code, of course).

If you are just looking into tracking a few CFs using the OptiTrack object tracker, we could also implement VRPN in libmotioncapture, which should solve most problems related to that mode.

The last option would be to port the code from mocap_optitrack (which seems to be another third-party implementation) to libmotioncapture.

glennliu, 2018-11-27:
Hi @whoenig-overleaf 

It seems implement VRPN/mocap_optitrack rather than NatNet is the easiest way. 
Are you going to update the changed code?
Or can I change it by myself? I'll read through the code in libmotioncapture to see if I can modify it.

Thanks

Mic542, 2018-11-27:
Hi.

Okay, I will try contact the Optitrack support team asking them about what you've said.
Also the possibility of putting the library code on Github.

whoenig, 2018-11-27:
I'd prefer if you can change the code.

1. For VRPN, you can look at vrpn_client_ros (see http://wiki.ros.org/vrpn_client_ros). 
2. For mocap_optitrack, take a look at their code.

In both cases, you need to implement the following interface: https://github.com/USC-ACTLab/libmotioncapture/blob/master/include/libmotioncapture/motioncapture.h#L90-L126. For VRPN, pointclouds and latencies are simply not supported. For mocap_optitrack, it should be technically possible to implement all 3 functions, although the pointcloud one would be the most important one.

glennliu, 2018-11-27:
Ok. I'll let you guys know my result after the test. @whoenig @Mic542

Mic542, 2018-11-27:
Thanks! I will inform you guys if OptiTrack support reply to my support ticket.

Thank you very much!

Mic542, 2018-11-28:
Hi, I got a reply from Optitrack team.

I am gonna quote what he said here:

"Thanks for using OptiTrack and for reaching out to us regarding these requests.
     Unfortunately, we do not have a direct Linux solution to provide for this application. CrazySwarm appears to work with the data streamed from various motion capture systems, which means you would need a version of our NatNet SDK for data streaming clients. 

     While the NatNet SDK does not directly support Linux, we do provide two samples that demonstrate how to connect to Motive and depacket the data without the use of our libraries. This process is sensitive to any changes we make to the data stream format, so if going between Motive versions you will need to download the latest samples and update the protocol in your client.
          https://v21.wiki.optitrack.com/index.php?title=NatNet_SDK_3.1#Direct_Depacketization

     If the CrazySwarm application is running in Linux and connecting to Motive, it likely uses this same direct depacketization method. If the code is provided, it may be possible to update the bit format so that the data is correctly decoded again. Otherwise, you may need to determine the version of Motive they developed off of and use that version.

I hope this information helps."

What do you all thinks?

whoenig, 2018-11-29:
That's too bad (and not entirely in line what the representative told me). I guess the easiest way forward would be to take the code from https://github.com/ros-drivers/mocap_optitrack. They recently switched to NatNet3, which, according to your reports, doesn't have that many issues.

glennliu, 2018-11-29:
But `mocap_optitrack` only offers data for rigid body, rather than pointcloud. Then, we can't use `mocap_optitrack` with `libobjecttracker`, can we?

glennliu, 2018-11-30:
Hi guys

I tried to modify `crasyswarm_server` code to work with `mocap_optitrack`. But there is conflict with `crazyswarm_server` and `mocap_optitrack` run together. Because they both trying to read mocap data stream from the same port and the program would just crash.

Besides, there is a large part in `crazyswarm_server` used lib from `optitrack.h`. It's hart to modify it for `mocap_optitrack. Considering I'm not that familiar with `crazyswarm_server`, I probably won't try.

But I was thinking can I change my Motive version? I was using Motive 1.10.0, while the wiki said its better to use with Motive 1.10.3.

glennliu, 2018-11-30:
Alright, it turns out our Optitrack License cannot support Motive 1.10.3 or higher. I have to figure out some ways else.

I'm looking at NatNet 2.10.0 from Optitrack(https://optitrack.com/downloads/developer-tools.html#natnet-sdk). It looks like I can also substitute this NatNet with the one in `crazyswarm`. The interface should be at /externalDependencies/include/libmotioncapture/src/optitrack.cpp

glennliu, 2018-11-30:
I think I made a mistake about coordinator frame before.

When building rigid body, crazyflie should point to x direction of **rviz** frame. But I was pointing it to x direction of **Motive** frame. After I changed it, I manually move crazyflie and the result on rviz is correct now.

But it still flipped at takeoff. So, it goes back to the problem with NatNet. NatNet interface issues cause `crazyswarm_server` only receive mocap data at 80Hz with delays insteady of 120 Hz.

paradox56, 2018-11-30:
I am currently trying to get Motive 2.0 work with crazyswarm. The default setup in my lab is Optitrack + Motive 2. So the only option I have is either modify crazyswarm so it works with moca or VRPN.

paradox56, 2018-11-30:
> Hi guys
> 
> I tried to modify `crasyswarm_server` code to work with `mocap_optitrack`. But there is conflict with `crazyswarm_server` and `mocap_optitrack` run together. Because they both trying to read mocap data stream from the same port and the program would just crash.
> 
> Besides, there is a large part in `crazyswarm_server` used lib from `optitrack.h`. And it used function like `Latency`,`PointCloud` that `mocap_optitrack` do not have these interfaces. It's hard to modify all of them for `mocap_optitrack`. Considering I'm not that familiar with `crazyswarm_server` , I probably won't try with `mocap_optitrack` anymore.
> 
> But I was thinking can I change my Motive version? I was using Motive 1.10.0, while the wiki said its better to use with Motive 1.10.3.

Have you tried to modify any codes in `libmotioncapture/src/optitrack.cpp`? The `optitrack.cpp` file is using NatNetLinux in back-end. I think the correct way to incorporate mocap_optitrack is to make changes in there.

glennliu, 2018-12-01:
Hi @paradox56 

It looks like `optitrack.cpp` have interface for getLatency, getPointcloud, which `vrpn` doesn't have. Does it work with libobjecttracker as well?

glennliu, 2018-12-01:
let's close this issues and discuss it in a new issue

whoenig, 2018-12-07:
Closing this issue, since we have other open issues (#121 and $122) that deal with VRPN and mocap_optitrack, respectively.

closed 2018-12-07 by whoenig