USC-ACTLab/crazyswarm #117: The problem of  synchronize of steering engine

roseyanpeng, 2018-11-11:
Hi, when I sent the takeoff command to the crazy, I found that there was a steering engine that started to work two seconds after the others started. Can anyone tell me why? Thank you.

whoenig, 2018-11-13:
What do you mean by "steering engine"?

roseyanpeng, 2018-11-13:
I mean the motor @whoenig .

whoenig, 2018-11-14:
It is a single Crazyflie, but one of the motors only starts with a two second delay? If that's the case, I have never seen such behavior.

roseyanpeng, 2018-11-14:
Four of the ten crazyflie have this phenomenon, not strictly delayed for two seconds, but the asynchronous start of their motors is obvious. Another question I want to ask is that every time you fly a  swarm crazyflies, whether they can successfully complete the fly mission, or occasionally one or two will be uncontrolled and fall down. Thank you.

The-SS, 2018-11-14:
I have never noticed any asynchronous behavior with the motors before, but I have had cases where a drone (or multiple drones) would suddenly drop. Regardless of whether I have 3-4 drones per radio or 13 on a single radio, a (a few) drones would just drop for no reason. 
At one point I noticed that these droppings increase when I would send `goto` commands with zero time duration (`goto(pos, 0, 0)`).
Other than that, I have also noticed that using certain USB ports on my machine gave more reliable performance. This might be related to how the USB port are connected on the motherboard.
There might be some problems with radio communications. (I've had a few cases where a command sent to one drone (say cf1 on radio 100) would cause another drone (cf2 on radio 100) to operate for only a split second).

roseyanpeng, 2018-11-15:
I am similar to what you have encountered. When more than three drones take off, one or two will drop randomly. I haven't found the reason until now. Like you, I also think there might be some problems with radio communications. Have you found a better solution?

The-SS, 2018-11-15:
In my case, having drones drop is not very frequent. It happens once in a while, but I am usually able to run my experiments. I have performed experiments with up to 24 drones over three radios. I try to distribute them as evenly as possible across the radios (about 8 drones/radio), and so far things are ok. Every once in a while I might have a case where 3-4 drones drop (out of 20-24 drones), but when I restart, everything operates normally more or less. 
One thing to note is that in my case when multiple drones drop, they usually drop sequentially (one after the other).

roseyanpeng, 2018-11-15:
Thank you for your advice. I will share my progress in time.

glennliu, 2018-11-15:
Hi @roseyanpeng 

I have similar drop case. May I know which branch for "crazyflie-firmware" and "crazyflie2-nrf-firmware" do you use?

Thanks

whoenig, 2019-11-12:
This is a pretty old thread. Closing - feel free to open a new issue if your problems persist with the latest version.

closed 2019-11-12 by whoenig