USC-ACTLab/crazyswarm #120: Problem with chooser.py

Mic542, 2018-11-21:
Hello, guys.
I am new at this crazyswarm.
So I am trying to flash the cf using the chooser.py but it always failed here is the screenshot:
[image]
I already flash the radio using the pre-built as the documentation said.

and this is the result of crazyflies_tools scan:
[image]

and this is my cf connected using cfclient at the same address E7E7E7E7E7.
[image]

Everything i try in chooser.py return timeout and when I try hover_swamp.launch the crazyswamp_server terminated with reason timeout.
Please help me solve this problem..

Thank you in advance.

glennliu, 2018-11-21:
Hi 

I met similar problems at the beginning and solved it by checking these,
1. check file `allCrazyflies.yaml` and `crazyflies.yaml`. Make sure the radio channel is correct.
2. check crazyradio device permission is opened. As you can use "scan" function, I think you already achieved this one.

BW

Mic542, 2018-11-21:
Hi,

this is my crazyflies.yaml:
[image]
The channel is already at 80.

and this is my allCrazyflies.yaml:
[image]
it is also with channel 80.

glennliu, 2018-11-21:
Hi 

Can you try these?
```
channel: 100
id:1
bandwidth: 2M
```
In using cfclient, you drone is connected correctly?

Mic542, 2018-11-21:
[image]
[image]
Is that right way to do it?
and I still failed when I try to flash it using chooser.py.
Still the same error, here is the screenshot:
[image]

glennliu, 2018-11-21:
Have you write the changed data into crazyflie? 
You can connect you cf through cfclient and write in channel and bandwidth you just changed. It should be able to work once your drone config and your .yaml are synced.
[image]

Mic542, 2018-11-21:
Hi.
As you can see here:
[image]

My cfclient that I have doesn't have any menu other than flight control.
Where can I get that?

Thank you very much for your response anyways.

glennliu, 2018-11-21:
I launched my cfclient via `sudo ./bin/cfclient`. You may also try that.
As you can see on your cfclient upper left, you are in channel:80 and bandwidth:250K. This is your currently real setting for radio, which is been wrote into your EEPROM on drone. **You need to change the setting and rewrote into drone.**
It can be changed through upper menu /connect/Configure 2.0, which you can see on my previous screenshot.

Mic542, 2018-11-21:
So you are using the one inside bin folder that come with this repository, right?
Maybe I need to launch it with sudo.
My cfclient right now does not have upper menu though.
Well, I will try it as you said tomorrow. I will post the update of this too.

Hopefully this can solve my problem too.

As always thank you very much.

glennliu, 2018-11-21:
yep, I launch it with in sudo. I'm not sure is it right. But it work very well in my PC.

whoenig, 2018-11-21:
Great to see users helping each other:-) If you need sudo, that likely means that you didn't deploy the udev-rules for the Crazyradio.

Mic542, 2018-11-22:
After I did it like @glennliu has said. Yes! I can flash it from chooser.py. However there is this timeout that keep appearing after sometimes.
[image]
As the above image, I was able to get the battery and version but after that everything I clicked resulted in time out or just empty. 

Anyways thank you very much @glennliu at least I can flash it using chooser.

glennliu, 2018-11-22:
You can try look ros_drivers folder: https://github.com/whoenig/crazyflie_tools/tree/0662e65b60b7577ca7fbb8118d687ba2ca679bda/src

At each file(battery.cpp, reboot.cpp), change its radio parameters to your radio setting. And it should be worked after that.

Mic542, 2018-11-22:
Even after doing that I still get timeout every so often and i need to re plug my radio to make it work again.

Mic542, 2018-11-27:
It seem this issue is resolved. As @glennliu said I need to launch cfclient from bin folder using sudo command, change the address of the cf, and the chooser.py worked. it just every so often the timeout occured and I just need to unplug and plug the radio again. Closing this issue.

closed 2018-11-27 by Mic542