USC-ACTLab/crazyswarm #121: Motive 2.0 with VRPN streaming

paradox56, 2018-11-29:
Hello,

The current setup in my lab is using Motive 2.0 with VRPN streaming. Does crazyswam compatible with this setting at this moment?

paradox56, 2018-11-30:
After checking a few previous issues, it looks like the motion capture tracking option uses NatNetLinux as its backend, which is out of support (The latest update was 4 years ago). Is it possible to utilize vrpn_client_ros for Optitrack?

glennliu, 2018-12-01:
I think there are two problems,
1. VRPN doesn't look like offered the interface for getPointCloud. It can work with `motionCapture`. But I think it won't work with `libobjecttracker`.
2. VRPN cannot run with NatNet together, or they would conflict and cause program shut down. So, every function in `crazyswarm_server` used `optitrack.h` need to be deleted and rewrite to work with VRPN.

whoenig, 2018-12-01:
crazyswarm_server doesn't directly rely on optitrack (and only on motionCapture and libobjecttracker). I think its fairly easy to add VRPN support, but as glennliu pointed out, it wont work for libobjecttracker.

glennliu, 2018-12-01:
I looked at NatNet from Optitrack. It looks like those lib are mainly for windows client. There isn't even makefile in its lib. So I don't know how to use it with ROS. That's probably why crazyswarm author use NatNetLinux lib rather than original NatNet.

I think VRPN and mocap_optitrack are similar and both would be easy to add. But mocap_optitrack(https://github.com/ros-drivers/mocap_optitrack) has interface for pointcloud and mocap_optitrack author keep updating it, while VRPN hasn't been updated for one year.

So, I would try with mocap_optitrack at the moment.

paradox56, 2018-12-01:
@glennliu 
Using mocap_optitrack makes a lot of sense to me. I contacted the author of NatNetLinux yesterday and he told me he was no longer involved in Optitrack project since 2014. Therefore, I don't think it will be updated in the future anymore. 

I will also look into mocap_optitrack and hopefully we can get it working soon.

glennliu, 2018-12-01:
@paradox56 
`mocap_optitrack` is updating to work with Motive 2.0. And I tested `mocap_optitrack` today and updated my progress at #122 .
It would be great if we can work together to solve it.

paradox56, 2018-12-01:
@glennliu 
Sounds good. I am not in Lab today but I can give your code a test tomorrow. Shall we get motionCapture work first? It looks like you get some error with `publishRigidBody` in #122

glennliu, 2018-12-02:
Yep. I am trying to get motionCapture work first. And `publishRigidBody` doesn't work correct.

whoenig, 2018-12-07:
I just pushed a change which adds VRPN support and tested it with a Vicon system (which, like Optitrack, has VRPN built-in). Its not a fully polished solution, but it should be good enough for trouble shooting etc. Please let me know if that helps.

paradox56, 2018-12-08:
Ok. I made a mistake in setting, it should be 
` object_tracking_type: "motionCapture"` and the error is gone. 

However, the drone took off and hit the wall when I use the `niceHover.py ` What is the coordinate system in vicon? I believe it is relate to the coordinate transform. I am going to try ` Z Up` in OptiTrack setting next to see if it solves the problem.

whoenig, 2018-12-08:
It should match the ROS coordinate system: positive Z = top of CF, positive X = front of CF, positive Y = left of CF. Where front of the CF is where the radio antenna is.

paradox56, 2018-12-09:
@whoenig  Got it. I will adjust it accordingly. Do I also need to update to the crazyswarm firmware? I'm currently running the latest bitcraze firmware.

whoenig, 2018-12-09:
The latest bitcraze firmware is fine. However, make sure you configure hover_swarm.launch accordingly: It needs to have "bitcraze" here: https://github.com/USC-ACTLab/crazyswarm/blob/master/ros_ws/src/crazyswarm/launch/hover_swarm.launch#L10, and "2" here: https://github.com/USC-ACTLab/crazyswarm/blob/master/ros_ws/src/crazyswarm/launch/hover_swarm.launch#L25.

paradox56, 2018-12-12:
Thanks! It works now.

paradox56, 2018-12-13:
@whoenig 
OK. Something really strange happened today. 
I tried to redo the test with everything unchanged and I am unable to get \tf data for the drone. Nothing is displayed in RVIZ and the drone flipped up-side down during take off. 

In addition, crazy swarm server is expecting some rigid bodies that is not even created in Optitrack. (Rigidbody 2 etc.)To double check if everything is correct, I launched a separate VRPN node and subscribed to the publishing node. Everything looked fine as well.

whoenig, 2018-12-14:
crazyswarm_server expects rigid bodies of the form "cf<id>" (for whatever CFs are currently enabled). It uses libmotioncapture, which tries to connect to all available rigid bodies over VRPN.

Can you test the behavior when running `roslaunch crazyswarm mocap_helper.launch`?

paradox56, 2018-12-20:
@whoenig Sorry about the delay. I was working on a some other projects for the past week. 
When I run `roslaunch crazyswarm mocap_helper.launch`, everything seems to be fine. Here is the output:
```

started roslaunch server http://RoboticsLab:38425/

SUMMARY
========

PARAMETERS
 * /mocap_helper/motion_capture_type: vicon
 * /mocap_helper/object_tracking_type: motionCapture
 * /mocap_helper/vicon_host_name: vicon
 * /rosdistro: kinetic
 * /rosversion: 1.12.14

NODES
  /
    mocap_helper (crazyswarm/mocap_helper)

auto-starting new master
process[master]: started with pid [30885]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to aad372be-0498-11e9-8cee-0cc47ab4cb7e
process[rosout-1]: started with pid [30902]
started core service [/rosout]
process[mocap_helper-2]: started with pid [30910]
```

bconvens, 2019-01-04:
Hi,
I had a working crazyswarm project for some time, today I recloned and downloaded the default version of the original crazyswarm code again.
Unfortunately I now get this build error. Does anyone know how to resolve this?
Since it has to do something with VRPN I put my questions in this open issue.
Thanks for the help!
[image]
[image]

whoenig, 2019-01-04:
You will need to install the vrpn library: `sudo apt install ros-kinetic-vrpn`.

paradox56, 2019-01-07:
Do I need to run a separate VRPN node to get the position working?

whoenig, 2019-01-08:
No, you just need that library installed for the compilation to succeed. The crazyswarm-server will connect to your mocap internally without relying on any third-party node.

paradox56, 2019-01-09:
@whoenig Thank you. I have managed to get the drone flying again. It looks like i need to run `pc_permissions.sh` before running the launch file.

closed 2019-01-09 by paradox56