USC-ACTLab/crazyswarm #122: Substitute NatNet with Mocap_optitrack to solve NATNET FRAME TIMEOUT

glennliu, 2018-12-01:
Hi

We've discussed problems in using Motive with crazyswarm in previous issues #115 

**Modify code**
And I found `mocap_optitrack` is better compatible with different version of Motive. So I substitute NatNet with `mocap_optitrack` in my `crazyswarm_server`. The code is updated here: (https://github.com/glennliu/crazyswarm/tree/test_mocap)

My Motive version: 1.10.0

After the change, `/tf` is published at 120Hz, synchronized with rate from Motive. 

**Confirm frame**
Because `mocap_optitrack` use a different coordinator frame. It requires "Y up" at Motive/Data_streaming, rather than "Z up" in NatNet. Also, I build rigid body when drone pointing to mocap x axis direction. 
[image]
[image]

To test the frame is correct, I placed drone pointing to mocap x axis (as showed in upper img). And move the drone forward along its own head direction. Then the result is as lower img. I take this as correct x translation. I use this method to test all 3d position and 3d orientation. And confirm frame correct in this way.

**LED red and takeoff flipped**
After program started, LED is in solid red. 
Then launch program and takeoff, drone flipped quickly. Terminal showed as below,
[image]

Because it keep warming 
> No updated pose for motion capture object %s

This is the warning come from this part of code, and I didn't change this part. I think there is something wrong when the program try to use function `publishRigidBody`. But I haven't found why. It would be great if @whoenig can help me take a look.

I'll test more and update the result here.

Best

glennliu, 2018-12-05:
I think I successful substitute NatNet mocapObject with `mocap_optitrack` pose. The changed code is updated here [https://github.com/glennliu/crazyswarm](url). (use "test_mocap" branch")

In launch hover_swarm.launch, there is warning

> NATNET FRAME TIMEOUT

This is caused by `getLatency` function. The old `mocap()` function still read latency from NatNet. But I think it should not affect cf receive mocap data. So I leave it unchanged.

The problem is that **LED always in solid red when placing cf on the ground.** 
But if I place it on my hand, its solid green. Once I place it back on the ground, it change back to red. Real time pose is updated on rviz correctly. I'm not sure what's cause it.

glennliu, 2018-12-15:
this issue has nothing common related to #103

Manakoy, 2019-05-03:
I also encountered a similar problem when running hover_mocap.launch. Can you help me to see what went wrong? @glennliu  @whoenig 
[image]
[image]
[image]
[image]
There is no /tf information displayed in rviz, as shown in the figure:
[image]
And I seem to have connected to crazyflie but the yellow light of crazyflie does not light up.
[image]

Manakoy, 2019-05-03:
I also encountered a similar problem when running hover_mocap.launch. I hope to get your help. @glennliu @whoenig Thanks！
I am using Ubuntu 18.04, melodic and motive1.10.3

1 I did not get the information from motive, showing NATNET FRAME TIMEOUT.
2 I seem to be connected to crazyflie but the yellow light of crazyflie does not light up. Is this normal?

[image]
[image]
[image]
[image]
[image]
[image]
[image]

whoenig, 2019-05-03:
Hi,
I just pushed a commit that adds proper optitrack support (see https://github.com/USC-ACTLab/crazyswarm/commit/c9362c5972b604f254061ea355c426e10eef7da0). This new version is directly adopted from the NatNetSDK (where I ported their direct packet deserialization example to Linux). I tested it with Motive 2.0 using both streaming of rigid bodies and raw point clouds. Make sure you select "Z-Up" in the streaming configuration in Motive (and enable streaming of markers if you want to use the pointcloud option.)

Let me know if that fixes the problems you are having. I am also planning to release a mocap ROS package based on libmotioncapture.

hanzheteng, 2019-05-04:
It would be great if libmocap can be an independent ROS package! As the number of robot grows, we are running out of unique configurations.

Manakoy, 2019-05-04:
Hi,@whoenig  I have tried the method which you responded to. It is not suitable for motive1.10. Fortunately, it can run in motive2.0 .But I also encountered a new problem, when I pressed the controller to takeoff, crazyflie did not react.
[image]
[image]
[image]

whoenig, 2019-05-05:
What is the problem with Motive 1.10? From my understanding it should work in both versions, although I did not have access to Motive 1.10 to test it myself.

For your issue: Which firmware version are you using? If you use the Crazyswarm firmware and have the LED ring, make sure that the LED ring is green. There could also be an issue with the sensor calibration - check the blinking pattern of the red LED (it should blink at 1 Hz).

Manakoy, 2019-05-06:
@whoenig HI!
I don't know why it can't be used in motive1.10.  Because I've tried running motive2.0 and motive1.10 in the same configuration, but only motive2.0 can display rviz's /tf information.
Regarding my question, The version information I queried with chooser.py is 1: 0, 1, 0. I think it is because there is a problem between crazyflie and PC communication. I found several phenomena：
1.I can't query the voltage when using the battery function of chooser.py. But other functions in chooser.py are valid. And I can query the voltage by running rosrun crazyflie_tools battery.
[image]
[image]
2.I am having trouble when i trying to brush crazyflie's nrf firmware. See the picture3
[image]
3.I don‘t have LED ring. Regarding the red LED light, it always blink at 2 Hz. Maybe the problem is here. But I don't know how to change it. When I was using cfclient to fly crazyflie, the flashing frequency of the red LED was not 1 Hz, it was 2 Hz. And carzyflie can fly normally.

whoenig, 2019-05-06:
It seems like you are running an old NRF firmware that's not compatible (which explains your issues in 1. and 2.). You can try "make BLE=0 flash", or use the crazyflie_tools to flash the pre-built nrf firmware.
Regarding 3., I might not remember the correct frequency; if it works with the cfclient, it is likely not a sensor calibration issue.

Manakoy, 2019-05-09:
I tried the following methods but did not solve the problem. I still can't check the voltage through chooser.py, and crazyflie can't fly.
1.I update STM and NRF using chooser.py.
2.Then I try to use 'make cload'.
[image]
3.I also tried to refresh the firmware with crazyflie_tool.
[image]
4.I also tried to use cfclient to refresh the firmware. The firmware package is downloaded from https://github.com/bitcraze/crazyflie-release/releases

whoenig, 2019-05-09:
In the previous screenshot of battery check you got an LIBUSB_ERROR_BUSY error, which indicates that the Crazyradio is in use. Make sure that ROS is *not* connected to the CF when using chooser.py (i.e., shut-down hover-swarm.launch before using chooser.py).

Manakoy, 2019-05-10:
Yes, I tried to run the voltage check when hover-swarm.launch is not running. But also did not successfully detect the voltage

lkyfly0765, 2019-05-13:
@glennliu 
Hi, I am using https://github.com/glennliu/crazyswarm/tree/test_mocap this package to make swarm flight.
There is something wrong when I launch the hover_swarm.launch file, show in the following pic.

> [ERROR] [1557752341.371983728]: Ignoring transform for child_frame_id "cf1" from authority "unknown_publisher" because of an invalid quaternion in the transform (-nan -nan -nan -nan)
> [ERROR] [1557752341.376308427]: Ignoring transform for child_frame_id "cf1" from authority "unknown_publisher" because of a nan value in the transform (0.000000 0.000000 0.000000) (-nan -nan -nan -nan)

[image]

I have no idea what's wrong with it. Thank!

lkyfly0765, 2019-05-13:
@glennliu 
Hi, I am using https://github.com/glennliu/crazyswarm/tree/test_mocap this package to make swarm flight.
There is something wrong when I launch the hover_swarm.launch file, show in the following pic.

> [ERROR] [1557752341.371983728]: Ignoring transform for child_frame_id "cf1" from authority "unknown_publisher" because of an invalid quaternion in the transform (-nan -nan -nan -nan)
> [ERROR] [...]

[image]

I have no idea what's wrong with it. Thank!

whoenig, 2019-11-12:
The latest crazyswarm has proper OptiTrack support, using the NatNet SDK. Please try if that resolves your problem.

lkyfly0765, 2019-11-17:
@whoenig 
Hi whoenig, I have tried the latest crazyswarm, together with the Natnetsdkcrossplatform package.
But there still have something wrong, I am not sure if there is any configuration errors.

 I am not sure about why it occurs the   **what():  set_option: Invalid argument**. Could you please help me to figure it out?

Following is part of the logs of the command `roslaunch crazyswarm hover_swarm.launch`

```
NODES
  /
    crazyswarm_server (crazyswarm/crazyswarm_server)
    crazyswarm_teleop (crazyswarm/crazyswarm_teleop)
    joy (joy/joy_node)
    rviz (rviz/rviz)

auto-starting new master
process[master]: started with pid [4630]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 171571d2-053a-11ea-ae18-34f64bc13130
process[rosout-1]: started with pid [4643]
started core service [/rosout]
process[crazyswarm_server-2]: started with pid [4660]
process[joy-3]: started with pid [4661]
process[crazyswarm_teleop-4]: started with pid [4662]
process[rviz-5]: started with pid [4684]
[ INFO] [1573555764.991148880]: Wait for services...
[ INFO] [1573555764.992241960]: waitForService: Service [/emergency] has not been advertised, waiting...
[ INFO] [1573555765.013889625]: waitForService: Service [/emergency] is now available.
[ INFO] [1573555765.016484269]: Manager ready.
terminate called after throwing an instance of 'boost::exception_detail::clone_impl<boost::exception_detail::error_info_injector<boost::system::system_error> >'
  what():  set_option: Invalid argument
[ERROR] [1573555765.158920318]: Couldn't open joystick force feedback!
[ INFO] [1573555765.159008863]: Opened joystick: /dev/input/js0. deadzone_: 0.050000.
[crazyswarm_server-2] process has died [pid 4660, exit code -6, cmd /home/eric/crazyswarm/ros_ws/devel/lib/crazyswarm/crazyswarm_server __name:=crazyswarm_server __log:=/home/eric/.ros/log/171571d2-053a-11ea-ae18-34f64bc13130/crazyswarm_server-2.log].
log file: /home/eric/.ros/log/171571d2-053a-11ea-ae18-34f64bc13130/crazyswarm_server-2*.log
```

My Motive server IP address is 169.254.41.1, and I have updated the Multicast Address in the cpp files in folder _Natnetsdkcrossplatform/src/_ . 

By the way, I wonder if it right to move the Natnetsdkcrosspaltform package into the Externaldependencies folder? 

If there is any instructions about how to configure the parameters correctly under the optitrack mocap system?

The hover_swarm.launch file is configured as follow:
```
<?xml version="1.0"?>
<launch>
  <arg name="joy_dev" default="/dev/input/js0" />

  <rosparam command="load" file="$(find crazyswarm)/launch/crazyflieTypes.yaml" />
  <rosparam command="load" file="$(find crazyswarm)/launch/crazyflies.yaml" />

  <node pkg="crazyswarm" type="crazyswarm_server" name="crazyswarm_server" output="screen" >
    <rosparam>
      world_frame: "/world"
      # Logging configuration (Use enable_logging to actually enable logging)
      genericLogTopics: ["log1"]
      genericLogTopicFrequencies: [10]
      genericLogTopic_log1_Variables: ["stateEstimate.x", "ctrltarget.x"]
      # firmware parameters for all drones (use crazyflieTypes.yaml to set per type, or
      # allCrazyflies.yaml to set per drone)
      firmwareParams:
        commander:
          enHighLevel: 1
        stabilizer:
          estimator: 2 # 1: complementary, 2: kalman
          controller: 2 # 1: PID, 2: mellinger
        ring:
          effect: 16 # 6: double spinner, 7: solid color, 16: packetRate
          solidBlue: 255 # if set to solid color
          solidGreen: 0 # if set to solid color
          solidRed: 0 # if set to solid color
          headlightEnable: 0
        locSrv:
          extPosStdDev: 1e-3
          extQuatStdDev: 0.5e-1
        kalman:
          pNAcc_xy: 1.0
          pNAcc_z: 2.0
      # tracking
      motion_capture_type: "optitrack" # one of none,vicon,optitrack,qualisys,vrpn
      object_tracking_type: "motionCapture" # one of motionCapture,libobjecttracker
      send_position_only: False # set to False to send position+orientation; set to True to send position only
      # vicon_host_name: "vicon"
      optitrack_host_name: "169.254.41.1"
      # qualisys_host_name: "10.0.5.219"
      # qualisys_base_port: 22222
      # vrpn_host_name: "vicon"
      save_point_clouds: ~/pointCloud.ot
      print_latency: False
      write_csvs: False
      force_no_cache: False
      enable_parameters: True
      enable_logging: False
    </rosparam>
  </node>

  <node name="joy" pkg="joy" type="joy_node" output="screen">
    <param name="dev" value="$(arg joy_dev)" />
  </node>

  <node pkg="crazyswarm" type="crazyswarm_teleop" name="crazyswarm_teleop" output="screen">
    <param name="csv_file" value="$(find crazyswarm)/launch/figure8_smooth.csv" />
    <param name="timescale" value="0.8" />
  </node>

  <node name="rviz" pkg="rviz" type="rviz" args="-d $(find crazyswarm)/launch/test.rviz"/>

  <!-- <node pkg="rqt_plot" type="rqt_plot" name="rqt_plot_x" args="/cf2/log1/values[0]"/> -->
  <!-- <node pkg="rqt_plot" type="rqt_plot" name="rqt_plot_roll" args="/cf1/log1/values[2] /cf1/log1/values[3]"/> -->

</launch>
```
Also, I am using the Ubuntu 16.04 with the boost version 1.58. When I try to comply the Natnetsdkcrossplatform using the command `cmake ..` , it requires the boost version 1.65 with errors. 

To solve it, I change the Line 3 in [https://github.com/whoenig/NatNetSDKCrossplatform/blob/master/CMakeLists.txt](https://github.com/whoenig/NatNetSDKCrossplatform/blob/master/CMakeLists.txt) from 
`find_package(Boost 1.65 REQUIRED COMPONENTS system thread)`
to
`find_package(Boost 1.58 REQUIRED COMPONENTS system thread)`

Then the `cmake .. `and `make` works well.

Wish you could help me to solve those questions!
Thanks a lot!

whoenig, 2019-11-18:
I have not seen this `set_option: Invalid argument` error before. Could you recompile with debugging enabled (`catkin_make -DBUILD_TYPE=DEBUG`) and launch the crazyswarm_server using gdb (see http://wiki.ros.org/roslaunch/Tutorials/Roslaunch%20Nodes%20in%20Valgrind%20or%20GDB). The callstack will help a lot to identify the problem. I assume it has something to do with the configuration, but I don't see an issue.

You should *not* need to update the multicast address. All you need to configure is the IP address of the computer that runs Motive (unless you changed the default settings of Motive, of course).

Your "trick" for NatNetSDKCrossplatform is correct. Do the sample applications in this repository work for you?

lkyfly0765, 2019-11-19:
I have recompiled with debug mode and the following is the log. I will try to launch the crazyswarm_server with gdb tomorrow as the lab is out of power today.

I see that in the compile logs, some dependencies is missing, will this cause some issues?

```
eric@eric-ThinkPad-S2:~/crazyswarm/ros_ws$ catkin_make -DBUILD_TYPE=DEBUG
Base path: /home/eric/crazyswarm/ros_ws
Source space: /home/eric/crazyswarm/ros_ws/src
Build space: /home/eric/crazyswarm/ros_ws/build
Devel space: /home/eric/crazyswarm/ros_ws/devel
Install space: /home/eric/crazyswarm/ros_ws/install
####
#### Running command: "cmake /home/eric/crazyswarm/ros_ws/src -DBUILD_TYPE=DEBUG -DCATKIN_DEVEL_PREFIX=/home/eric/crazyswarm/ros_ws/devel -DCMAKE_INSTALL_PREFIX=/home/eric/crazyswarm/ros_ws/install -G Unix Makefiles" in "/home/eric/crazyswarm/ros_ws/build"
####
-- The C compiler identification is GNU 5.4.0
-- The CXX compiler identification is GNU 5.4.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Using CATKIN_DEVEL_PREFIX: /home/eric/crazyswarm/ros_ws/devel
-- Using CMAKE_PREFIX_PATH: /opt/ros/kinetic
-- This workspace overlays: /opt/ros/kinetic
-- Found PythonInterp: /usr/bin/python (found version "2.7.12") 
-- Using PYTHON_EXECUTABLE: /usr/bin/python
-- Using Debian Python package layout
-- Using empy: /usr/bin/empy
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /home/eric/crazyswarm/ros_ws/build/test_results
-- Found gmock sources under '/usr/src/gmock': gmock will be built
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Found gtest sources under '/usr/src/gmock': gtests will be built
-- Using Python nosetests: /usr/bin/nosetests-2.7
-- catkin 0.7.18
-- BUILD_SHARED_LIBS is on
-- BUILD_SHARED_LIBS is on
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- ~~  traversing 10 packages in topological order:
-- ~~  - crazyflie (metapackage)
-- ~~  - crazyflie_cpp
-- ~~  - crazyflie_description
-- ~~  - crazyflie_tools
-- ~~  - libmotioncapture
-- ~~  - libobjecttracker
-- ~~  - crazyflie_demo
-- ~~  - crazyflie_controller
-- ~~  - crazyflie_driver
-- ~~  - crazyswarm
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- +++ processing catkin metapackage: 'crazyflie'
-- ==> add_subdirectory(crazyflie_ros/crazyflie)
-- +++ processing catkin package: 'crazyflie_cpp'
-- ==> add_subdirectory(crazyflie_ros/crazyflie_cpp)
[... 416 lines cut ...]
[ 98%] Built target crazyswarm_teleop
[100%] Linking CXX executable /home/eric/crazyswarm/ros_ws/devel/lib/crazyswarm/mocap_helper
[100%] Built target mocap_helper
In file included from /home/eric/crazyswarm/ros_ws/src/crazyswarm/src/crazyswarm_server.cpp:62:0:
/home/eric/crazyswarm/ros_ws/src/externalDependencies/libobjecttracker/include/libobjecttracker/cloudlog.hpp: In member function ‘void libobjecttracker::PointCloudPlayer::play(libobjecttracker::ObjectTracker&) const’:
/home/eric/crazyswarm/ros_ws/src/externalDependencies/libobjecttracker/include/libobjecttracker/cloudlog.hpp:103:57: warning: format ‘%d’ expects argument of type ‘int’, but argument 2 has type ‘std::size_t {aka long unsigned int}’ [-Wformat=]
     printf("\n  %d  ------------------------------\n", i);
                                                         ^
/home/eric/crazyswarm/ros_ws/src/externalDependencies/libobjecttracker/include/libobjecttracker/cloudlog.hpp: In member function ‘void libobjecttracker::PointCloudDebugger::convert(libobjecttracker::ObjectTracker&, std::vector<boost::shared_ptr<pcl::PointCloud<pcl::PointXYZ> > >&)’:
/home/eric/crazyswarm/ros_ws/src/externalDependencies/libobjecttracker/include/libobjecttracker/cloudlog.hpp:139:57: warning: format ‘%d’ expects argument of type ‘int’, but argument 2 has type ‘std::size_t {aka long unsigned int}’ [-Wformat=]
     printf("\n  %d  ------------------------------\n", i);
                                                         ^
/home/eric/crazyswarm/ros_ws/src/externalDependencies/libobjecttracker/include/libobjecttracker/cloudlog.hpp:152:55: warning: format ‘%d’ expects argument of type ‘int’, but argument 2 has type ‘std::vector<libobjecttracker::Object>::size_type {aka long unsigned int}’ [-Wformat=]
      printf("Object vector size: %d\n", objects.size());
                                                       ^
[100%] Linking CXX executable /home/eric/crazyswarm/ros_ws/devel/lib/crazyswarm/crazyswarm_server
[100%] Built target crazyswarm_server
```

> Your "trick" for NatNetSDKCrossplatform is correct. Do the sample applications in this repository work for you?

You mean the mcr and natnettest application? I have tried them but the `./mcr` will only show the Motive version and the natnet version; and after I execute the `./natnettest` there is nothing in the xterm. I will upload the log tomorrow. 

And, could you please tell me what is the normal output in the xterm if a execute the mcr and natnettest application?
Thanks!

whoenig, 2019-11-19:
Your build looks normal. Since your issue occurs at runtime, only the gdb output will help to figure out where the exception is thrown (and hopefully why).

For NatNetSDK Crossplatform: `./natnettest` just prints the version and exits. `./mcr` is supposed to continuously print the actual data packets (that is: poses if any objects are enabled; locations of the individual markers otherwise). If you don't see anything, that's certainly concerning, because it means that streaming over multicast does not work for you. I assume you did enable that feature in Motive?

lkyfly0765, 2019-11-20:
1. I have reset the Multicast Address in the cpp file of the Natnetsdkcrossplatform/src folder to the 239.255.42.99. And rebuilt it. The `./mcr` and `./natnettest` tests well. 
```
eric@eric-ThinkPad-S2:~/crazyswarm/ros_ws/src/externalDependencies/libmotioncapture/externalDependencies/NatNetSDKCrossplatform/built$ ./natnettest 169.254.41.1
1 279
NatNetVersion: 236.164.53.0
ServerVersion: 3.1.0.0

# Part of the ./mcr program are shown as follow:
eric@eric-ThinkPad-S2:~/crazyswarm/ros_ws/src/externalDependencies/libmotioncapture/externalDependencies/NatNetSDKCrossplatform/built$ ./mcr 169.254.41.1
Begin Packet
-------
Message ID : 7
Byte count : 325
Frame # : 987569
Marker Set Count : 2
Model Name: crazyflie1
Marker Count : 3
	Marker 0 : [x=0.03,y=-0.03,z=0.02]
	Marker 1 : [x=0.02,y=0.01,z=0.02]
	Marker 2 : [x=-0.02,y=-0.03,z=0.02]
Model Name: all
Marker Count : 3
	Marker 0 : [x=0.03,y=-0.03,z=0.02]
	Marker 1 : [x=0.02,y=0.01,z=0.02]
	Marker 2 : [x=-0.02,y=-0.03,z=0.02]
Unidentified Marker Count : 3
	Marker 0 : pos = [0.03,-0.03,0.02]
	Marker 1 : pos = [0.02,0.01,0.02]
	Marker 2 : pos = [-0.02,-0.03,0.02]
Rigid Body Count : 1
ID : 4
pos: [0.01,-0.02,0.02]
ori: [0.00,-0.00,0.01,-1.00]
Mean marker error: 0.00
Skeleton Count : 0
Labeled Marker Count : 3
ID  : [MarkerID: 1] [ModelID: 4]
pos : [0.03,-0.03,0.02]
size: [0.02]
err:  [0.00]
ID  : [MarkerID: 2] [ModelID: 4]
pos : [0.02,0.01,0.02]
size: [0.01]
err:  [0.00]
ID  : [MarkerID: 3] [ModelID: 4]
pos : [-0.02,-0.03,0.02]
size: [0.02]
err:  [0.00]
Timestamp : 8229.742
Mid-exposure timestamp : 32188639183
Camera data received timestamp : 32188655821
Transmit timestamp : 32188657531
End Packet
```
2. The launch file with `launch-prefix="xterm -e gdb --args"` :
Could you please help me to find out what's wrong?
[image]

3. The connection between crazyradio and crazyflie is not established. I not sure if it's because of question 2. The communication led on crazyradio and crazyflie is not light up and the LED ring under the crazyflie if also flashing same with after powered on.

On Windows computer which is running Motive, I am using the Motive 2.20 Final version.
I follow the instructions about how to build the rigid body on [https://github.com/USC-ACTLab/libmotioncapture/pull/3](https://github.com/USC-ACTLab/libmotioncapture/pull/3).

The data stream panel in Motive is set as follow:

```
Optitrack Streaming Engine
Broadcast Frame Data: On
Local Interface:169.254.41.1
Labeled Markers: On
Unlabeled Markers: On
Assets Markers: On
Rigid Bodies: On
Up Axis: Z Up
Remote Triggers: On
Transmission Type: Multicast

VRPN Steaming Engine
Broadcast VRPN Data: Off
```

Does the latest version of Motive supported by the crazyswarm package? And if there are any wrong configuration of the motive software?

Thanks!
Tracked Streaming Engine
Broadcast Tracked Data: Off

```

whoenig, 2019-11-20:
1. Looks good!
2. Could you press enter in gdb to get a longer stack trace? It looks like the the error is caused when the network stack is initialized. I assume you don't run anything else in the background that might try using the same multicast address? (This is strange, because your experiment in 1. works, and both use the same code to initialize the network...)
3. Yes, probably because the crazyswarm_server crashes. So if 2) is resolved, this issue should go away as well.

Your Motive settings look good (and so does the output in 1), so I think you are configured well. And yes, crazyswarm does support Motive 2.X (although I personally only tested with 2.0).

lkyfly0765, 2019-11-21:
Today I try to test it again. I'm not sure about how to get a longer stack trace. I just type in the `set backstace limit 100` and type `bt` to get the stack trace, but it still have only frame #0 to frame #13, I think it's all the stack trace. Well, I am still new in gdb, so if there is other command need to type in, please tell me.

I am quite sure that I am not running any other application on my Ubuntu system. And on Windows, there is no other software using the multicast address, the port 1510 adn 1511 is only used by Motive.
It's strange that I can not get the data correctly in the crazyswarm package but I can get it in the mcr application separately.

[image]

whoenig, 2019-11-22:
Your stack trace is complete, although the line numbers are missing. I guess it didn't really build a debug version. Try `catkin_make clean` followed by `catkin_make -DCMAKE_BUILD_TYPE=Debug`.

Please also check that your optitrack.cpp does not have any local changes (i.e., it should be identical to https://github.com/USC-ACTLab/libmotioncapture/blob/master/src/optitrack.cpp). I compared the code of the relevant network parts of the mcr portion and this file and it seems to be identical.

lkyfly0765, 2019-11-23:
You are right! 
As you said, I forgot that I change the multicast address to the Motive running computer IP in optitrack.cpp. Now I fixed it and the `set_option: Invalid argument` has been solved. It's all because the Multicast Address in optitrack.cpp is wrong. The default multicast address is `239.255.42.99` , see [https://www.optitrack.com/products/natnet-sdk/](https://www.optitrack.com/products/natnet-sdk/).

But there occurs another error, I used the gbd and find that the error comes from the optitrack.cpp:367
`int nRigidBodies = 0; memcpy(&nRigidBodies, ptr, 4); ptr += 4;`
I just go through the optitrack.cpp but I cannot find out why it occurs an error.

The gbd screenshot is shown as follow:
[image]

whoenig, 2019-11-26:
This does indicate that there is an issue with decoding the packet (essentially, the format that it receives does not match the expected format). However, if `mcr` works correctly, I don't understand why that would be happening. (I didn't see a mismatch between the https://github.com/whoenig/NatNetSDKCrossplatform/blob/master/src/PacketClient.cpp#L620 and optitrack.cpp.)

Which tracking mode are you using? If you want to use optitrack, make sure you only enable rigidBody streaming in motive. If you want to use libobjecttracker, try to only enable unlabeled markers in Motive.

lkyfly0765, 2019-11-26:
The error has been solved. It’s because that I turn on both Stream labeled markers and Stream Rigid Bodies, this will cause the data decoding issue. So if you are using motioncapture as the tracking mode, only stream rigid bodies should be turned on and any other related to the stream markers (labeled or unlabeled) must be turn off. 

Well, thanks @whoenig a lot that I can run the hover_swarm.launch correctly and receive the tracking data perfectly. Now, in Rviz the cf’s position is shown and the LED ring under the cf is flashing green. 

Unfortunately, I always have trouble with issue a take off order with the joystick start(green) button. When I press the green button, in the xterm, there is no log shows the `takeoff` and the cf has no respond with this order. I tested on another cf and it has the same situation. May this cause by the joy node is not connected with the crazyswarm _server? In rqt_graph I see that there is no connection between crazyswarm_teleop and crazyswarm_server. Could you please give me some advice ?

By the way, today I tested to fly an advanced swarm script. So I start the figure _8 script in another xterm. The cf has responded to the takeoff order, but it just slipping on the ground and then flipped. In simulation mode, the cf should takeoff firstly, then fly the figure 8 trajectory and land. But it seems that the cf isn’t takeoff but starts the figure 8 trajectory directly on the ground. Did I do something wrong?

whoenig, 2019-11-27:
Joystick: The start button is not the green button, but the small button labeled start next to the xbox logo in the middle, see https://github.com/USC-ACTLab/crazyswarm/blob/master/ros_ws/src/crazyswarm/src/crazyswarm_teleop.cpp#L70-L72.

Figure8: Sounds like there is still an issue with the state estimation. Did you verify that the pose estimate is correct now?

lkyfly0765, 2019-11-27:
Thanks! It’s because I followed the instructions for the crazyflie_ros package to press the green button to issue a take off command. The joystick now is working.

But the flip issue still remains. In Motive, I set the axis to Z-up. And I checked the state estimates in Rviz by moving the crazyflie  from origin manually on the ground. When I move the cf to it’s front direction( the antenna side), in Rviz, the crazyflie moves to the positive X axis. I checked the Y position with the same technique, the Y position is correct, too. It seems that the state estimate is right. But the cf still cannot do a nice hover but drifting once the motor start and then flipped onto the ground. Both takeoff-hover and figure 8 has the same issue.

whoenig, 2019-11-27:
This looks like there might be a mismatch in orientation between Motive and the Crazyflie (the antenna should be at yaw=0). Perhaps you can try using position-only EKF by switching https://github.com/USC-ACTLab/crazyswarm/blob/master/ros_ws/src/crazyswarm/launch/hover_swarm.launch#L38 to True. If that works, then your rigid body in Motive is oriented incorrectly.

lkyfly0765, 2020-11-06:
@whoenig 
Hi whoenig, I didn't studied in crazyswarm for a few months. Recently, I need to re-using the crazyswarm project to do some more experiments. So it re-clone the crazyswarm project to update it to the latest version, but here I came across some issues.
The issue is quite same with what we discussed last year. The error information is :  `[crazyswarm_server-2] process has died`. Initially, I think it's because of the misconfiguration, so I checked all the things which I made mistakes. But it didn't work.

The configuration of the important files is shown below:
1. hover_swarm.launch
>    #tracking
      motion_capture_type: "optitrack" # one of none,vicon,optitrack,qualisys,vrpn
      object_tracking_type: "motionCapture" # one of motionCapture,libobjecttracker
      send_position_only: False # set to False to send position+orientation; set to True to send position only
      # vicon_host_name: "192.168.1.200"
      optitrack_host_name: "192.168.1.200"
      # qualisys_host_name: "10.0.5.219"
      # qualisys_base_port: 22222
      # vrpn_host_name: "192.168.1.200"
      save_point_clouds: ~/pointCloud.ot
      print_latency: False
      write_csvs: False
      force_no_cache: False
      enable_parameters: True
      enable_logging: False
2. Motive Data_streaming pane  (Motive version is : 2.1.1)
[image]

 3. The `allcrazyflies.yaml `and `crazyflies.yaml` is correctly configured.
4. The `optitrack.cpp` has been changed to the correct multicast_address.
    The Motive computer's IP address is 192.168.1.200, and the Multicast interface is set to 192.168.1.200 in Motive-Streaming pane

>   L10: constexpr const char* MULTICAST_ADDRESS = "192.168.1.200";

To check if there are any connection issue between my ubuntu system with the Motive computer, I use the **NatNetSDKCrossplatform** and the `mcr` or `natnettest` works well. The `mcr`  returns the correct information of the rigid body in motive, this means that the connection between two computer works well, also the Natnet.

> Begin Packet
    -------
Message ID : 7
Byte count : 116
Frame # : 935750
Marker Set Count : 0
Unidentified Marker Count : 0
Rigid Body Count : 1
ID : 1
pos: [0.02,-0.02,0.02]
ori: [0.01,1.00,-0.00,0.01]
Mean marker error: 0.00
Skeleton Count : 0
Labeled Marker Count : 0
Timestamp : 7797.917
Mid-exposure timestamp : 1866713199529
Camera data received timestamp : 1866713245528
Transmit timestamp : 1866713253075
End Packet

I try to use the gdb tool to debug the program and it returns as folllowing:
[image]
It shows that the error may comes from the [https://github.com/USC-ACTLab/libmotioncapture/blob/master/src/optitrack.cpp#L311](https://github.com/USC-ACTLab/libmotioncapture/blob/master/src/optitrack.cpp#L311), it seems that there are some trouble with **Join the multicast group** , but the IP address is correct in gdb `host_name="192.168.1.200".` Could you please help me to solve it?

whoenig, 2020-11-08:
I don't think you should change the multicast address in the code, especially not if you didn't change it in NatNetSDKCrossplatform.

whoenig, 2021-10-01:
We have now an improved OptiTrack backend that also automatically identifies the correct multicast address. Please start a new discussion, if you continue facing problems with the latest version.

closed 2021-10-01 by whoenig