USC-ACTLab/crazyswarm #123: Process has died when set mocation_capture_type to "none"

paradox56, 2018-12-04:
Hello, 

I was trying to do some testing without external object tracking and I got the following error when `motion_capture_type` is set to `none` in `hover_swarm.launch`. 

Here is the error 
```
xx@xx-MS-7922:~/crazyswarm/ros_ws/src/crazyswarm/launch$ roslaunch crazyswarm hover_swarm.launch 
... logging to /home/xx/.ros/log/5fecd6c8-f7dc-11e8-bf31-38baf81d0dcb/roslaunch-xx-MS-7922-699.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://xx-MS-7922:43883/

SUMMARY
========

PARAMETERS
 * /crazyflieTypes/default/batteryVoltageWarning: 3.8
 * /crazyflieTypes/default/batteryVoltateCritical: 3.7
 * /crazyflieTypes/default/bigQuad: False
 * /crazyflieTypes/default/dynamicsConfiguration: 0
 * /crazyflieTypes/default/firmwareParams/ctrlMel/i_range_m_z: 1500
 * /crazyflieTypes/default/firmwareParams/ctrlMel/i_range_xy: 2.0
 * /crazyflieTypes/default/firmwareParams/ctrlMel/i_range_z: 0.4
 * /crazyflieTypes/default/firmwareParams/ctrlMel/kR_xy: 70000
 * /crazyflieTypes/default/firmwareParams/ctrlMel/kR_z: 60000
 * /crazyflieTypes/default/firmwareParams/ctrlMel/kd_omega_rp: 200
 * /crazyflieTypes/default/firmwareParams/ctrlMel/kd_xy: 0.2
 * /crazyflieTypes/default/firmwareParams/ctrlMel/kd_z: 0.4
 * /crazyflieTypes/default/firmwareParams/ctrlMel/ki_m_z: 500
 * /crazyflieTypes/default/firmwareParams/ctrlMel/ki_xy: 0.05
 * /crazyflieTypes/default/firmwareParams/ctrlMel/ki_z: 0.05
 * /crazyflieTypes/default/firmwareParams/ctrlMel/kp_xy: 0.4
 * /crazyflieTypes/default/firmwareParams/ctrlMel/kp_z: 1.25
 * /crazyflieTypes/default/firmwareParams/ctrlMel/kw_xy: 20000
 * /crazyflieTypes/default/firmwareParams/ctrlMel/kw_z: 12000
 * /crazyflieTypes/default/firmwareParams/ctrlMel/mass: 0.032
 * /crazyflieTypes/default/firmwareParams/ctrlMel/massThrust: 132000
 * /crazyflieTypes/default/markerConfiguration: 0
 * /crazyflieTypes/large/batteryVoltageWarning: 11.4
 * /crazyflieTypes/large/batteryVoltateCritical: 11.1
 * /crazyflieTypes/large/bigQuad: True
 * /crazyflieTypes/large/dynamicsConfiguration: 0
 * /crazyflieTypes/large/firmwareParams/ctrlMel/i_range_m_z: 1500
 * /crazyflieTypes/large/firmwareParams/ctrlMel/i_range_xy: 2.0
 * /crazyflieTypes/large/firmwareParams/ctrlMel/i_range_z: 4.0
 * /crazyflieTypes/large/firmwareParams/ctrlMel/kR_xy: 30000
 * /crazyflieTypes/large/firmwareParams/ctrlMel/kR_z: 40000
 * /crazyflieTypes/large/firmwareParams/ctrlMel/kd_omega_rp: 100
 * /crazyflieTypes/large/firmwareParams/ctrlMel/kd_xy: 3.0
 * /crazyflieTypes/large/firmwareParams/ctrlMel/kd_z: 3.0
 * /crazyflieTypes/large/firmwareParams/ctrlMel/ki_m_z: 500
 * /crazyflieTypes/large/firmwareParams/ctrlMel/ki_xy: 1.0
 * /crazyflieTypes/large/firmwareParams/ctrlMel/ki_z: 2.0
 * /crazyflieTypes/large/firmwareParams/ctrlMel/kp_xy: 8.0
 * /crazyflieTypes/large/firmwareParams/ctrlMel/kp_z: 8.0
 * /crazyflieTypes/large/firmwareParams/ctrlMel/kw_xy: 9000
 * /crazyflieTypes/large/firmwareParams/ctrlMel/kw_z: 10000
 * /crazyflieTypes/large/firmwareParams/ctrlMel/mass: 0.38
 * /crazyflieTypes/large/firmwareParams/ctrlMel/massThrust: 5400
 * /crazyflieTypes/large/markerConfiguration: 2
 * /crazyflieTypes/medium/batteryVoltageWarning: 7.6
 * /crazyflieTypes/medium/batteryVoltateCritical: 7.4
 * /crazyflieTypes/medium/bigQuad: True
 * /crazyflieTypes/medium/dynamicsConfiguration: 0
 * /crazyflieTypes/medium/firmwareParams/ctrlMel/i_range_m_z: 1500
 * /crazyflieTypes/medium/firmwareParams/ctrlMel/i_range_xy: 2.0
 * /crazyflieTypes/medium/firmwareParams/ctrlMel/i_range_z: 2.0
 * /crazyflieTypes/medium/firmwareParams/ctrlMel/kR_xy: 50000
 * /crazyflieTypes/medium/firmwareParams/ctrlMel/kR_z: 60000
 * /crazyflieTypes/medium/firmwareParams/ctrlMel/kd_omega_rp: 200
 * /crazyflieTypes/medium/firmwareParams/ctrlMel/kd_xy: 1.0
 * /crazyflieTypes/medium/firmwareParams/ctrlMel/kd_z: 1.0
 * /crazyflieTypes/medium/firmwareParams/ctrlMel/ki_m_z: 500
 * /crazyflieTypes/medium/firmwareParams/ctrlMel/ki_xy: 0.1
 * /crazyflieTypes/medium/firmwareParams/ctrlMel/ki_z: 0.5
 * /crazyflieTypes/medium/firmwareParams/ctrlMel/kp_xy: 2.0
 * /crazyflieTypes/medium/firmwareParams/ctrlMel/kp_z: 3
 * /crazyflieTypes/medium/firmwareParams/ctrlMel/kw_xy: 16000
 * /crazyflieTypes/medium/firmwareParams/ctrlMel/kw_z: 12000
 * /crazyflieTypes/medium/firmwareParams/ctrlMel/mass: 0.11
 * /crazyflieTypes/medium/firmwareParams/ctrlMel/massThrust: 23000
 * /crazyflieTypes/medium/markerConfiguration: 1
 * /crazyflies: [{'initialPositio...
 * /crazyswarm_server/broadcast_address: FFE7E7E7E701
 * /crazyswarm_server/enable_logging: False
 * /crazyswarm_server/enable_parameters: True
 * /crazyswarm_server/firmware: bitcraze
 * /crazyswarm_server/firmwareParams/commander/enHighLevel: 1
 * /crazyswarm_server/firmwareParams/ring/effect: 16
 * /crazyswarm_server/firmwareParams/ring/headlightEnable: 0
 * /crazyswarm_server/firmwareParams/ring/solidBlue: 255
 * /crazyswarm_server/firmwareParams/ring/solidGreen: 0
 * /crazyswarm_server/firmwareParams/ring/solidRed: 0
 * /crazyswarm_server/firmwareParams/stabilizer/controller: 2
 * /crazyswarm_server/firmwareParams/stabilizer/estimator: 2
 * /crazyswarm_server/force_no_cache: False
 * /crazyswarm_server/genericLogTopicFrequencies: [10]
 * /crazyswarm_server/genericLogTopic_log1_Variables: ['ctrlStat.edist']
 * /crazyswarm_server/genericLogTopics: ['log1']
 * /crazyswarm_server/motion_capture_type: none
 * /crazyswarm_server/object_tracking_type: motionCapture
 * /crazyswarm_server/optitrack_local_ip: localhost
 * /crazyswarm_server/optitrack_server_ip: optitrack
 * /crazyswarm_server/print_latency: False
 * /crazyswarm_server/save_point_clouds: ~/pointCloud.ot
 * /crazyswarm_server/world_frame: /world
 * /crazyswarm_server/write_csvs: False
 * /crazyswarm_teleop/csv_file: /home/xx/crazyswa...
 * /crazyswarm_teleop/timescale: 0.8
 * /dynamicsConfigurations/0/maxFitnessScore: 0.001
 * /dynamicsConfigurations/0/maxPitch: 1.4
 * /dynamicsConfigurations/0/maxPitchRate: 20.0
 * /dynamicsConfigurations/0/maxRoll: 1.4
 * /dynamicsConfigurations/0/maxRollRate: 20.0
 * /dynamicsConfigurations/0/maxXVelocity: 2.0
 * /dynamicsConfigurations/0/maxYVelocity: 2.0
 * /dynamicsConfigurations/0/maxYawRate: 10.0
 * /dynamicsConfigurations/0/maxZVelocity: 3.0
 * /joy/dev: /dev/input/js0
 * /markerConfigurations/0/numPoints: 4
 * /markerConfigurations/0/offset: [0.0, -0.01, -0.04]
 * /markerConfigurations/0/points/0: [0.0177184, 0.013...
 * /markerConfigurations/0/points/1: [-0.0262914, 0.05...
 * /markerConfigurations/0/points/2: [-0.0328889, -0.0...
 * /markerConfigurations/0/points/3: [0.0431307, -0.03...
 * /markerConfigurations/1/numPoints: 4
 * /markerConfigurations/1/offset: [0.0, 0.0, -0.03]
 * /markerConfigurations/1/points/0: [-0.00896228, -0....
 * /markerConfigurations/1/points/1: [-0.0156318, 0.09...
 * /markerConfigurations/1/points/2: [0.0461693, -0.08...
 * /markerConfigurations/1/points/3: [-0.0789959, -0.0...
 * /markerConfigurations/2/numPoints: 4
 * /markerConfigurations/2/offset: [0.0, 0.0, -0.06]
 * /markerConfigurations/2/points/0: [0.0558163, -0.00...
 * /markerConfigurations/2/points/1: [-0.0113941, 0.00...
 * /markerConfigurations/2/points/2: [-0.0306277, 0.05...
 * /markerConfigurations/2/points/3: [0.0535816, -0.04...
 * /numDynamicsConfigurations: 1
 * /numMarkerConfigurations: 3
 * /rosdistro: kinetic
 * /rosversion: 1.12.14

NODES
  /
    crazyswarm_server (crazyswarm/crazyswarm_server)
    crazyswarm_teleop (crazyswarm/crazyswarm_teleop)
    joy (joy/joy_node)
    rviz (rviz/rviz)

auto-starting new master
process[master]: started with pid [709]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 5fecd6c8-f7dc-11e8-bf31-38baf81d0dcb
process[rosout-1]: started with pid [722]
started core service [/rosout]
process[crazyswarm_server-2]: started with pid [742]
process[joy-3]: started with pid [743]
process[crazyswarm_teleop-4]: started with pid [744]
process[rviz-5]: started with pid [758]
[ INFO] [1543938651.564491100]: Wait for services...
[ INFO] [1543938651.564936766]: waitForService: Service [/emergency] has not been advertised, waiting...
[ INFO] [1543938651.587777661]: waitForService: Service [/emergency] is now available.
[ INFO] [1543938651.589066327]: Manager ready.
ch: 1
[ INFO] [1543938651.726936806]: Opened joystick: /dev/input/js0. deadzone_: 0.050000.
terminate called after throwing an instance of 'std::runtime_error'
  what():  timeout
[crazyswarm_server-2] process has died [pid 742, exit code -6, cmd /home/xx/crazyswarm/ros_ws/devel/lib/crazyswarm/crazyswarm_server __name:=crazyswarm_server __log:=/home/xx/.ros/log/5fecd6c8-f7dc-11e8-bf31-38baf81d0dcb/crazyswarm_server-2.log].
log file: /home/xx/.ros/log/5fecd6c8-f7dc-11e8-bf31-38baf81d0dcb/crazyswarm_server-2*.log

```

paradox56, 2018-12-06:
@whoenig 
Hello whoenig,

Thanks for the reply. Here is what my crazyflies.yaml looks like
```
crazyflies:
- channel: 80
  id: 1
  initialPosition: [1.5, 1.5, 0.0]
  type: default
```

The crazyflie is set to chanel 80 with address `0xE7E7E7E701`

glennliu, 2018-12-06:
I tried change `motion_capture_type` to `none`. I didn't meet the program died problem.
As whoenig mentioned, it does looks like connection problem. I suggest you also check your chooser.py. The connection should work fine if chooser.py can connect to cf.

paradox56, 2018-12-06:
@whoenig @glennliu 
Sorry about the late reply. I was working on something else earlier today. 
As you suggested, I think I do have issues with radio after checking the connection. When I click `version` button in chooser.py, it returns `timeout 1:` in terminal. 

After changing the radio channel to radio://0/80/2M, the problem still exists. In addition, I tried `roslaunch crazyflie_demo teleop_ps3.launch uri:=radio://0/80/2M` and get

```
auto-starting new master
process[master]: started with pid [30197]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to dc0d69e8-f98f-11e8-9b46-38baf81d0dcb
process[rosout-1]: started with pid [30210]
started core service [/rosout]
process[crazyflie_server-2]: started with pid [30226]
process[crazyflie/crazyflie_add-3]: started with pid [30228]
process[crazyflie/joy-4]: started with pid [30235]
process[crazyflie/quadrotor_teleop-5]: started with pid [30251]
process[crazyflie/crazyflie_demo_controller-6]: started with pid [30269]
process[rviz-7]: started with pid [30280]
process[rqt_plot2-8]: started with pid [30287]
process[rqt_plot3-9]: started with pid [30293]
[ INFO] [1544125690.883326165]: wait_for_service /add_crazyflie
[ INFO] [1544125690.884535959]: found /add_crazyflie
[ INFO] [1544125690.889757537]: Adding radio://0/80/2M as crazyflie with trim(0.000000, 0.000000). Logging: 1, Parameters: 1, Use ROS time: 1
[ INFO] [1544125690.934826344]: Opened joystick: /dev/input/js0. deadzone_: 0.050000.
[ERROR] [1544125691.894307424]: Exception thrown while processing service call: timeout
[ERROR] [1544125691.894474443]: Service call failed: service [/add_crazyflie] responded with an error: timeout
[crazyflie/crazyflie_add-3] process has finished cleanly
log file: /home/xx/.ros/log/dc0d69e8-f98f-11e8-9b46-38baf81d0dcb/crazyflie-crazyflie_add-3*.log
^C[rqt_plot3-9] killing on exit
[rqt_plot2-8] killing on exit
[rviz-7] killing on exit
[crazyflie/crazyflie_demo_controller-6] killing on exit
[crazyflie/quadrotor_teleop-5] killing on exit
[crazyflie/joy-4] killing on exit
[crazyflie_server-2] killing on exit
Traceback (most recent call last):
  File "/home/xx/crazyswarm/ros_ws/src/crazyflie_ros/crazyflie_demo/scripts/controller.py", line 62, in <module>
    controller = Controller(use_controller, joy_topic)
  File "/home/xx/crazyswarm/ros_ws/src/crazyflie_ros/crazyflie_demo/scripts/controller.py", line 10, in __init__
    rospy.wait_for_service('update_params')
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 159, in wait_for_service
    raise ROSInterruptException("rospy shutdown")
rospy.exceptions.ROSInterruptException: rospy shutdown
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
```
The connection is perfectly normal when I use crazyflie-clients-python/bin/cfclient (work for both sudo and without sudo). I also ran `pc_permissions.sh ` previously to give my local user account the permission to use radio dongle. I can also get `Bus 001 Device 010: ID 1915:7777 Nordic Semiconductor ASA ` when
 I type `lsusb`. Are there anything else in configuration that I might miss?

paradox56, 2018-12-06:
It ends up that I made a mistake in configuration and the connection works now. 

```
NODES
  /
    crazyswarm_server (crazyswarm/crazyswarm_server)
    crazyswarm_teleop (crazyswarm/crazyswarm_teleop)
    joy (joy/joy_node)
    rviz (rviz/rviz)

auto-starting new master
process[master]: started with pid [6010]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to b62ee75a-f995-11e8-9b46-38baf81d0dcb
process[rosout-1]: started with pid [6023]
started core service [/rosout]
process[crazyswarm_server-2]: started with pid [6040]
process[joy-3]: started with pid [6041]
process[crazyswarm_teleop-4]: started with pid [6042]
process[rviz-5]: started with pid [6057]
[ INFO] [1544128204.307599596]: Wait for services...
[ INFO] [1544128204.308108839]: waitForService: Service [/emergency] has not been advertised, waiting...
[ INFO] [1544128204.329047885]: waitForService: Service [/emergency] is now available.
[ INFO] [1544128204.330472666]: Manager ready.
ch: 1
[ INFO] [1544128204.350816965]: Opened joystick: /dev/input/js0. deadzone_: 0.050000.
[ INFO] [1544128204.396498440]: Adding CF: cf1 (radio://0/80/2M/E7E7E7E701, cf1)...
[ INFO] [1544128204.401687093]: CF ctor: 0.005142 s
[ INFO] [1544128204.409530424]: [cf1] Requesting parameters...
[ INFO] [1544128204.417107601]: Found variables in cache.
[ INFO] [1544128204.632556802]: [cf1] reqParamTOC: 0.230840 s
[ INFO] [1544128204.632576661]: Requesting memories...
[ INFO] [1544128204.639971567]: Memories: 5
[ INFO] [1544128204.663884930]: [cf1] Ready. Elapsed: 0.262170 s
[ INFO] [1544128204.663905022]: CF run: 0.262226 s
update commander/enHighLevel to 1
update ring/effect to 16
update ring/headlightEnable to 0
update ring/solidBlue to 255
update ring/solidGreen to 0
update ring/solidRed to 0
update stabilizer/controller to 2
update stabilizer/estimator to 2
update ctrlMel/i_range_m_z to 1500
update ctrlMel/i_range_xy to 2
update ctrlMel/i_range_z to 0.4
update ctrlMel/kR_xy to 70000
update ctrlMel/kR_z to 60000
update ctrlMel/kd_omega_rp to 200
update ctrlMel/kd_xy to 0.2
update ctrlMel/kd_z to 0.4
update ctrlMel/ki_m_z to 500
update ctrlMel/ki_xy to 0.05
update ctrlMel/ki_z to 0.05
update ctrlMel/kp_xy to 0.4
update ctrlMel/kp_z to 1.25
update ctrlMel/kw_xy to 20000
update ctrlMel/kw_z to 12000
update ctrlMel/mass to 0.032
update ctrlMel/massThrust to 132000
[ INFO] [1544128204.676034358]: [cf1] Update parameters
[ INFO] [1544128204.715525274]: Update params: 0.051601 s
[ INFO] [1544128204.715660258]: Started 1 threads
[ INFO] [1544128218.070790118]: Takeoff!
[ INFO] [1544128218.200667741]: Takeoff!
^C[rviz-5] killing on exit

```
That being said, the drone immediately flipped to one side when I pressed R2 on my PS4 controller to let it take off. Closing this particular issue

closed 2018-12-06 by paradox56