USC-ACTLab/crazyswarm #124: Trajectory optimization

jchen93nj, 2018-12-16:
Hi,

I was trying to find some useful script to optimize drone trajectory, basically using the state(pos, vel, acc, jerk, etc) of a start point and a goal point for each segment and to generate a smooth trajectory passing each waypoints. Then I found the script demo_optimize_waypoints.py in old folder which seems to have everything I want. However after fixing some bugs and running, I ended up with getting the error like this:

('x:', [[1, 0, 0, 0], [1], [1, 0, 0, 0]])
('y:', [[0, 0, 0, 0], [0], [1, 0, 0, 0]])
('z:', [[0, 0, 0, 0], [0, 0, 0, 0], [0, 0, 0, 0]])
('yaw:', [[0, 0, 0, 0], [0, 0, 0, 0], [0, 0, 0, 0]])
Traceback (most recent call last):
  File "demo_optimize_waypoints.py", line 40, in <module>
    traj.optimize_waypoints(x_wp, y_wp, z_wp, yaw_wp, duration)
  File "/home/junchen/crazyswarm/ros_ws/src/crazyswarm/scripts/trajectory.py", line 76, in optimize_waypoints
    T = path.optimize()
  File "/home/junchen/crazyswarm/ros_ws/src/crazyswarm/scripts/snap.py", line 61, in optimize
    Tr_res = minimize(self.total_cost, Tr0, method='Nelder-Mead', options={'disp':False, 'maxiter':20})
  File "/home/junchen/.local/lib/python2.7/site-packages/scipy/optimize/_minimize.py", line 591, in minimize
    return _minimize_neldermead(fun, x0, args, callback, **options)
  File "/home/junchen/.local/lib/python2.7/site-packages/scipy/optimize/optimize.py", line 552, in _minimize_neldermead
    fsim[k] = func(sim[k])
  File "/home/junchen/.local/lib/python2.7/site-packages/scipy/optimize/optimize.py", line 293, in function_wrapper
    return function(*(wrapper_args + args))
  File "/home/junchen/crazyswarm/ros_ws/src/crazyswarm/scripts/snap.py", line 92, in total_cost
    return self.x.cost(T) + self.y.cost(T) + self.z.cost(T)
  File "/home/junchen/crazyswarm/ros_ws/src/crazyswarm/scripts/snap.py", line 351, in cost
    self.dP = -np.linalg.inv(RPP)
  File "/usr/lib/python2.7/dist-packages/numpy/linalg/linalg.py", line 526, in inv
    ainv = _umath_linalg.inv(a, signature=signature, extobj=extobj)
  File "/usr/lib/python2.7/dist-packages/numpy/linalg/linalg.py", line 90, in _raise_linalgerror_singular
    raise LinAlgError("Singular matrix")
numpy.linalg.linalg.LinAlgError: Singular matrix

I noticed that snap.py was written by someone else but I am still wondering if you can provide some tips of how to get a smooth trajectory using states of each waypoints. The exact thing I want to get is the coefficients of the trajectory polynomials.

Thank you so much!

Best,
Jun

whoenig, 2018-12-18:
We should probably delete those files - they are not used. Here are some resources you can look at:

* Single robot: https://github.com/whoenig/uav_trajectories
* Multi-Robot: https://github.com/jpreiss/smoothener, https://github.com/baskinburak/mrtrajreplan-dars2018 (you can find related publications at act.usc.edu).

jchen93nj, 2018-12-18:
Great! These are really useful. I'll look at them. Thank you so much for sharing.

closed 2019-01-04 by whoenig