USC-ACTLab/crazyswarm #125: LED Solid Red in launch hover_swarm.launch

glennliu, 2019-01-03:
Hi @whoenig 

So far I did a lot of tests on crazyswarm. In launch ``hover_swarm.launch``, cf LED is always in solid red. I tried it with libmotioncapture, vrpn and mocap_optitrack. I have topic /tf receive data at 120Hz and rviz showed cf pose correct, but LED never goes right. And so it cause cf flip every time it takeoff.

Is there possible reasons that can cause LED in red other than mocap data wrong?

Thanks

glennliu, 2019-01-08:
The LED ring board attached to the bottom of CF board.

whoenig, 2019-01-08:
This indicates that the broadcast communication is not working correctly, or that you didn't use the correct configuration for your firmware. Make sure that:
a) the radio firmware is flashed,
b) the nrf firmware is flashed,
c) the crazyswarm stm firmware is flashed (otherwise the LED ring indication won't work), and
d) the crazyswarm firmware is selected in the launch file (https://github.com/USC-ACTLab/crazyswarm/blob/master/ros_ws/src/crazyswarm/launch/hover_swarm.launch#L10).

glennliu, 2019-01-09:
For d), the crazyswarm firmware is correctly selected in launch file.
For a)-c), I flashed the prebuild firmware. Besides, chooser.py runs correct, which should prove radio and nrf firmware are correct. 
Maybe I should reflash stm firmware and try again?

whoenig, 2019-01-10:
Chooser.py does not use broadcast commands, but point-to-point communication, so chooser.py working unfortunately doesn't mean that the broadcasts are working correctly. When you flash, try to use crazyflie_tools or a JTAG debugger, as those are currently the only two ways of ensuring integrity of the flashed file.

glennliu, 2019-01-10:
I was using cfloader to flash firmwrae. I'll try with crazyfile_tools to
flash.

Thanks

-- 
刘楚豪
Chuhao Liu

Hong Kong University of Science and Technology
Integrated System and Design, School of Engineering
Room 2014, CYT Building, HKUST
Kowloon, Hong Kong

Email: chuhao.liu@gmail.com

tel:+86 18928456513 (China)| +852 52235056 (HK)
skype:chuhao.liu

jpreiss, 2020-06-13:
@glennliu were you able to resolve  this?

glennliu, 2020-06-15:
@jpreiss I didn't solve it.

jpreiss, 2020-09-28:
It seems like the problem here is essentially getting the right STM32 firmware onto the Crazyflie. Please reopen if there is any further diagnostic information.

closed 2020-09-28 by jpreiss