USC-ACTLab/crazyswarm #126: Plotting crazyflie controller and estimator signals

emmasmith100, 2019-01-04:
I think this is a very interesting project! I am trying to understand what’s possible with this crazyswarm project. After some struggling I was able to successfully run some of the standard scripts in crazyswarm/ros_ws/src/crazyswarm/scripts
I have a question about plotting and saving data/signals of a specific crazyflie.
When running for example niceHover.py (I slightly adapted it so it follows some waypoints), how can I plot desired/actual position/orientation (is it expressed with quaternions or RPY?), velocity, acceleration, in x,y,z /, motor forces, raw and estimated attitude, ... during that flight experiment?
I would like to make some plots of this data for some experiments to see what happens with the signals in the estimator/controller.
I am not sure, but I thought this could be done with logging in the hover_swarm.launch code. But I don’t really understand how to use it, neither I see which signals can be logged in the default code... :(
I never used ROS before, neither I used much Python or C++ so sorry if it are basic questions… I am more experienced with Matlab/simulink.
Thank you!!

whoenig, 2019-01-04:
The radio bandwidth is quite limited to get all the information you are interested in. If you just fly a single CF, you can get around 5 logging variables in realtime at about 100 Hz (however, the on-board control loop runs at 250Hz). If you are fine with post-processing data, I suggest using the uSD card deck instead. This allows you to log all the data at once (there is even some preliminary version that can gather data synchronous with the loop, so you won't miss any information). You can find more information about data logging with the uSD card deck here: https://wiki.bitcraze.io/projects:crazyflie2:expansionboards:microsd#data_logging. 

For the real-time logging, you can select logging variable names: https://github.com/USC-ACTLab/crazyswarm/blob/master/ros_ws/src/crazyswarm/launch/hover_swarm.launch#L15-L20; enable logging: https://github.com/USC-ACTLab/crazyswarm/blob/master/ros_ws/src/crazyswarm/launch/hover_swarm.launch#L54; and plot: https://github.com/USC-ACTLab/crazyswarm/blob/master/ros_ws/src/crazyswarm/launch/hover_swarm.launch#L70.

In all cases, logging variables are defined in the firmware, and there is no documentation apart from the firmware code as of now. If you have trouble finding a variable you are interested in, the official Bitcraze forum is usually very helpful: https://forum.bitcraze.io/.

whoenig, 2019-11-12:
Please create a new issue if you have further questions.

closed 2019-11-12 by whoenig