USC-ACTLab/crazyswarm #127: Online cmdPosition & cmdFullState

hanzheteng, 2019-01-14:
Hi @whoenig ,

Nice to see your latest update on cmdFullState!

From the most recent commit, I noticed that you added trajectories for takeoff and landing processes. Does it mean we can only do offline planning for now?

I tried my script with `goTo` and `cmdFullState` API. The crazyflie responses to `goTo` command but ignores `cmdFullState` command. Does it have some relation with the high-level/low-level command privilege? 

Usually it should response to `cmdFullState` (low level cmd) first, even though some high level command like `goTo` is running. (In my case they are not running at the same time, I run `goTo` first and then `cmdFullState`.) 

My goal is to make the crazyflie track a wand in my hand. I also added similar structures in crazyflie.py so that I can use `cmdPosition` API as well, because I saw this part was added to crazyswarm_server node but not in crazyflie.py wrapper yet. But anyway, it's not working just like `cmdFullState`.
 
By the way, on the ros node/topic graph, I can only see `crazyflieAPI` node publish `/cf1/cmdFullState` topic, but no other node subscribe it. The `crazyswarm_server` node only publish to `/tf` and  `/pointCloud` topic.

Thank you again for your great contribution.

whoenig, 2019-01-15:
The offline trajectories are just examples. Of course, you can generate setpoints dynamically instead (as long as you can generate it realtime, it should be fine).

You are right that there is a priority, and that `cmdFullState` is given higher priority. Did you enable the high level mode in the firmware by setting the `enHighLevel` parameter to 1?

`cmdPosition` should work in a similar fashion, although I did not test it. Is your code somewhere online in a fork?

`crazyswarm_server` should subscribe to that topic, see https://github.com/USC-ACTLab/crazyswarm/blob/master/ros_ws/src/crazyswarm/src/crazyswarm_server.cpp#L207.

hanzheteng, 2019-01-15:
Thank you for your quick reply. The main issue is that the crazyflie doesn't response to real-time `cmdPosition` or `cmdFullState` commands in the current framework. I have to check firmware later on. Thanks for this info. We do have a fork repo but I haven't pushed the latest code yet. Will get back to this later. Thanks!

whoenig, 2019-01-15:
Does executing `cmdFullState.py` work for you? That's the script I tested the `cmdFullState` topic with in our lab. I never tested `cmdPosition`.

hanzheteng, 2019-01-21:
Thank you for the explanation to firmware. 

You're right. The `commander.c` file shows that `enHighLevel` parameter is set to `False` by default. That's the reason why the crazyflie didn't response to my `takeoff` service call. After executing `test_high_level.py` script for once, it can response to the service call. Thanks!

However, during my experiments, it seems like `test_high_level.py` can only be used when Kalman filter is enabled (estimator 2). The robot will go to random places if I set the estimator to be 1 or 3. 

Also, if using VICON (by publishing to `external position`) and flow deck at the same time, it seems like the robot will consider flow desk first as the localization method, because I saw the robot swing a bit before reaching some certain height.

closed 2019-02-01 by hanzheteng