USC-ACTLab/crazyswarm #135: Build Error

arxk9, 2019-02-22:
When trying to build the project with ./build.sh on ubuntu 16.04 and ros-kinetic, we get this error:
```
####
#### Running command: "make -j8 -l8" in "/home/um/crazyswarm/ros_ws/build"
####
[  0%] Built target std_msgs_generate_messages_cpp
[  1%] Built target crazyflie_controller
[  1%] Built target geometry_msgs_generate_messages_cpp
make[2]: *** No rule to make target '/usr/lib/x86_64-linux-gnu/libproj.so', needed by '/home/um/crazyswarm/ros_ws/devel/lib/liblibobjecttracker.so'.  Stop.
[  6%] Built target ViconDataStreamSDK_CPP
CMakeFiles/Makefile2:1053: recipe for target 'externalDependencies/libobjecttracker/CMakeFiles/libobjecttracker.dir/all' failed
make[1]: *** [externalDependencies/libobjecttracker/CMakeFiles/libobjecttracker.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[  7%] Built target quadrotor_teleop
[ 11%] Built target crazyflie_cpp
[ 11%] Built target _crazyflie_driver_generate_messages_check_deps_GoTo
[ 11%] Built target _crazyflie_driver_generate_messages_check_deps_AddCrazyflie
[ 11%] Built target _crazyflie_driver_generate_messages_check_deps_Takeoff
[ 11%] Built target _crazyflie_driver_generate_messages_check_deps_Hover
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j8 -l8" failed
```

This probably has something to do with libmotioncapture, but we don't really know how to resolve it.

whoenig, 2019-02-22:
I have not seen this issue before and I am not sure what `libproj.so` is. In `ros_ws` folder, could you run `catkin_make -j1 -k` twice and post the output of the second run?

arxk9, 2019-02-22:
Absolutely! Here's the output of the second run:
```
Base path: /home/um/crazyswarm/ros_ws
Source space: /home/um/crazyswarm/ros_ws/src
Build space: /home/um/crazyswarm/ros_ws/build
Devel space: /home/um/crazyswarm/ros_ws/devel
Install space: /home/um/crazyswarm/ros_ws/install
####
#### Running command: "make cmake_check_build_system" in "/home/um/crazyswarm/ros_ws/build"
####
####
#### Running command: "make -j1 -k" in "/home/um/crazyswarm/ros_ws/build"
####
[  4%] Built target crazyflie_cpp
[  5%] Built target listLogVariables
[  6%] Built target scan
[  7%] Built target listParams
[  8%] Built target listMemories
[  9%] Built target reboot
[ 10%] Built target battery
[ 12%] Built target version
[ 15%] Built target console
[ 16%] Built target flash
[ 21%] Built target ViconDataStreamSDK_CPP
make[2]: *** No rule to make target '/usr/lib/x86_64-linux-gnu/libproj.so', needed by '/home/um/crazyswarm/ros_ws/devel/lib/liblibmotioncapture.so'.
make[2]: Target 'externalDependencies/libmotioncapture/CMakeFiles/libmotioncapture.dir/build' not remade because of errors.
CMakeFiles/Makefile2:904: recipe for target 'externalDependencies/libmotioncapture/CMakeFiles/libmotioncapture.dir/all' failed
make[1]: *** [externalDependencies/libmotioncapture/CMakeFiles/libmotioncapture.dir/all] Error 2
[ 22%] Built target ViconDataStreamSDK_CPPTest
make[2]: *** No rule to make target '/usr/lib/x86_64-linux-gnu/libproj.so', needed by '/home/um/crazyswarm/ros_ws/devel/lib/liblibobjecttracker.so'.
make[2]: Target 'externalDependencies/libobjecttracker/CMakeFiles/libobjecttracker.dir/build' not remade because of errors.
CMakeFiles/Makefile2:1053: recipe for target 'externalDependencies/libobjecttracker/CMakeFiles/libobjecttracker.dir/all' failed
make[1]: *** [externalDependencies/libobjecttracker/CMakeFiles/libobjecttracker.dir/all] Error 2
[ 23%] Built target quadrotor_teleop
[ 24%] Built target crazyflie_controller
[ 24%] Built target _crazyflie_driver_generate_messages_check_deps_GoTo
[ 24%] Built target std_msgs_generate_messages_cpp
[ 24%] Built target geometry_msgs_generate_messages_cpp
[ 24%] Built target _crazyflie_driver_generate_messages_check_deps_Takeoff
[ 24%] Built target _crazyflie_driver_generate_messages_check_deps_AddCrazyflie
[ 24%] Built target _crazyflie_driver_generate_messages_check_deps_Hover
[ 24%] Built target _crazyflie_driver_generate_messages_check_deps_crtpPacket
[ 24%] Built target _crazyflie_driver_generate_messages_check_deps_Land
[ 24%] Built target _crazyflie_driver_generate_messages_check_deps_LogBlock
[ 24%] Built target _crazyflie_driver_generate_messages_check_deps_SetGroupMask
[ 24%] Built target _crazyflie_driver_generate_messages_check_deps_StartTrajectory
[ 24%] Built target _crazyflie_driver_generate_messages_check_deps_GenericLogData
[ 24%] Built target _crazyflie_driver_generate_messages_check_deps_sendPacket
[ 24%] Built target _crazyflie_driver_generate_messages_check_deps_TrajectoryPolynomialPiece
[ 24%] Built target _crazyflie_driver_generate_messages_check_deps_UpdateParams
[ 24%] Built target _crazyflie_driver_generate_messages_check_deps_Position
[ 24%] Built target _crazyflie_driver_generate_messages_check_deps_UploadTrajectory
[ 24%] Built target _crazyflie_driver_generate_messages_check_deps_RemoveCrazyflie
[ 24%] Built target _crazyflie_driver_generate_messages_check_deps_FullState
[ 24%] Built target _crazyflie_driver_generate_messages_check_deps_Stop
[ 24%] Generating C++ code from crazyflie_driver/GenericLogData.msg
Traceback (most recent call last):
  File "/opt/ros/kinetic/share/gencpp/cmake/../../../lib/gencpp/gen_cpp.py", line 41, in <module>
    import genmsg.template_tools
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/genmsg/template_tools.py", line 39, in <module>
    import em
ImportError: No module named 'em'
crazyflie_ros/crazyflie_driver/CMakeFiles/crazyflie_driver_generate_messages_cpp.dir/build.make:80: recipe for target '/home/um/crazyswarm/ros_ws/devel/include/crazyflie_driver/GenericLogData.h' failed
make[2]: *** [/home/um/crazyswarm/ros_ws/devel/include/crazyflie_driver/GenericLogData.h] Error 1
[ 25%] Generating C++ code from crazyflie_driver/crtpPacket.msg
Traceback (most recent call last):
  File "/opt/ros/kinetic/share/gencpp/cmake/../../../lib/gencpp/gen_cpp.py", line 41, in <module>
    import genmsg.template_tools
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/genmsg/template_tools.py", line 39, in <module>
    import em
[... 302 lines cut ...]
seaborn==0.9.0
sensor-msgs==1.12.7
six==1.12.0
smach==2.0.1
smach-ros==2.0.1
smclib==1.8.3
stable-baselines==2.4.1
tensorboard==1.12.2
tensorflow-gpu==1.12.0
termcolor==1.1.0
tf==1.11.9
tf-conversions==1.11.9
tf2-kdl==0.5.20
tf2-py==0.5.20
tf2-ros==0.5.20
topic-tools==1.12.14
tqdm==4.31.1
urllib3==1.24.1
vcstools==0.1.40
vispy==0.5.3
Werkzeug==0.14.1
wstool==0.1.17
xacro==1.11.3
zmq==0.0.0
```

arxk9, 2019-02-22:
Ok so i tried a `pip install em`, and now I get this:

```
File "/opt/ros/kinetic/lib/python2.7/dist-packages/genmsg/template_tools.py", line 77, in _generate_from_spec
    interpreter = em.Interpreter(output=ofile, globals=g, options={em.RAW_OPT:True,em.BUFFERED_OPT:True})
AttributeError: 'module' object has no attribute 'Interpreter'
```

whoenig, 2019-02-22:
This seems like a generic ROS problem, perhaps this will help: https://answers.ros.org/question/257331/python-module-empy-missing-tutorials/ ?

arxk9, 2019-02-26:
Ok thank you! We fixed the issue by using system python instead of a virtualenv.

closed 2019-02-26 by arxk9