USC-ACTLab/crazyswarm #136: Unable to connect to Vicon System

arya-k, 2019-02-25:
Hey,

The documentation isn't super clear on how I'm supposed to connect to a Vicon system:

When I try to run `mocap_helper` it just hangs, because it isn't ableto connect to the client. How do I figure out my `vicon_host_name`? I currently have the desktop running crazyswarm directly connected to the PoE switch. Is this config ok?

Here's my `mocap_helper.launch` file:
```
<?xml version="1.0"?>
<launch>
  <node pkg="crazyswarm" type="mocap_helper" name="mocap_helper" output="screen" >
    <rosparam>
      # tracking
      motion_capture_type: "vicon" # one of vicon,optitrack,qualisys,vrpn
      object_tracking_type: "libobjecttracker" # one of motionCapture,libobjecttracker
      vicon_host_name: "vicon"
      # optitrack_local_ip: "localhost"
      # optitrack_server_ip: "optitrack"
      # qualisys_host_name: "10.0.5.219"
      # qualisys_base_port: 22222
      # vrpn_host_name: "vicon"
    </rosparam>
  </node>
</launch>
```

jpreiss, 2019-02-25:
No, you need to have 2 network cards in the vicon machine, one for the poe
switch and another for a LAN shared with the crazyswarm Linux PC.

arya-k, 2019-02-26:
Ok thank you! I'm still a little confused though. So the physical setup has to look like this, right?
`Crazyswarm Linux PC <-- ethernet --> Windows VICON PC <-- ethernet --> VICON Switch`

1. Do I specifically need two network cards, or would a network card and a switch suffice?
2. On the VICON PC, do I just run Tracker? How do I make it communicate with the Linux PC?

Thank you!

whoenig, 2019-02-26:
1. You need two cards.
2. Yes, Tracker runs a server, the Crazyswarm operates as client using the DataStream SDK.

I suggest that you read the VICON documentation, in particular for the physical setup.

hanzheteng, 2019-02-26:
You can buy a 1000 rate Ethernet to USB 3.0 adapter as the second card.

jpreiss, 2019-02-27:
@arya-k probably more like

    crazyswarm Linux PC --- ethernet switch --- windows VICON PC --- VICON switch
                                   |
                                   +----------- internet connection

 @hanzheteng have you tested that configuration, using the usb3-ethernet adapter? Did you measure latency?

hanzheteng, 2019-03-13:
@jpreiss  For time being, we use this configuration in our lab. It works well with our small scale team. I used to test it by normal `ping` command and get an average of 0.7ms transmission speed. For a large group of robots, I do agree that it's better to use a PC with two network cards.

closed 2019-03-15 by whoenig

Hurisa, 2020-10-27:
> @arya-k probably more like
>
> ```
> crazyswarm Linux PC --- ethernet switch --- windows VICON PC --- VICON switch
>                                |
>                                +----------- internet [...]

Hi @jpreiss,

I have that same configuration but when I run the mocap helper it still hangs without outputting anything... is there any troubleshoot you suggest?

Thanks

Hurisa, 2020-10-27:
ok! Quick update

Issue #37 solved it