USC-ACTLab/crazyswarm #137: Unable to connect to Crazyflie

whoenig, 2019-03-04:
Hi,

I believe that I am facing the same/similar issue when running either of the following commands using ROS;

```
cd ros_ws/src/crazyswarm/scripts
python chooser.py
```
or 

```
source ros_ws/devel/setup.bash
roslaunch crazyswarm hover_swarm.launch
```

I have read and understood majority of this existing issue raised and have thereafter implemented the changes I deemed necessary to try and diagnose the problem better and try to get the command(s) to work. However, I have now spent much time on this and can not get it to work. I receive the following issue/error when I run either of the commands (as stated above);

`timeout`

or 

```
[crazyswarm_server-2] process has died [pid 25697, exit code -6, cmd /home/huzayfaa/crazyswarm/ros_ws/devel/lib/crazyswarm/crazyswarm_server __name:=crazyswarm_server __log:=/home/huzayfaa/.ros/log/061f4cf4-39f2-11e9-ad17-e839355c91e0/crazyswarm_server-2.log].
log file: /home/huzayfaa/.ros/log/061f4cf4-39f2-11e9-ad17-e839355c91e0/crazyswarm_server-2*.log
```

I have made the following changes to the configuration files, as stated in the installation guide
(https://crazyswarm.readthedocs.io/en/latest/usage.html) and some of the older issues that are now solved.The process I follow  includes: 

1. Assigned CF2.0 using a USB cable and the cfclient to **uri:=radio://0/100/2M/E7E7E7E701**
(Using the cfclient I can then scan for the CF2.0 on that address and channel and am able to
connect to it too)
2. Upgraded the firmware of the CF2.0 using the cfclient and bootloader mode to release **2019.02.1**
3. Upgraded the firmware of the Crazyradio using option 2 (prebuilt binary) and the version of the Crazyradio
is now **99.55**
4. Ran the ** ./pc_permissions.sh** script and this too has executed successfully
5. My **allCrazyflies.yaml** configuration file reads
```
crazyflies:
  - id: 1
    channel: 100
    initialPosition: [1.5, 1.5, 0.0]
    type: default
```
6. ** crazyflieTypes.yaml** - No changes were made to this configuration file
7. The following changes were made to ** hover_swarm.launch**
```
firmware: "bitcraze"
estimator: 2
motion_capture_type: "none"
object_tracking_type: "motionCapture"
```

The cfclient works absolutely fine and I am able to connect the CF2.0 using the same channel and address whenusing the client. However, both **chooser.py** and **hover_swarm.launch** do not work  and thia I understand may be due to the timeout. 

I understand that this may be a connection issue and that the Crazyradio is unable to connect to the CF. However, I don’t understand what I am doing wrong or even not doing and how to go about fixing this issue. I have two crazyflies and have now tried with both. 

Many thanks in advance, any help and direction would be highly appreciated.

------------------------------------------------------------

p.s. when I run `./build.sh` it all seems to compile fine and reaches 100%, however the following warning message is also displayed four times and I believe this relates to Issue #16 

```
CMake Warning at /opt/ros/kinetic/share/catkin/cmake/catkin_package.cmake:166 (message):
  catkin_package() DEPENDS on '/usr/lib/x86_64-linux-gnu/libusb-1.0.so' but
  neither '/usr/lib/x86_64-linux-gnu/libusb-1.0.so_INCLUDE_DIRS' nor
  '/usr/lib/x86_64-linux-gnu/libusb-1.0.so_LIBRARIES' is defined.
```

_Originally posted by @Huzayfaa in https://github.com/USC-ACTLab/crazyswarm/issues/105#issuecomment-467553874_

whoenig, 2019-03-04:
The build warning should not be an issue. In order to troubleshoot, can you try the command line tools:

```
rosrun crazyflie_tools scan --address E7E7E7E701
```
It should find the Crazyflie as well. Then try:
```
rosrun crazflie_tools reboot --uri radio://0/100/2M/E7E7E7E701
```
and see if your Crazyflie reboots successfully.

Finally, did you update your NRF firmware as well? I am not sure if the firmware release you mentioned includes both STM and NRF firmwares.

Huzayfaa, 2019-03-04:
Hi, 

I have tried and achieved the following results

`rosrun crazyflie_tools scan --address E7E7E7E701`
ROS reply `radio://0/100/2M/E7E7E701` 

`rosrun crazyflie_tools reboot --uri radio://0/100/2M/E7E7E7E701`
CF2.0 also respond to this command, however not always and if I remove and plug the CrazyRadio in again and/or reboot the CF2.0 this fixes it and I am then able to re-send the command it and works. 

I have also updated the CF2.0 firmware following **Option 2** as part of the 'usage' section in the latest documentation and then re-ran the commands and same behaviour as above.   

Continuing with the latest documentation, I run chooser.py and this detects the battery just fine, however, when I select sysOff or reboot, it very rarely works and on most occasions the ROS terminal displays timeout - whereas, again, when I remove the radio and plug it back then it seems to be fine. I don't seem to understand why the battery function works absolutely fine but the other two flag up inconsistencies. 

Also, when running the script thereafter (roslaunch crazyswarm hover_swarm.launch) it does not flag up with the pid error like before and is able to run without any errors. However, the .rviz window is always empty and also when the CF is commanded to take off it flies out and flips over. I believe this may have something to do with the configuration of the files and the positioning system I am using? 

I am using LPS in TDoA3 and have made the following changes  to ** hover_swarm.launch**
```
firmware: "bitcraze"
estimator: 2
motion_capture_type: "none"
object_tracking_type: "motionCapture" 
```
Many thanks for your help! 

Huzayfaa

whoenig, 2019-03-04:
Hi,

The behavior of your Crazyradio is strange. I have seen this kind of behavior when the cfclient is used in between (which seems to alter the state somehow), and using the latest NRF firmware (which I need to debug), but not with the prebuilt firmwares that are part of the Crazyswarm.

If you use LPS, it is normal that the rviz window is empty. If you use a motion capture system, rviz will show the current position estimate by the mocap. In order to debug your crashes, I suggest that you add logging of your current state position estimate (`stateEstimate.x`,`stateEstimate.y`,`stateEstimate.z`) here: https://github.com/USC-ACTLab/crazyswarm/blob/master/ros_ws/src/crazyswarm/launch/hover_swarm.launch#L19. Then enable logging https://github.com/USC-ACTLab/crazyswarm/blob/master/ros_ws/src/crazyswarm/launch/hover_swarm.launch#L54, and plot the data (e.g., similar to https://github.com/USC-ACTLab/crazyswarm/blob/master/ros_ws/src/crazyswarm/launch/hover_swarm.launch#L69). Move your CF around the space manually (w/o takeoff) and make sure that the state estimate is roughly correct. If that works, try changing the controller to PID (https://github.com/USC-ACTLab/crazyswarm/blob/master/ros_ws/src/crazyswarm/launch/hover_swarm.launch#L26), the mellinger controller has very high gains which might cause issues if the state estimate is noisy.

Huzayfaa, 2019-03-06:
Hi, 

I'd like to start by thanking you for your support. 

I have successfully been able to follow the instructions as above and using the following line of code am also able to plot the stateEstimator values

`<node pkg="rqt_plot" type="rqt_plot" name="rqt_plot_xyz" args="/cf1/log1/values[0] /cf1/log1/values[1] /cf1/log1/values[2]"/>`

I believe I had flashed the CF2.0 incorrectly which was causing the drift and this has now been fixed by updating the firmware again (both nrf and stm). This also means that following the latest installation guide I was able to achieve both **Basic** and **Advance Flight**. So, Thank You! 

However, the issue with the Crazyradio is still present and I have to always remove and re-connect the Crazyradio after each run - which is concerning. BUt - only then does the 'timeout' or 'process has died' error not flag up (although at times it still occasionally does). 

Would you say that this sort of behaviour is only observed in some of the Crazyradios? would a new Crazyradio help address this issue?

Thanks, 
Huzayfaa

whoenig, 2019-03-06:
Good to hear that it is partially working now!

The behavior you observe is certainly not normal. By "after each run" you mean after you killed the `hover_swarm` launch file and try to run it again? Note that at the end (i.e., Ctrl+C) of the crazyswarm_server, all CFs are set into sysoff (that is, the STM32 is powered off to save battery). Before reconnecting, you'll need to reboot the CFs using chooser.py.

I have not seen any difference between different Crazyradios. I have seen one PC where it matters which USB port to use, but that seems a PC HW problem (i.e., it happens with both cfclient and crazyflie_ros with any Crazyradio).

Huzayfaa, 2019-03-06:
Hi, 

Yes, when I used to kill the launch file, I had to then each time remove the radio and restart the CF. However, I now understand completely why and following your method of utilising chooser.py, this addresses the issue (mostly, but not always) and allows me to repeat the hover_swarm launch file without a timeout error. 

I only now get the following warnings and I would like to believe that these warning are what are also affecting chooser.py - whereby it doesn't always reboot as per the the command and only really does it about 30% of the time. 

[ WARN] [1551897997.422150153]: [cf1] Link Quality low (0.550000)
[ WARN] [1551897999.666812112]: [cf1] Link Quality low (0.620000)
[ WARN] [1551898003.062417428]: [cf1] Link Quality low (0.210000)
[ WARN] [1551898004.186780398]: [cf1] Link Quality low (0.580000)
[ WARN] [1551898005.308186843]: [cf1] Link Quality low (0.450000)
[ WARN] [1551898008.678308459]: [cf1] Link Quality low (0.550000)
[ WARN] [1551898016.540881859]: [cf1] Link Quality low (0.610000)
[ WARN] [1551898025.542165428]: [cf1] Link Quality low (0.670000)

But apart from this, and while these warning were being displayed, the basic and advance flights (as per the latest installation guide) do operate as would be expected. 

Thanks, 
Huzayfaa

whoenig, 2019-03-06:
This warning indicates that there are a lot of packet drops. I assume that you have a Crazyradio PA? Also, try a different channel (e.g., using https://github.com/bitcraze/crazyflie-lib-python/blob/master/examples/radio-test.py to find a good channel).

AhmedA92, 2019-03-13:
Hello,
First I want to thank you for your support.
Secondly, I still have the same error as here, I followed the old thread and this one, updated the crazyflies and the crazyradio and still get this error: 
`process[master]: started with pid [34898]
ROS_MASTER_URI=http://localhost:11311
setting /run_id to 8e171b5c-458f-11e9-aa8e-f4939ff2c1cb
process[rosout-1]: started with pid [34911]
started core service [/rosout]
process[crazyswarm_server-2]: started with pid [34928]
ERROR: cannot launch node of type [joy/joy_node]: joy
ROS path [0]=/opt/ros/kinetic/share/ros
ROS path [1]=/home/studentadm/Desktop/crazyswarm/ros_ws/src
ROS path [2]=/opt/ros/kinetic/share
process[crazyswarm_teleop-4]: started with pid [34929]
process[rqt_plot_xyz-5]: started with pid [34930]
[ INFO] [1552481848.483980056]: Wait for services...
[ INFO] [1552481848.485107484]: waitForService: Service [/emergency] has not been advertised, waiting...
[ INFO] [1552481848.506696507]: waitForService: Service [/emergency] is now available.
[ INFO] [1552481848.508831485]: Manager ready.
ch: 1
terminate called after throwing an instance of 'XmlRpc::XmlRpcException'
[crazyswarm_server-2] process has died [pid 34928, exit code -6, cmd /home/studentadm/Desktop/crazyswarm/ros_ws/devel/lib/crazyswarm/crazyswarm_server __name:=crazyswarm_server __log:=/home/studentadm/.ros/log/8e171b5c-458f-11e9-aa8e-f4939ff2c1cb/crazyswarm_server-2.log].
log file: /home/studentadm/.ros/log/8e171b5c-458f-11e9-aa8e-f4939ff2c1cb/crazyswarm_server-2*.log
[rqt_plot_xyz-5] process has finished cleanly
log file: /home/studentadm/.ros/log/8e171b5c-458f-11e9-aa8e-f4939ff2c1cb/rqt_plot_xyz-5*.log
`
Also important note: crazyflie-server node dies everytime I change (motion_capture_type: "none") in hover_swarm.launch while this doesn't occure when I actually leave it as "vicon". Still both cases don't get me to connect to the crazyflie.
I tried plotting "stateEstimate" variables but i get blank plot.
however `chooser.py` is working fine (battery and rebooting), `crazyflie_tools` scan is also working fine.
Please any suggestion will be appreciated.

luozhexin, 2019-06-07:
> Hi,
> I have tried and achieved the following results
> rosrun crazyflie_tools scan --address E7E7E7E701
> ROS reply radio://0/100/2M/E7E7E701
> rosrun crazyflie_tools reboot --uri [...]

hi,Sorry to bother you,i have similar problem comparing to yours.and i follow the instruction well in [https://crazyswarm.readthedocs.io/en/latest/usage.html](url),i am sure i have fine crazyflie and PA,but i can not takeoff the crazyflie when i run hover_swarm.launch in a terminal.i am so confused about this.So,can i ask you some question about this?especially the settings needed to be done before i run hover_swarm.launch.thanks a lot.

jpreiss, 2021-01-30:
closing due to inactivity.

closed 2021-01-30 by jpreiss