USC-ACTLab/crazyswarm #140: XmlRpc::XmlRpcException

whoenig, 2019-03-15:
Hello,
First I want to thank you for your support.
Secondly, I still have the same error as here, I followed the old thread and this one, updated the crazyflies and the crazyradio and still get this error: 
`process[master]: started with pid [34898]
ROS_MASTER_URI=http://localhost:11311
setting /run_id to 8e171b5c-458f-11e9-aa8e-f4939ff2c1cb
process[rosout-1]: started with pid [34911]
started core service [/rosout]
process[crazyswarm_server-2]: started with pid [34928]
ERROR: cannot launch node of type [joy/joy_node]: joy
ROS path [0]=/opt/ros/kinetic/share/ros
ROS path [1]=/home/studentadm/Desktop/crazyswarm/ros_ws/src
ROS path [2]=/opt/ros/kinetic/share
process[crazyswarm_teleop-4]: started with pid [34929]
process[rqt_plot_xyz-5]: started with pid [34930]
[ INFO] [1552481848.483980056]: Wait for services...
[ INFO] [1552481848.485107484]: waitForService: Service [/emergency] has not been advertised, waiting...
[ INFO] [1552481848.506696507]: waitForService: Service [/emergency] is now available.
[ INFO] [1552481848.508831485]: Manager ready.
ch: 1
terminate called after throwing an instance of 'XmlRpc::XmlRpcException'
[crazyswarm_server-2] process has died [pid 34928, exit code -6, cmd /home/studentadm/Desktop/crazyswarm/ros_ws/devel/lib/crazyswarm/crazyswarm_server __name:=crazyswarm_server __log:=/home/studentadm/.ros/log/8e171b5c-458f-11e9-aa8e-f4939ff2c1cb/crazyswarm_server-2.log].
log file: /home/studentadm/.ros/log/8e171b5c-458f-11e9-aa8e-f4939ff2c1cb/crazyswarm_server-2*.log
[rqt_plot_xyz-5] process has finished cleanly
log file: /home/studentadm/.ros/log/8e171b5c-458f-11e9-aa8e-f4939ff2c1cb/rqt_plot_xyz-5*.log
`
Also important note: crazyflie-server node dies everytime I change (motion_capture_type: "none") in hover_swarm.launch while this doesn't occure when I actually leave it as "vicon". Still both cases don't get me to connect to the crazyflie.
I tried plotting "stateEstimate" variables but i get blank plot.
however `chooser.py` is working fine (battery and rebooting), `crazyflie_tools` scan is also working fine.
Please any suggestion will be appreciated.

_Originally posted by @AhmedA92 in https://github.com/USC-ACTLab/crazyswarm/issues/137#issuecomment-472428046_

whoenig, 2019-03-15:
@AhmedA92 this is actually unrelated, so I created a new issue.

You get a XmlRpc::XmlRpcException error, which likely means there is an issue with parsing one of the config files (hover_swarm.launch, or crazyflieTypes.yaml, or crazyflies.yaml). Could you double check and if you can't find anything post your files?

AhmedA92, 2019-03-15:
Thank you so much, the problem was with hover_swarm.launch, something is wrong with the characters in my system. In order to fix it i had to copy "none" from the comment in the file and same goes to number "2" when choosing Mellinger controller. 

Thanks again for your patience

closed 2019-03-16 by whoenig