USC-ACTLab/crazyswarm #141: Crazyradio not detected by scan

arya-k, 2019-03-15:
Not really sure where the error is coming from:
the python client is successfully able to connect to and control crazyflies over a crazyradio (without sudo, since permissions are set correctly).

However, `rosrun crazyflie_tools scan` doesn't return any radios despite both firmwares flashing successfully:

```
um@kurtz:~/crazyswarm$ python crazyradio-firmware/usbtools/launchBootloader.py
Launch bootloader .
Bootloader started
um@kurtz:~/crazyswarm$ sudo python crazyradio-firmware/usbtools/nrfbootload.py flash prebuilt/cradio.bin
('Found nRF24LU1 bootloader version', '18.0')
Flashing:
  Flashing 7471 bytes...
Flashing done!
Verifying:
  Reading prebuilt/cradio.bin...
  Reading 7471 bytes from the flash...
Verification succeded!
um@kurtz:~/crazyswarm$ rosrun crazyflie_tools scan
um@kurtz:~/crazyswarm$ 
```

whoenig, 2019-03-16:
Try running with `-v`, i.e., `rosrun crazyflie_tools scan -v` to get some more verbose output.

arya-k, 2019-03-18:
My mistake, when looking at the verbose output i realized that I had not specified the right address. It now works!
```
um@kurtz:~/crazyswarm/ros_ws/src/crazyflie_ros/crazyflie_tools/src$ rosrun crazyflie_tools scan -v --address 0xE7E7E7E701
Found Crazyradio with version 99.55
radio://0/16/250K/E7E7E7E701
radio://0/28/250K/E7E7E7E701
radio://0/48/250K/E7E7E7E701
radio://0/80/250K/E7E7E7E701
radio://0/112/250K/E7E7E7E701
No Crazyflie connected via USB cable found.
```

closed 2019-03-18 by arya-k