USC-ACTLab/crazyswarm #142: Crazyradio only has 250K, not 2M

arya-k, 2019-03-18:
Honestly, I'm not 100% sure what to call this:

When connecting to a crazyflie, all the crazyradio channels have a 250K as the third argument:
```
um@kurtz:~/crazyswarm/ros_ws/src/crazyflie_ros/crazyflie_tools/src$ rosrun crazyflie_tools scan -v --address 0xE7E7E7E701
Found Crazyradio with version 99.55
radio://0/16/250K/E7E7E7E701
radio://0/28/250K/E7E7E7E701
radio://0/48/250K/E7E7E7E701
radio://0/80/250K/E7E7E7E701
radio://0/112/250K/E7E7E7E701
```

However, in the sysOff.py command, it seems that it's looking for a *2M* instead of a *250K*, as in `radio://0/16/2M/E7E7E7E701`

arya-k, 2019-03-18:
More info:
```
um@kurtz:~/crazyswarm$ rosrun crazyflie_tools scan --address E7E7E7E701
radio://0/80/250K/E7E7E701
um@kurtz:~/crazyswarm$ rosrun crazyflie_tools reboot --uri radio://0/100/250K/E7E7E7E701
timeout
```

allCrazyflies.yaml:
```
crazyflies:
  - id: 1
    channel: 100
    initialPosition: [0.0, 0.0, 0.0]
    type: default
```

`mocap_helper` is able to detect the 4 points perfectly.

whoenig, 2019-03-18:
The Crazyswarm assumes that the datarate is set to 2M, which is the datarate which achieves best performance in most indoor settings. You will need to use cfclient to reconfigure your CFs to use the 2M datarate.

closed 2019-03-21 by arya-k