USC-ACTLab/crazyswarm #143: Switching between goTo and sendPositionSetpoint causes motors to stop

williamdwarke, 2019-03-18:
I'm trying to implement a trajectory tracker using your Crazyflie::sendPositionSetpoint function, and it works decently well, however, if I call Crazyflie::goTo and then Crazyflie::sendPositionSetpoint during the same runtime the motors stop and drone falls out of the sky. I believe this is a bug in the firmware and I'm going to try and investigate it myself but I'd appreciate any help you all can provide.

Thanks,
William

whoenig, 2019-03-18:
This switching between low-level control (sendPositionSetpoint) and high-level mode (on-board trajectory computation and evaluation) is currently not supported. There are a number of issues that currently prevent proper switching, including issues in the firmware (bugs and missing features).

williamdwarke, 2019-03-18:
Thanks for the quick response! Does that include takeoff and landing? There's a weird ~3s delay (and drift) that occurs when I go from a takeoff command to a sendPositionSetpoint command. If not I will just make sure and send setpoints for the entire trajectory, including takeoff and landing.

whoenig, 2019-03-18:
Unfortunately yes. The current recommend workaround is to generate a start/landing trajectory yourself (e.g., by using tools in https://github.com/whoenig/uav_trajectories), and start/land using sendPositionSetpoint (or better sendFullPoseSetpoint) instead. I am aware of the drift you are describing - this is part of a "safety" feature in the firmware when switching from high-level back to low-level mode.

williamdwarke, 2019-03-18:
I'm actually using Mellinger's trajectory generation approach, but thanks for the info! This really helps.

hanzheteng, 2019-03-19:
@williamdwarke I have encountered similar issue before. As I remember, I was able to switch from goTo to cmdPosition but not the inverse. I will work on it next week and get back to this.

jpreiss, 2019-03-19:
@hanzheteng you are right, cmdPosition will immediately supersede but there is a timeout period of approx 2 seconds before the controller setpoint reverts to the high level commander. This is really a bug and needs to be fixed at the firmware (and possibly rradio protocol) level.

jpreiss, 2021-01-30:
This feature was implemented in the firmware in https://github.com/bitcraze/crazyflie-firmware/issues/536 and added to Crazyswarm in #202.

closed 2021-01-30 by jpreiss